espressif/esp-idf #5703: Wifi has no reconnect strategy whereas Ethernet, MQTT etc. do

will-emmerson, 2020-08-06:
## Environment
- IDF version: master

## Problem Description
Anyone who uses wifi and wants to maintain a connection over long periods of time (i.e. almost everyone) has to program their own reconnect strategy. This is already built into the ethernet and mqtt components (and probably others I haven't used), and considering the ESP32 is known primarily as a wifi chip it seems strange this is missing.

### Expected Behavior
There should be some kind of `reconnect` flag when initialising a wifi connection which causes connection to be maintained. If the reconnect strategy doesn't fit someone's use-case, they can create their own.

### Actual Behavior
Everyone who uses wifi over long periods of time is surprised the wifi connection isn't maintained, then has to hunt around the documentation/issues/forum looking for a solution which then probably doesn't work that well.

Link to relevant documentation:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-reconnect

will-emmerson, 2020-08-07:
Actually the wifi example does automatically reconnect now so maybe this doesn't matter so much:
https://github.com/espressif/esp-idf/blob/c77c4ccf6c43ab09fd89e7c907bf5cf2a3499e3b/examples/wifi/getting_started/station/main/station_example_main.c#L51

esp-HarveyRong, 2020-08-25:
Hi @will-emmerson,

Due to many reasons for disconnection, WiFi cannot guarantee in which case the application needs to reconnect, so the application is required to perform this operation

HarveyRong-Esp, 2020-11-09:
@will-emmerson，Thanks for reporting, will close due to short of feedback, feel free to reopen.

closed 2020-11-09 by Alvin1Zhang