espressif/esp-idf #5704: MQTT_EVENT_ERROR event missed when use mqtt to connect mqtts broker

AxelLin, 2020-08-06:
## Environment

Module or chip used: ESP32-WROOM-32D
IDF version: v4.3-dev-771-gc77c4ccf6c43
Build System: idf.py
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
Operating System: Linux
Power Supply: USB

## Problem Description

When use mqtt to connect mqtts broker, the mqtt connect will fail as expect.
I expect to get MQTT_EVENT_ERROR event but it does not happen.

E (33429) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=128
E (33430) MQTT_CLIENT: esp_mqtt_connect: mqtt_message_receive() returned -1
E (33434) MQTT_CLIENT: MQTT connect failed

### Expected Behavior
Should get MQTT_EVENT_ERROR if connect failed.

### Actual Behavior
No MQTT_EVENT_ERROR in such test case.

### Steps to reproduce

Just use mqtt to connect mqtts broker.

ESP-Marius, 2020-08-07:
Hi @AxelLin 

I agree, the error reporting/handling in the MQTT client could need some cleanup. We'll take a look at it.

closed 2020-09-16 by espressif-bot

AxelLin, 2020-11-10:
A side-effect of this commit is now I got MQTT_EVENT_ERROR but the error_type is always MQTT_ERROR_TYPE_ESP_TLS even though I connect to mqtt 1883 port.

see https://github.com/espressif/esp-idf/issues/5906

ESP-Marius, 2020-11-10:
@AxelLin There is an in-progress MR to fix #5906, which will also add better reporting of the underlying socket errors. So that should take care of the issue mentioned here as well.