espressif/esp-idf #5706: use HTTP Restful API Server Example with softAP

SERIDJ, 2020-08-06:
hi, @david-cermak  

please i can use this example  : https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/restful_server  
with mode softAP  ? 

I dont know but i change mode connection with wifi_init_softap : 

```

void app_main()
{
    ESP_ERROR_CHECK(nvs_flash_init());
    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    initialise_mdns();

   // ESP_ERROR_CHECK(example_connect());
    ESP_LOGI(TAG, "ESP_WIFI_MODE_AP");
    wifi_init_softap();

    ESP_ERROR_CHECK(init_fs());
    ESP_ERROR_CHECK(start_rest_server(CONFIG_EXAMPLE_WEB_MOUNT_POINT));
}

```

SERIDJ, 2020-08-06:
i thinks that work but just size of windows is not compatible phone or tablette screen

suda-morris, 2020-08-10:
Suggestion:
* Have you tried other browsers (e.g. Chrome)?
* Enable developer tools, to see why the image resources are missing, if it's something related your network issue?

And if you think it's an issue from esp-idf side, can you share your application log? Usually if the http server is not working properly, you can see warnings or errors there.

Alvin1Zhang, 2020-08-18:
Thanks for reporting, will close for now.

closed 2020-08-18 by Alvin1Zhang