espressif/esp-idf #5709: Modbus master SEND example or any reference

MarkEvens, 2020-08-07:
I am trying send data on modbus to modbus slave simulator if i get example of it. This will be very helpful

alisitsyn, 2020-08-07:
Hi @MarkEvens,

Could you clarify which Modbus Simulator you are going to use? I would recommend to use Modbus Poll/Slave software on PC (https://www.modbustools.com/modbus_slave.html). Also it is possible to use pymodbus (https://pymodbus.readthedocs.io/en/latest) as a modbus client/server on PC side.
In this case you can use official examples of esp-idf for serial and tcp: https://github.com/espressif/esp-idf/tree/master/examples/protocols/modbus
You just need to configure the simulator SW accordingly to work with parameters in these examples. I can provide the project files for ModbusPoll/Slave to read/write parameters of master/slave examples on ESP32.
Do you need examples on how to configure it?

MarkEvens, 2020-08-10:
finally solved i have hardware problem my max485 module is broken

Thank you for your help

alisitsyn, 2020-08-11:
@MarkEvens, ok, let me know if you have questions.

Alvin1Zhang, 2020-08-11:
Thanks for reporting and updates, feel free to reopen.

closed 2020-08-11 by Alvin1Zhang