espressif/esp-idf #5710: Cannot build with freeRTOS

jakubpav, 2020-08-08:
Hello, 
I've got a problem with building even a simple program.

So if I have code: 
`
#include <iostream>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"

extern "C"
{
    void app_main(void);
}

void app_main(void)
{
    for(;;)
    {
        std::cout << "HELLO" << std::endl;
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}
`
Then I run `idf.py build`.

And i still get a 
`FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj 
...
...
../main/main.cpp: In function 'void app_main()':
../main/main.cpp:15:9: error: 'vTaskDelay' was not declared in this scope
         vTaskDelay(1000 / portTICK_PERIOD_MS);
         ^~~~~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
`
error. 

My questions are: Do i need to set some extra paths? What i did wrong?

(all other source is the same as in hello-world example)

chegewara, 2020-08-08:
> (all other source is the same as in hello-world example)

I see some difference:
https://github.com/espressif/esp-idf/blob/master/examples/get-started/hello_world/main/hello_world_main.c#L10-L12

jakubpav, 2020-08-10:
oh, I see `#include "freertos/tasks.h"` :D sorry.

closed 2020-08-11 by jakubpav

projectgus, 2020-08-11:
Thanks @chegewara for helping out. :)