espressif/esp-idf #5713: Half-duplex SPI in DMA mode reads 0 on some positions

yaqwsx, 2020-08-10:
## Environment

- Development Kit: none
- Module or chip used: ESP32-WROVER
- IDF version: v4.1-rc
- Build System: CMake
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System: Linux
- Using an IDE?: no
- Power Supply: Battery

## Problem Description

There is a half-duplex SPI bus, where ESP32 is master and talks to custom hardware. The write operation works as expected, the read operation reads several bytes correctly, the other bytes it reads as zero. The data are physically present on the bus (verified by an oscilloscope). This happens only in the DMA mode. When I call `spi_bus_initialize` with the `dma_chan` 0, everything starts to work as expected. However, DMA is not used.

Also note, that the function `spi_device_transmit` writes to all bytes of the output buffer, but the values are zero. 

### Steps to reproduce

Unfortunately, the bug is present in a fairly large codebase involving custom hardware. I wasn't able to create a simple reproducible example using only devkits and commonly available hardware (yet).
 
### Code to reproduce this issue

Code to initialize the bus (the code uses custom C++ wrappers which are fairly simple and I can guarantee they work properly):

```cpp
spi_bus_config_t busConfig = SpiBus()
    .mosiIoNum( dataPin )
    .sclkIoNum( clkPin )
    .flags( SPICOMMON_BUSFLAG_MASTER );
spi_device_interface_config_t devConfig = SpiDeviceInterface()
    .mode( 0 )
    .flags( SPI_DEVICE_3WIRE | SPI_DEVICE_HALFDUPLEX )
    .queueSize( 1 )
    .clockSpeedHz( clkFrequency );
auto ret = spi_bus_initialize( bus, &busConfig, 1 ); // When we pass 0 as DMA channel, code starts to work
ESP_ERROR_CHECK( ret );
ret = spi_bus_add_device( bus, &devConfig, &_spiDev );
ESP_ERROR_CHECK( ret );
```

Reading from the bus:
```cpp
inline void spiRead( spi_device_handle_t dev, uint8_t* where, int count ) {
    spi_transaction_t t = {};
    t.rx_buffer = where;
    memset( where, 0xAA, count ); // To recognize whether spi_device_transmit wrote something into the buffer
    t.length = t.rxlength = 8 * count;
    spi_device_transmit( dev, &t );
	// When reading 14 bytes, the last byte is zero;
	// When reading 4 bytes, the last two bytes are zero
    std::cout << "SPI READ: " << hexDump( where, count ) << "\n";
    std::cout << "SPI READ: " << charDump( where, count ) << "\n";
}
```

bill1davis, 2020-08-13:
Possibly lending weight to this issue, we have a 128MB serial flash part on HSPI.  Works like a champ on idf 3.3, works like a champ in 4.1beta2 at 80 and 160Mhz system clock, but at 240Mhz returns the entire 4096 block as zeroes, or the entire block as 0xffs randomly maybe one in 30 reads or so.  Writes to the serial flash part work fine at 240Mhz.  Not sure why the system clock would matter as I understand spi runs off of APB clock, but it does.

@yaqwsx have you tried lowering system clock?

davidjade, 2020-09-18:
I am seeing similar problems. Using half duplex with DMA just the first byte is read then zeros for the rest. I pre-filled the buffer and it is actively changing it to zeros. I can see the SPI data with a logic analyzer and the data is on the lines, it is just not getting into the read buffer (except the first byte).