espressif/esp-idf #5714: Insufficient handling of Bluetooth GATT memory problems

itmm, 2020-08-10:
## Environment

- IDF version: v4.3-dev-771-gc77c4ccf6
- Build System: idf.py
- Compiler version: (crosstool-NG esp-2020r2) 8.2.0
- Operating System: Linux
- Using an IDE?: No

## Problem Description

in the bluetooth GATT implementation `components/bt/host/bluedroid/bta/gatt/bta_gatts_api.c` nothing happens, if a memory allocation fails. Other files may also have the same problem.

### Expected Behavior

generate an error response to the app or at least log an error

### Actual Behavior

nothing happens; from the application side the bluetooth stack froze as neither a success nor a failure message arrives

### Steps to reproduce

generate a BLE service with more that 500 characteristics

Alvin1Zhang, 2020-08-11:
Thanks for reporting, we will look into.

WCCWCC, 2020-08-11:
Hi， @itmm 

* Can you provide a test program?

* I will provide MR to fix this problem after I reproduce it.

* Similarly, I will also check whether the same problem exists elsewhere

itmm, 2020-08-11:
Hi,

a test program is available at

https://github.com/itmm/esp-btle-chars

You can configure the number of characteristics using menuconfig (entry BLE Characteristics). With the default of 200 you should get a completion message on the console. With 500 no additional event is logged an also no completion message is written.

Kind regards,
Timm.