espressif/esp-idf #5715: Full Unix-like time & timezone API support

huming2207, 2020-08-10:
**Is your feature request related to a problem? Please describe.**

ESP32 does not have full support on time and timezone support. For example, `tm_gmtoff` in `struct tm` is not available. Some `TZ` variables are also not available. This may be very problematic for some use cases like:

- e-calendar applications (e.g. parsing iCalendar files)
- some embedded JS interpreters like QuickJS (so that Date class may not work properly any more)
- some other inter-timezone conversions

**Describe the solution you'd like**

I understand that adding these features may dramatically increase the firmware size. But I think if possible, could you guys please leave an option in `Kconfig` so that can let the user decide whether to enable it or not, instead of hard-coded macros.

**Describe alternatives you've considered**

But meanwhile seems like ESP32 uses the time APIs from the internal ROM. Maybe we need to implement a separate time library/wrapper instead?

**Additional context**

N/A

Alvin1Zhang, 2020-08-11:
Thanks for raising this feature request.

KonstantinKondrashov, 2020-08-11:
Hi @huming2207!
Thanks for the feature request. We will add this to the future IDF versions.
I can suggest trying to use CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS option to disable ROM time functions, and after that, you can [build the toolchain yourself](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/toolchain-setup-scratch.html?#customized-setup-of-toolchain) with supporting TZ database.

huming2207, 2020-08-11:
Hello @KonstantinKondrashov 

Thanks! I will have a look for that!

huming2207, 2020-08-11:
Hello again @KonstantinKondrashov 

I just had a look on the menuconfig of that crosstool-ng. But I couldn't find any options for enabling the full timezone. Is it not available in the current version of newlib and I have to add it by myself?

KonstantinKondrashov, 2020-08-11:
Actually, the [CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS](https://github.com/espressif/esp-idf/blob/master/Kconfig#L78-L92) option is related to ESP-IDF. If you do not see this option in IDF it means that you can not apply this approach. Sorry, I am not familiar with crosstool-ng, and I do not know how to enable the full timezone there.

huming2207, 2020-08-12:
Hi @KonstantinKondrashov , 

Yea I also couldn't find a way of getting timezone database in in the crosstool-ng or newlib configure's args. 

But for the `tm_gmtoff`, seems like I just need to do something like this before including the `time.h`:

```c
#ifdef ESP_PLATFORM
#define __TM_GMTOFF tm_gmtoff
#endif
```

I guess it should work. But I haven't got time to test it.

crackwitz, 2023-04-12:
@huming2207 I just tested that (define before include) with esp-idf v5.0.1 and it had no effect for me. the definition needs to be visible in whatever library code implements `strptime`. I can't find the implementation by diving into the esp-idf code using VSCode.

diplfranzhoepfinger, 2024-05-30:
i support this too.

wuyuanyi135, 2025-11-17:
Is there any progress to this issue?