espressif/esp-idf #5716: esp_console error

aledoors, 2020-08-11:
Hi friends, i have the next errors:

In file included from C:/esp-idf/components/console/esp_console.h:22,
                 from ../main/i2ctools_example_main.c:14:
../main/i2ctools_example_main.c: In function 'app_main':
../main/i2ctools_example_main.c:44:18: error: implicit declaration of function 'esp_console_start_repl'; did you mean 'esp_console_split_argv'? [-Werror=implicit-function-declaration]
  ESP_ERROR_CHECK(esp_console_start_repl(repl));
                  ^~~~~~~~~~~~~~~~~~~~~~
C:/esp-idf/components/esp_common/include/esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
../main/i2ctools_example_main.c:44:41: error: 'repl' undeclared (first use in this function)
  ESP_ERROR_CHECK(esp_console_start_repl(repl));
                                         ^~~~
C:/esp-idf/components/esp_common/include/esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
../main/i2ctools_example_main.c:44:41: note: each undeclared identifier is reported only once for each function it appears in
  ESP_ERROR_CHECK(esp_console_start_repl(repl));
                                         ^~~~
C:/esp-idf/components/esp_common/include/esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
../main/i2ctools_example_main.c:46:5: error: unknown type name 'esp_console_repl_t'; did you mean 'esp_console_cmd_t'?
     esp_console_repl_t *repl = NULL;
     ^~~~~~~~~~~~~~~~~~
     esp_console_cmd_t
../main/i2ctools_example_main.c:47:5: error: unknown type name 'esp_console_repl_config_t'; did you mean 'esp_console_config_t'?
     esp_console_repl_config_t repl_config = ESP_CONSOLE_REPL_CONFIG_DEFAULT();
     ^~~~~~~~~~~~~~~~~~~~~~~~~
     esp_console_config_t
../main/i2ctools_example_main.c:47:45: error: implicit declaration of function 'ESP_CONSOLE_REPL_CONFIG_DEFAULT'; did you mean 'SDSPI_SLOT_CONFIG_DEFAULT'? [-Werror=implicit-function-declaration]
     esp_console_repl_config_t repl_config = ESP_CONSOLE_REPL_CONFIG_DEFAULT();
                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                             SDSPI_SLOT_CONFIG_DEFAULT
../main/i2ctools_example_main.c:48:5: error: unknown type name 'esp_console_dev_uart_config_t'; did you mean 'esp_console_config_t'?
     esp_console_dev_uart_config_t uart_config = ESP_CONSOLE_DEV_UART_CONFIG_DEFAULT();
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     esp_console_config_t
../main/i2ctools_example_main.c:48:49: error: implicit declaration of function 'ESP_CONSOLE_DEV_UART_CONFIG_DEFAULT'; did you mean 'SDSPI_SLOT_CONFIG_DEFAULT'? [-Werror=implicit-function-declaration]
     esp_console_dev_uart_config_t uart_config = ESP_CONSOLE_DEV_UART_CONFIG_DEFAULT();
                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                 SDSPI_SLOT_CONFIG_DEFAULT
../main/i2ctools_example_main.c:53:16: error: request for member 'prompt' in something not a structure or union
     repl_config.prompt = "i2c-tools>";
                ^
In file included from C:/esp-idf/components/console/esp_console.h:22,
                 from ../main/i2ctools_example_main.c:14:
../main/i2ctools_example_main.c:55:21: error: implicit declaration of function 'esp_console_new_repl_uart'; did you mean 'esp_console_cmd_register'? [-Werror=implicit-function-declaration]
     ESP_ERROR_CHECK(esp_console_new_repl_uart(&uart_config, &repl_config, &repl));
                     ^~~~~~~~~~~~~~~~~~~~~~~~~
C:/esp-idf/components/esp_common/include/esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
At top level:
../main/i2ctools_example_main.c:19:20: warning: 'TAG' defined but not used [-Wunused-variable]
 static const char *TAG = "i2c-tools";
                    ^~~
cc1.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.
Build complete (9 errors, 1 warnings): D:\Eclipse\Trabajo\I2C_Tools\build

suda-morris, 2020-08-11:
@aledoors Have you update the esp-idf to latest master/release/v4.2? The change was introduced in this commit: 98d7f13c24089a090e156b9820b3e2b3d0a3a8bb

aledoors, 2020-08-11:
@suda-morris maybe not... almost certainly not. I will do the update and then comment to you. Thanks in advance.

aledoors, 2020-08-12:
@suda-morris same error. Nothing change

suda-morris, 2020-08-12:
Hi @aledoors  I checked the i2ctool example on master and release/v4.2, no build error. Maybe the IDF_PATH doesn't point to the right one in your eclipse setting?

aledoors, 2020-08-12:
Hello @suda-morris i tried from the espressif IDE and I have the same error that appears in Eclipse. Maybe I'm doing wrong the update, but i followed the steps as the procedure says.

suda-morris, 2020-08-13:
Could you go into your esp-idf repo, build the example in your command line? To see if you have installed the right version of esp-idf.

aledoors, 2020-08-17:
Hi Suda, sorry for the delay in answering you, below is the error

C:\esp-idf\examples\peripherals\i2c\i2c_tools>idf.py build
Checking Python dependencies...
Python requirements from C:\esp-idf\requirements.txt are satisfied.
Executing action: all (aliases: build)
Running ninja in directory c:\esp-idf\examples\peripherals\i2c\i2c_tools\build
Executing "ninja all"...
[1/10] cmd.exe /C "cd /D C:\esp-idf\examples\peripherals\i...*********************************************************"
Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
storage,data,fat,0x110000,1M,
*******************************************************************************
[2/10] Performing build step for 'bootloader'
ninja: no work to do.
[3/8] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/i2ctools_example_main.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/i2ctools_example_main.c.obj
ccache C:\Users\C765B~1.ALE\ESPRES~1\tools\XTENSA~1\ESP-20~1.0\XTENSA~1\bin\XT8994~1.EXE -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../main -I../../../../../components/newlib/platform_include -I../../../../../components/freertos/include -I../../../../../components/heap/include -I../../../../../components/log/include -I../../../../../components/soc/esp32/include -I../../../../../components/soc/include -I../../../../../components/esp_rom/include -I../../../../../components/esp_common/include -I../../../../../components/xtensa/include -I../../../../../components/xtensa/esp32/include -I../../../../../components/esp32/include -I../../../../../components/driver/include -I../../../../../components/esp_ringbuf/include -I../../../../../components/esp_event/include -I../../../../../components/tcpip_adapter/include -I../../../../../components/lwip/include/apps -I../../../../../components/lwip/include/apps/sntp -I../../../../../components/lwip/lwip/src/include -I../../../../../components/lwip/port/esp32/include -I../../../../../components/lwip/port/esp32/include/arch -I../../../../../components/vfs/include -I../../../../../components/esp_wifi/include -I../../../../../components/esp_wifi/esp32/include -I../../../../../components/esp_eth/include -I../../../../../components/efuse/include -I../../../../../components/efuse/esp32/include -I../../../../../components/app_trace/include -I../../../../../components/mbedtls/port/include -I../../../../../components/mbedtls/mbedtls/include -I../../../../../components/wpa_supplicant/include -I../../../../../components/wpa_supplicant/port/include -I../../../../../components/wpa_supplicant/include/esp_supplicant -I../../../../../components/bootloader_support/include -I../../../../../components/app_update/include -I../../../../../components/spi_flash/include -I../../../../../components/nvs_flash/include -I../../../../../components/pthread/include -I../../../../../components/espcoredump/include -I../../../../../components/asio/asio/asio/include -I../../../../../components/asio/port/include -I../../../../../components/coap/port/include -I../../../../../components/coap/port/include/coap -I../../../../../components/coap/libcoap/include -I../../../../../components/coap/libcoap/include/coap2 -I../../../../../components/console -I../../../../../components/nghttp/port/include -I../../../../../components/nghttp/nghttp2/lib/includes -I../../../../../components/esp-tls -I../../../../../components/esp_adc_cal/include -I../../../../../components/esp_gdbstub/include -I../../../../../components/tcp_transport/include -I../../../../../components/esp_http_client/include -I../../../../../components/esp_http_server/include -I../../../../../components/esp_https_ota/include -I../../../../../components/protobuf-c/protobuf-c -I../../../../../components/protocomm/include/common -I../../../../../components/protocomm/include/security -I../../../../../components/protocomm/include/transports -I../../../../../components/mdns/include -I../../../../../components/esp_local_ctrl/include -I../../../../../components/esp_websocket_client/include -I../../../../../components/expat/expat/expat/lib -I../../../../../components/expat/port/include -I../../../../../components/wear_levelling/include -I../../../../../components/sdmmc/include -I../../../../../components/fatfs/diskio -I../../../../../components/fatfs/vfs -I../../../../../components/fatfs/src -I../../../../../components/freemodbus/common/include -I../../../../../components/idf_test/include -I../../../../../components/jsmn/include -I../../../../../components/json/cJSON -I../../../../../components/libsodium/libsodium/src/libsodium/include -I../../../../../components/libsodium/port_include -I../../../../../components/mqtt/esp-mqtt/include -I../../../../../components/openssl/include -I../../../../../components/spiffs/include -I../../../../../components/ulp/include -I../../../../../components/unity/include -I../../../../../components/unity/unity/src -I../../../../../components/wifi_provisioning/include -I../../../../system/console/components/cmd_nvs -I../../../../system/console/components/cmd_system -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unus [... cut ...]
../main/i2ctools_example_main.c:70:21: error: implicit declaration of function 'esp_console_start_repl'; did you mean 'esp_console_split_argv'? [-Werror=implicit-function-declaration]
     ESP_ERROR_CHECK(esp_console_start_repl(repl));
                     ^~~~~~~~~~~~~~~~~~~~~~
c:\esp-idf\components\esp_common\include\esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
cc1.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\esp-idf\examples\peripherals\i2c\i2c_tools>

suda-morris, 2020-08-17:
Seen from your log ` -DIDF_VER="v4.0.1-dirty" `
But esp_console_repl was introduced from esp-idf v4.2
So please upgrade your ESP-IDF repo correctly.

aledoors, 2020-08-17:
Hi Suda, I understand that I am not doing the update steps correctly, but I don't know what I am doing wrong?
I did de procedure below

Updating to Master Branch
Note

Using Master branch means living “on the bleeding edge” with the latest ESP-IDF code.

To use the latest version on the ESP-IDF master branch, this is the process to follow:

Check out the master branch locally:

cd $IDF_PATH
git checkout master
git pull
git submodule update --init --recursive

suda-morris, 2020-08-17:
The steps are correct, but I'm afraid you have multiple IDF_PATH environment, can you check if the IDF_PATH points to the repo you run the i2ctool example in?

aledoors, 2020-08-18:
I did the procedure below:

Add IDF_PATH to User Profile
[中文]

To preserve setting of IDF_PATH environment variable between system restarts, add it to the user profile, following instructions below.

Windows
The user profile scripts are contained in C:/msys32/etc/profile.d/ directory. They are executed every time you open an MSYS2 window.

Create a new script file in C:/msys32/etc/profile.d/ directory. Name it export_idf_path.sh.

Identify the path to ESP-IDF directory. It is specific to your system configuration and may look something like C:\msys32\home\user-name\esp\esp-idf

Add the export command to the script file, e.g.:

export IDF_PATH="C:/msys32/home/user-name/esp/esp-idf"
Remember to replace back-slashes with forward-slashes in the original Windows path.

Save the script file.

Close MSYS2 window and open it again. Check if IDF_PATH is set, by typing:

printenv IDF_PATH
The path previusly entered in the script file should be printed out.

If you do not like to have IDF_PATH set up permanently in user profile, you should enter it manually on opening of an MSYS2 window:

export IDF_PATH="C:/msys32/home/user-name/esp/esp-idf"
If you got here from section Setup Path to ESP-IDF, while installing s/w for ESP32 development, then go back to section Start a Project.

suda-morris, 2020-08-18:
So? What you get if you run `printenv IDF_PATH` just in your example folder?

aledoors, 2020-08-19:
I get this: "printenv" is not recognized as an internal or external command, executable program or batch file.

aledoors, 2020-08-26:
@suda-morris I found the problem and resolved it. what i did was reinstall everything again and that way it worked. Thanks for your help and time.

closed 2020-08-31 by Alvin1Zhang