espressif/esp-idf #5717: Unable to get WIFI_EVENT_STA_DISCONNECTED event after router turned OFF.

turgu1, 2020-08-11:
## Environment

- Development Kit:      None
- Module or chip used:  ESP32-WROOM-32 (DoIt ESP32 DEVKIT v1)
- IDF version (run ``git describe --tags`` to find it): v4.0.1
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

This is a WiFi related issue. The ESP32 received its IP address and mask from the router and is able to communicate. Forcing the router to no power, the wifi driver doesn't indicate the issue to the application.

### Expected Behavior

The event handler is expected to receive a WIFI_EVENT_STA_DISCONNECTED event.

### Actual Behavior

No event is received.

### Steps to reproduce

1. Using the station example located in esp-idf/examples/wifi/getting_started/station without any change.
2. Configure with idf.py menuconfig, set wifi ssid and password in example configuration
3. Router is ON, board is connected to USB port
4. Compile and load using the following command:  idf.py -p /dev/ttyUSB0 flash monitor

### Code to reproduce this issue

The code used to reproduce is the station example from esp-idf/examples/wifi/getting_started/station

## Debug Logs

```
$ idf.py -p /dev/ttyUSB0 flash monitor
Checking Python dependencies...
Python requirements from /home/turgu1/esp/esp-idf/requirements.txt are satisfied.
Adding flash's dependency "all" to list of actions
Executing action: all (aliases: build)
Running ninja in directory /home/turgu1/Dev/station/build
Executing "ninja all"...
[1/4] cd /home/turgu1/Dev/station/build/esp-idf/partition_table && /usr/bin/cmake -E echo "Parti.../cmake -E echo "*******************************************************************************"
Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[2/4] Performing build step for 'bootloader'
ninja: no work to do.
Executing action: flash
Running esptool.py in directory /home/turgu1/Dev/station/build
Executing "/home/turgu1/.espressif/python_env/idf4.0_py3.8_env/bin/python /home/turgu1/esp/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset write_flash @flash_project_args"...
esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 wifi_station.bin
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting......
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 0)
Features: WiFi, BT, Dual Core, Coding Scheme None
Crystal is 40MHz
MAC: 30:ae:a4:05:0e:34
Uploading stub...
Running stub...
Stub running...
[... 103 lines cut ...]
I (852) wifi:state: init -> auth (b0)
I (852) wifi:state: auth -> assoc (0)
I (862) wifi:state: assoc -> run (10)
I (1912) wifi:connected with toto, aid = 1, channel 6, BW20, bssid = 00:21:29:68:41:64
I (1922) wifi:security: WPA2-PSK, phy: bg, rssi: -54
I (1922) wifi:pm start, type: 1

I (1972) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (5042) tcpip_adapter: sta ip: 10.1.1.100, mask: 255.255.255.0, gw: 10.1.1.1
I (5042) wifi station: got ip:10.1.1.100
I (5042) wifi station: connected to ap SSID:toto password:snoopy1967
I (24462) wifi:bcn_timout,ap_probe_send_start
I (26962) wifi:ap_probe_send over, resett wifi status to disassoc
I (26962) wifi:state: run -> init (c800)
I (26962) wifi:pm stop, total sleep time: 15241907 us / 25040626 us

I (26972) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1

```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2020-08-12:
Thanks for reporting and letting us know, we will look into.

zhangyanjiaoesp, 2020-08-12:
hi @turgu1 ,
According to the log `state: run -> init (c800)`, the STA has issued the disconnect event, but the application doesn't report the disconnect event.  The reason is as the following: 
In the station example, after the STA has connected to AP successfully, the event handler will be deleted, so the when you power off the AP, the disconnect event doesn't show.
https://github.com/espressif/esp-idf/blob/820621687ce04b3c85276ca0eb3fe72479573cfd/examples/wifi/getting_started/station/main/station_example_main.c#L123

If you want the example show the disconnect event, you can delete line 122~124 in the example.

turgu1, 2020-08-12:
Well, I need a new pair of glasses. Thank you!

closed 2020-08-12 by turgu1