espressif/esp-idf #5718: Generating an RSA key pair triggers the task watchdog

ferrarimarco, 2020-08-11:
## Environment

- Development Kit: N/A
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
    v4.3-dev-771-gc77c4ccf6
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     macOS
- Using an IDE?: No
- Power Supply: USB and external

## Problem Description

I'm generating an 4096-bit RSA private key, but the watchdog complains that the task didn't reset it in time.

### Expected Behavior

The watchdog should be correctly reset during key generation

### Actual Behavior

The watchdog is not reset in time, so it panics.

### Steps to reproduce

1. Initialize mbedTLS context to generate RSA keys (see below)
2. Call the `mbedtls_rsa_gen_key` function.

### Code to reproduce this issue

```c
    int ret = 1;
    int exit_code = MBEDTLS_EXIT_FAILURE;
    mbedtls_pk_context key;
    mbedtls_mpi N, P, Q, D, E, DP, DQ, QP;
    mbedtls_entropy_context entropy;

    ESP_LOGI(TAG, "Initializing contexts...");

    mbedtls_mpi_init(&N);
    mbedtls_mpi_init(&P);
    mbedtls_mpi_init(&Q);
    mbedtls_mpi_init(&D);
    mbedtls_mpi_init(&E);
    mbedtls_mpi_init(&DP);
    mbedtls_mpi_init(&DQ);
    mbedtls_mpi_init(&QP);

    mbedtls_pk_init(&key);

    mbedtls_ctr_drbg_context ctr_drbg;
    mbedtls_ctr_drbg_init(&ctr_drbg);

    char buf[1024];
    memset(buf, 0, sizeof(buf));

    ESP_LOGI(TAG, "Seeding the random number generator...");
    mbedtls_entropy_init(&entropy);
    if ((ret = mbedtls_ctr_drbg_seed(&ctr_drbg, mbedtls_entropy_func, &entropy, NULL, 0)) != 0)
    {
        ESP_LOGE(TAG, "mbedtls_ctr_drbg_seed returned %d", ret);
        return -1;
    }

    ESP_LOGI(TAG, "Initializing key context...");
    if ((ret = mbedtls_pk_setup(&key, mbedtls_pk_info_from_type(MBEDTLS_PK_RSA))) != 0)
    {
        ESP_LOGE(TAG, "mbedtls_pk_info_from_type returned %d", ret);
        return -1;
    }

    ESP_LOGI(TAG, "Generating the private key...");
    ret = mbedtls_rsa_gen_key(mbedtls_pk_rsa(key), mbedtls_ctr_drbg_random, &ctr_drbg, rsa_key_generation_options.key_size, 65537);
    if (ret != 0)
    {
        ESP_LOGE(TAG, "mbedtls_rsa_gen_key returned %d", ret);
        return -1;
    }

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
E (9288) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (9288) task_wdt:  - IDLE1 (CPU 1)
E (9288) task_wdt: Tasks currently running:
E (9288) task_wdt: CPU 0: IDLE0
E (9288) task_wdt: CPU 1: rsa_key_gen

.....

Backtrace:0x4008ACB7:0x3FFBEC70 0x40082E35:0x3FFBEC90 0x4017ADB1:0x3FFCB7E0 0x40128FB4:0x3FFCB800 0x4012A435:0x3FFCB830 0x4012A827:0x3FFCBEA0 0x4012ABD9:0x3FFCBF10 0x4012AD35:0x3FFCBF40 0x401339AA:0x3FFCBF80 0x400D8789:0x3FFCBFE0 0x400D8141:0x3FFCC840 0x4008FB95:0x3FFCC860
0x4008acb7: esp_crosscore_isr at /Users/ferrarim/Workspaces/esp-idf/components/esp32/crosscore_int.c:77

0x40082e35: _xt_lowint1 at Workspaces/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x4017adb1: mpi_mul_hlp at Workspaces/esp-idf/components/mbedtls/mbedtls/library/bignum.c:1529 (discriminator 2)

0x40128fb4: mpi_montmul at Workspaces/esp-idf/components/mbedtls/mbedtls/library/bignum.c:2004 (discriminator 3)

0x4012a435: mbedtls_mpi_exp_mod at Workspaces/esp-idf/components/mbedtls/mbedtls/library/bignum.c:2197

0x4012a827: mpi_miller_rabin at Workspaces/esp-idf/components/mbedtls/mbedtls/library/bignum.c:2570

0x4012abd9: mbedtls_mpi_is_prime_ext at Workspaces/esp-idf/components/mbedtls/mbedtls/library/bignum.c:2641

0x4012ad35: mbedtls_mpi_gen_prime at Workspaces/esp-idf/components/mbedtls/mbedtls/library/bignum.c:2731

0x401339aa: mbedtls_rsa_gen_key at Workspaces/esp-idf/components/mbedtls/mbedtls/library/rsa.c:597

0x400d8789: generate_rsa_keypair at Workspaces/xxxxxx/build/../components/crypto_utils/rsa_utils.c:96 (discriminator 9)

0x400d8141: vRsaKeyGenerationTask at Workspaces/xxx/build/../main/smart_desk_main.c:74

0x4008fb95: vPortTaskWrapper at Workspaces/esp-idf/components/freertos/xtensa/port.c:169

```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder) - see below
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

### sdkconfig.defaults

```ini
# Store the current time/date timestamp in the app description structure
CONFIG_APP_COMPILE_TIME_DATE=y

# Enable Bluetooth and show the submenu with Bluetooth configuration choices.
CONFIG_BT_ENABLED=y
CONFIG_BT_NIMBLE_ENABLED=y

# Detect flash size when flashing bootloader
CONFIG_ESPTOOLPY_FLASHSIZE_DETECT=y

# Set flash size to 4MB
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y

# Configure the partition scheme to hold a factory app, and two OTA definitions
CONFIG_PARTITION_TABLE_TWO_OTA=y

# Enable FreeRTOS tracing and formatting functions
CONFIG_FREERTOS_USE_TRACE_FACILITY=y
CONFIG_FREERTOS_USE_STATS_FORMATTING_FUNCTIONS=y

# Enable hardware accelerated multiple precision integer operations, AES, and SHA
CONFIG_MBEDTLS_HARDWARE_AES=y
CONFIG_MBEDTLS_HARDWARE_MPI=y
CONFIG_MBEDTLS_HARDWARE_SHA=y
```

ferrarimarco, 2020-08-11:
Note, I'm creating a task to run on CPU1, but the very same situation applies even if I run everything on CPU0.

ferrarimarco, 2020-08-11:
Is changing the value of CONFIG_MBEDTLS_HARDWARE_MPI  enough to turn on/off MPI hardware acceleration? Enabling or disabling that flag made no difference in my case.

Alvin1Zhang, 2020-08-12:
Thanks for reporting and letting us know, we will look into.

projectgus, 2020-08-12:
Hi @ferrarimarco,

Thanks for the detailed report of your config and what you're seeing.

The purpose of the task watchdog is to provide a warning if a task is monopolising the CPU without yielding to lower priority tasks. In the default configuration, if triggers if the idle task (lowest priority task) on either CPU is not run within the timeout period. There is a potential conflict of purpose here when the firmware actually needs to calculate something which takes a lot of CPU power (generating an RSA key, even with MPI hardware acceleration, takes a lot of CPU power on the ESP32.)

There are a few different ways you can work around this:

1. Increase the [default task timeout timeout period](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-esp-task-wdt-timeout-s) to the length of the longest intensive calculation you expect to do (probably generating a 4096-bit RSA key), plus a small amount of headroom.
2. Immediately before the RSA key generation starts, set the task WDT timeout higher by calling [esp_task_wdt_init()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html#_CPPv417esp_task_wdt_init8uint32_tb) with a higher timeout value. Then call `esp_task_wdt_init()` again after key generation is complete to drop it back down. (Note that you can call `esp_task_wdt_init()` multiple times without disabling the WDT in between, each call will reconfigure it with the new values.)
2. [Disable the task watchdog timer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-esp-task-wdt) if you don't think you need it. It's designed as a programming aid to catch "stuck" tasks but it's not mandatory, especially if you are comfortable programming in an RTOS environment.
3. [Don't enable the task watchdog for the idle task on the CPU that you are using to generate the RSA key](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-esp-task-wdt-check-idle-task-cpu0). It will still be active from startup on the other CPU.
4. If key generation happens during a startup sequence, set the config item to not enable the task watchdog on that CPU's idle task by default but then enable it from your app later, after key generation is completed. This is done by calling `esp_task_wdt_add(xTaskGetIdleTaskHandle()))` or `esp_task_wdt_add(xTaskGetIdleTaskHandleForCPU(cpu_num))`.

(For more background on the runtime options for Task WDT, see the [API reference page](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html#task-watchdog-timer).

> The watchdog is not reset in time, so it panics.

Just to check, the expected behaviour by default would be that the WDT triggers and prints a warning but the task continues running. "Panic" on Task WDT timeout (as in, resetting the system) is an option but it is not enabled by default. Is that consistent with what you're seeing?

> Is changing the value of CONFIG_MBEDTLS_HARDWARE_MPI enough to turn on/off MPI hardware acceleration? Enabling or disabling that flag made no difference in my case.

Yes, that's correct. The calculations used to generate an RSA key are substantial even with accelerated MPI calculations, and the cost increases exponentially with the key size. (Even on my laptop here with a current generation Core i7, running `openssl genrsa 4096` takes 0.8s).

ferrarimarco, 2020-08-12:
Thanks for the detailed reply!

As a workaround, I already implement your suggestion no.5, and I'm happy with that.

Wouldn't it be possible to add some sleeps during key generation, to make the watchdog happy? A couple hundred milliseconds here and there shouldn't be to much, considering that key generation takes O(seconds) on an ESP32@240 MHz.

Thanks!

negativekelvin, 2020-08-13:
>Wouldn't it be possible to add some sleeps during key generation
Not practical. If a single mbedtls function call takes too long then alternate versions of those functions with sleeps would have to be created and maintained and may still only work within specific configurations.

projectgus, 2020-08-13:
> Wouldn't it be possible to add some sleeps during key generation, to make the watchdog happy? A couple hundred milliseconds here and there shouldn't be to much, considering that key generation takes O(seconds) on an ESP32@240 MHz.

In addition to what @negativekelvin says, we can't know whether a particular firmware prefers to have the operation complete as quickly as possible or to have the task stop and go to sleep in case something else wants to run. It's a perfectly legitimate use of an RTOS to have a high priority task monopolise the CPU if that's the most important thing in the system. Automatically adding sleeps works actively against the user in this case.

The task watchdog is intended as a coarse check for the common beginner pitfall of porting code to an RTOS and forgetting that you have to let other tasks run sometimes[*], but there's no "one size fits all" way to configure it.

[*] This was a particular concern when ESP32 was new and we knew a lot of our users were coming from ESP8266 with non-OS SDK or Arduino.

closed 2020-08-26 by projectgus

m1dnight, 2023-07-09:
I too have this issue, and I chose to try out option 2. A small change since that comment is that you cannot call the configure function multiple times.  At the time of writing you need to use `esp_task_wdt_reconfigure` if the WDT has been configured before. 

Other than that, the suggestion works fine.  Here is what I did: 

```
void generate_keys_task(void *params)
{
    ESP_LOGI(TAG, "generating pki");

    // this task will take longer than the default 5 seconds
    // increasing the wdt timeout to 10 seconds avoids it being triggered
    esp_task_wdt_config_t twdt_config = {
        .timeout_ms = 1000000,
        .idle_core_mask = (1 << portNUM_PROCESSORS) - 1, // Bitmask of all cores
        .trigger_panic = false,
    };
    ESP_ERROR_CHECK(esp_task_wdt_reconfigure(&twdt_config));

    initialize_keys();

    ESP_LOGI(TAG, "pki generated");

    xSemaphoreGive(binSemaphore);

    // set the watchdog timer back to its original settings.
    twdt_config.timeout_ms = CONFIG_ESP_INT_WDT_TIMEOUT_MS;
    ESP_ERROR_CHECK(esp_task_wdt_reconfigure(&twdt_config));

    vTaskDelete(NULL);
}