espressif/esp-idf #5719: esp_now_fetch_peer  function suspected issue

sagi2313, 2020-08-12:
INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB]
- IDF version (run ``git describe --tags`` to find it):
   v4.3-dev-771-gc77c4ccf6
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     [Linux]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No and Yes ] 
Using eclipse but also do my compilation and tests on shell
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description
Unless there is something wrong with my code that I missed, here it is:
I have tried to retrieve all known peers from esp_now_peer list, by using function 'esp_now_fetch_peer()'
As I saw that not all peers came up, I wrote a small test program to help understand the issue.
It seems that the esp_now_get_peer_num() function returns the correct number of peers stored in the list, but the 
loop with esp_now_fetch_peer() only returns the odd members(1,3,5 ..).

### Expected Behavior
I expect that
1.  esp_now_fetch_peer returns as many peers as  esp_now_get_peer_num says
2. Both are working correct, so they actually return all peers I  added
### Actual Behavior
Already described... esp_now_fetch_peer fetches every other entry, not all
### Steps to reproduce
1. add 10 peers in peer list using esp_now_add_peer()
2. get Number of peers using esp_now_get_peer_num()
3.  Using esp_now_fetch_peer() loop and print all peers found 

// If possible, attach a picture of your setup/wiring here.
No special wiring and setup. also its a software bug, a  broken iterator cant relate to wires

### Code to reproduce this issue

https://gist.github.com/sagi2313/6c8bf49e99e63382e142cdf15adc208d

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2020-08-13:
Thanks for reporting,  we will look into.

zhangyanjiaoesp, 2020-08-14:
hi @sagi2313 ,

The `esp_now_fetch_peer()` only return the peer which address is unicast, for the multicat/broadcast address, the function will ignore and try do find the next in the peer list.

sagi2313, 2020-08-25:
Thank you for your answer. 
I don't see any benefit in this behavior, it is only more complex , but at  least I think this would be best clearly documented in the relevant API docs. 
Br,
Alex

zhangyanjiaoesp, 2020-08-25:
We will update the API docs as soon as possible.

closed 2020-12-13 by espressif-bot