espressif/esp-idf #5720: https://dl.espressif.com/dl/xtensa-esp32-elf-osx-1.22.0-96-g2852398-5.2.0.tar.gz 404 not found

btukacs, 2020-08-12:
please correct the link: https://dl.espressif.com/dl/xtensa-esp32-elf-osx-1.22.0-96-g2852398-5.2.0.tar.gz

Alvin1Zhang, 2020-08-13:
Thanks for reporting, we will check, and thanks for the notes in https://github.com/espressif/esp-idf/issues/279

projectgus, 2020-08-13:
Hi @btukacs,

Yes, thanks for letting us know. We renamed the toolchain tuple to "macos" but the URL wasn't updated correctly when we updated v3.3. It should be https://dl.espressif.com/dl/xtensa-esp32-elf-macos-1.22.0-96-g2852398-5.2.0.tar.gz

I've updated the old URL so that they will both work for now. This issue will be closed once we update the URL in the docs.

Angus

Alvin1Zhang, 2020-11-05:
Thanks for reporting and sorry for slow turnaround, the fix has been available at https://github.com/espressif/esp-idf/commit/10d40d1448dd9d1cb1b71a26e843bb2f8237471b, feel free to reopen.

closed 2020-11-05 by Alvin1Zhang