espressif/esp-idf #5721: smartconfig does not respect log default level

AxelLin, 2020-08-12:
## Environment

- Module or chip used: ESP32-WROOM-32D
- IDF version: v4.3-dev-771-gc77c4ccf6c43
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

Even though I already set CONFIG_LOG_DEFAULT_LEVEL_NONE=y, I still got below smartconfig messages:

I (10340) smartconfig: SC version: V2.9.0
I (12998) smartconfig: Start to find channel...
I (13962) smartconfig: TYPE: ESPTOUCH
I (13963) smartconfig: T|AP MAC: xx:xx:xx:xx:xx:xx
I (13963) smartconfig: Found channel on 4-0. Start to get ssid and password...
I (18755) smartconfig: T|pswd: xxxxxxxxxx
I (18755) smartconfig: T|ssid: TEST-AP

### Expected Behavior
With CONFIG_LOG_DEFAULT_LEVEL_NONE=y, it should not print smartconfig related messages.

### Actual Behavior
It shows above messages.

### Steps to reproduce

Test using smart_config config.
idf.py menuconfig:
Log output: select Default log verbosity (No output).

Then run the example code, you will see the smartconfig messages.

Alvin1Zhang, 2020-08-13:
Thanks for reporting, we will look into.

zhangyanjiaoesp, 2020-08-14:
hi @AxelLin,
I have tested on the smartconfig example, the smartconfig log can be printed even I set the CONFIG_LOG_DEFAULT_LEVEL_NONE=y. The reason is that the smartconfig loga are independent of wifi logs.
To solve this problem, you can call '**esp_log_level_set("smartconfig", CONFIG_LOG_DEFAULT_LEVEL);**' after this line:
https://github.com/espressif/esp-idf/blob/c77c4ccf6c43ab09fd89e7c907bf5cf2a3499e3b/components/esp_wifi/src/wifi_init.c#L63

And I have tested with this change, it can close the logs.

AxelLin, 2020-08-14:
@zhangyanjiaoesp 
So you just point out this bug is *valid*.
All other libraries do not need to explicitly call esp_log_level_set for CONFIG_LOG_DEFAULT_LEVEL, only smartconfig needs to explicitly call it.

zhangyanjiaoesp, 2020-08-14:
> @zhangyanjiaoesp
> So you just point out this bug is _valid_.
> All other libraries do not need to explicitly call esp_log_level_set for CONFIG_LOG_DEFAULT_LEVEL, only smartconfig needs to explicitly call it.

Explicitly call `esp_log_level_set` is a workaround for your problem. For other libraries, we will check the logs, thanks.

closed 2020-09-16 by espressif-bot