espressif/esp-idf #5722: IDF4.2 : PPPOS_Client : issue sim800_init() : issue in the initialisation logic

mmggbe, 2020-08-12:
Dear,

I have an issue with the modem initialisation in the example PPPOS_Client

Api version is : ESP-IDF v4.2-dev-1995-g636b964c8-dirty
Modem :  SIM800l on TTGO board

When powering the SIM800L Modem, the Modem generates a glitch (some random characters) that should be managed by the UART.
The problem is : 

> if you  turn the power of the SIM800 before the sim800_init function(), there is no UART handler installed and it generates sometimes a crash

> If you turn the modem after the function sim800_init function(), the init function crashes since the modem is not responding (the function esp_modem_dce_sync() send an AT to the modem and expects an OK)

Below the original code

```
modem_dce_t *sim800_init(modem_dte_t *dte)
{
    DCE_CHECK(dte, "DCE should bind with a DTE", err);
    /* malloc memory for sim800_dce object */
    sim800_modem_dce_t *sim800_dce = calloc(1, sizeof(sim800_modem_dce_t));
    DCE_CHECK(sim800_dce, "calloc sim800_dce failed", err);
    /* Bind DTE with DCE */
    sim800_dce->parent.dte = dte;
    dte->dce = &(sim800_dce->parent);
    /* Bind methods */
    sim800_dce->parent.handle_line = NULL;
    sim800_dce->parent.sync = esp_modem_dce_sync;
    sim800_dce->parent.echo_mode = esp_modem_dce_echo;
...
    sim800_dce->parent.deinit = sim800_deinit;
    /* Sync between DTE and DCE */
    **DCE_CHECK(esp_modem_dce_sync(&(sim800_dce->parent)) == ESP_OK, "sync failed", err_io);**
...
}
```
I think the init function should be reworked as below

1° create a new handler to receive the Modem hardware init glitch
2° add a piece of code for the modem power ON as below

```
modem_dce_t *sim800_init(modem_dte_t *dte
{
....

    sim800_dce->parent.power_down = sim800_power_down;
    sim800_dce->parent.deinit = sim800_deinit;

   /* To be added */
    dce->handle_line = glitch_handle_function_to_be_written;
    modem_hardware_power_on_sequence();

   /* Sync between DTE and DCE */
    DCE_CHECK(esp_modem_dce_sync(&(sim800_dce->parent)) == ESP_OK, "sync failed", err_io);
    /* Close echo */
    DCE_CHECK(esp_modem_dce_echo(&(sim800_dce->parent), false) == ESP_OK, "close echo mode failed", err_io);
...
}
```

mmggbe, 2020-08-12:
Follwing code seemes to fix the issue, but to be deeply tested and cleaned.
Pls add in sim800l.c

`#include "driver/gpio.h"				//MaGe
`
....
```
**
 * @power ON sequence : handler that discards the glitches
 */
static esp_err_t sim800_handle_power_ON(modem_dce_t *dce, const char *line)		//MaGe
{
    ESP_LOGI(DCE_TAG, "Glitch from Modem: %s", line);
    return ESP_OK;
}
```
....
```
static esp_err_t sim800_power_ON(sim800_modem_dce_t *sim800_dce)				//MaGe
{
	modem_dte_t *dte = sim800_dce->parent.dte;
    sim800_dce->parent.handle_line = sim800_handle_power_ON;

	#define LOW 				0
	#define HIGH				1

	#define MODEM_RST            5
	#define MODEM_PWKEY          4
	#define MODEM_POWER_ON       23

	gpio_pad_select_gpio(MODEM_RST);
	gpio_set_direction(MODEM_RST, GPIO_MODE_OUTPUT);

	gpio_pad_select_gpio(MODEM_PWKEY);
	gpio_set_direction(MODEM_PWKEY, GPIO_MODE_OUTPUT);

	gpio_pad_select_gpio(MODEM_POWER_ON);
	gpio_set_direction(MODEM_POWER_ON, GPIO_MODE_OUTPUT);

	// restart module : needed in case of watchdog
	gpio_set_level(MODEM_RST, LOW);
	gpio_set_level(MODEM_PWKEY, LOW);
	gpio_set_level(MODEM_POWER_ON, LOW);		// Turn off the Modem LDO external power supply
	ESP_LOGI(DCE_TAG, "Module GSM un-powered");
	vTaskDelay(500/portTICK_PERIOD_MS);			// must be between 0,5 and 1s

	gpio_set_level(MODEM_RST, HIGH);
	gpio_set_level(MODEM_PWKEY, HIGH);
	gpio_set_level(MODEM_POWER_ON, HIGH);		// Turn on the Modem LDO external power supply first
	ESP_LOGI(DCE_TAG, "Module GSM powered");
	vTaskDelay(500/portTICK_PERIOD_MS);			// must be 1,5s up before the raise of PWKEY

	gpio_set_level(MODEM_PWKEY, LOW);
	ESP_LOGI(DCE_TAG, "Module GSM PWKEY LOW");
	vTaskDelay(1100/portTICK_PERIOD_MS);		// Pull down PWRKEY for more than 1 second

	gpio_set_level(MODEM_PWKEY, HIGH);
	ESP_LOGI(DCE_TAG, "Module GSM PWKEY HIGH");
	vTaskDelay(10000/portTICK_PERIOD_MS);		// modem is ready 10s after pull down of PWKEY

    return ESP_OK;
}
```

....
```
modem_dce_t *sim800_init(modem_dte_t *dte)
{
....
    sim800_dce->parent.deinit = sim800_deinit;

    /* Turn modem Power ON */
    DCE_CHECK( sim800_power_ON(sim800_dce) == ESP_OK, "Power ON failed", err_io);                //MaGe

    /* Sync between DTE and DCE */
    DCE_CHECK(esp_modem_dce_sync(&(sim800_dce->parent)) == ESP_OK, "sync failed", err_io);
    /* Close echo */
...
}
```

david-cermak, 2020-08-20:
Hi @mmggbe 

Thanks for sharing your initialization code. Indeed the current version of our PPPoS example is not very robust. We aimed to provide a demonstration of using AT commands and PPP mode, not really dealing with correct and stable power up sequence, as this is more up to the end application and also depends on custom HW for integrating of a GSM modem.

On the other hand, we have to provide a standard `esp-modem` component which should be stable enough and  set the modem board to a known state (possibly configurable) at least using RST signal.

Please keep this open, there's a plan to refactor the `esp_modem` into a standalone component which should have this (re)starting capabilities. Will close this issue once this new component is introduced.

closed 2021-09-10 by espressif-bot