espressif/esp-idf #5723: STA can not get an IP

stdenits, 2020-08-12:
## Environment

Development Kit: [none]
Kit version: [none]
Module or chip used: [ESP32-PICO-D4]
IDF version: v4.2
Build System: [???]
Compiler version: [???]
Operating System: [Windows]
(Windows only) environment type: [???].
Using an IDE?: [Yes (VSCode + ESP-IDF Extension)]
Power Supply: [external 3.3V]

## Problem Description

ESP32 in STA mode can not get an IP.

### Expected Behavior

1. check if ssid and pwd are blank;
1.1. starting STA mode if ssid and pwd are not blank;
1.2. starting AP mode if ssid and pwd are blank;
1.2.1. loading STA wifi config from NVS;
1.2.2. if loaded ssid and pwd are blank -> proceed AP mode
1.2.3. if loaded ssid and pwd are not blank -> switching to STA mode to proceed
2. if ssid and pwd are not correct when connecting - erasing them and save to NVS
3. go to p. 0

### Actual Behavior

STA hangs when connected to WiFi router and cannot get IP when ssid and pwd are loaded from Wi-Fi NVS configuration. SSID and PWD are correct.

P.S. ESP32 panick'ed when esp_event_loop and mDNS service was not deinitialized / freed.

### Steps to reproduce

### Code to reproduce this issue

Code: https://gist.github.com/stdenits/38f4b6de4fee9035c99b37bd8c0223c5

## Debug Logs

**Correct behavior log**
```
I (30) boot: ESP-IDF v4.2-dev-1905-g625bd5eb1 2nd stage bootloader
I (30) boot: compile time 19:30:13
I (30) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot:  3 storage          Unknown data     01 82 00110000 000f0000
I (101) boot: End of partition table
I (106) boot_comm: chip revision: 1, min. application chip revision: 0
I (113) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1d3d4 (119764) map
I (167) esp_image: segment 1: paddr=0x0002d3fc vaddr=0x3ffb0000 size=0x02c1c ( 11292) load
I (172) esp_image: segment 2: paddr=0x00030020 vaddr=0x400d0020 size=0x91910 (596240) map
0x400d0020: _stext at ??:?

I (401) esp_image: segment 3: paddr=0x000c1938 vaddr=0x3ffb2c1c size=0x00ee4 (  3812) load
I (403) esp_image: segment 4: paddr=0x000c2824 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/Users/Nox/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (408) esp_image: segment 5: paddr=0x000c2c30 vaddr=0x40080404 size=0x16ad4 ( 92884) load
[... 154 lines cut ...]
I (870) wifi:config NVS flash: enabled
I (870) wifi:config nano formating: disabled
I (870) wifi:Init dynamic tx buffer num: 32
I (870) wifi:Init data frame dynamic rx buffer num: 32
I (880) wifi:Init management frame dynamic rx buffer num: 32
I (890) wifi:Init management short buffer num: 32
I (890) wifi:Init static rx buffer size: 1600
I (890) wifi:Init static rx buffer num: 10
I (900) wifi:Init dynamic rx buffer num: 32
I (900) ESP32_Server: WiFi TEST 
I (910) ESP32_Server: PSW TEST 
I (1010) phy: phy_version: 4182, f1ba940, Jun  4 2020, 19:40:07, 0, 0
I (1010) wifi:mode : sta (d8:a0:1d:4a:3c:a0)
I (1140) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:13
I (1140) wifi:state: init -> auth (b0)
I (1150) wifi:state: auth -> assoc (0)
I (1160) wifi:state: assoc -> run (10)
I (1240) wifi:connected with TEST, aid = 3, channel 6, BW20, bssid = 3c:19:7d:af:86:bd
I (1240) wifi:security type: 4, phy: bgn, rssi: -89
I (1240) wifi:pm start, type: 1

I (1350) wifi:AP's beacon interval = 102400 us, DTIM period = 1
```

## Other items if possible

Alvin1Zhang, 2020-08-13:
Thanks for reporting and letting us know, we will look into.

stdenits, 2020-08-13:
The same thing on ESP IDF v4.3.

The code based on older version of ESP IDF. I'm interested if i screwed up with modifying the code for the current master branch (v4.*).

liuzfesp, 2020-08-13:
Hi @stdenits , the log `rssi: -89` indicates that the AP is far away from ESP32 devices, suggest keep AP closer to ESP32.

stdenits, 2020-08-14:
@liuzfesp
Hello. 
Check the log of the correct behavior: the signal strength there is also not very good and it works fine, as expected.

liuzfesp, 2020-08-14:
Hi @stdenits, it's a bug about `esp_wifi_set_storage(WIFI_STORAGE_RAM)`, we already have fix for it, but it need sometime for it to be merged into IDF.

stdenits, 2020-08-14:
>it's a bug about esp_wifi_set_storage(WIFI_STORAGE_RAM)
Could you please explain how my situation and this bug are related?

Any way to test the fix?

AxelLin, 2020-08-14:
> Hi @stdenits, it's a bug about `esp_wifi_set_storage(WIFI_STORAGE_RAM)`, we already have fix for it, but it need sometime for it to be merged into IDF.

I'm wondering if you have some policy to speed up merging critical fixes into IDF.
Especially for panic issues or this issue which causes connecting AP failure.

AxelLin, 2020-08-20:
> Hi @stdenits, it's a bug about `esp_wifi_set_storage(WIFI_STORAGE_RAM)`, we already have fix for it, but it need sometime for it to be merged into IDF.

Can you provide the link to the commit? I don't find such fix in esp32-wifi-lib.
Thanks.

liuzfesp, 2020-08-20:
HI @AxelLin, we haven't merge the fix yet, it's still under internal reviewing and we still need some time before it can be merged.

AxelLin, 2020-09-12:
Hi @liuzfesp 

I still don't find the fix (from the commit log) in esp32-wifi-lib after waiting for 3 weeks.
Now I'm worry that the fix won't be available in recent esp-idf update.
Can you tell which branches are impacted by this issue?

liuzfesp, 2020-09-14:
Hi @AxelLin, since this is a big change for us, we still need some time for it. 

The root cause of this issue is that the configuration of AP is NOT store into NVS correctly. As a workaround, you can fix this issue by storing the AP configuration into NVS **twice**.

To be detailed,

1. Call esp_wifi_get_config(WIFI_IF_AP, &ap_config) to get WiFi configuration
2. Fill ap_config with new ssid/password and other info
2. **Increase ap_config.ap.beacon_interval by one: ap_config.ap.beacon_internal +=1**
3. Call esp_wifi_set_config(WIFI_IF_AP, &ap_config)
4. **Recovery original beacon_interval: ap_config.ap.beacon_internal -= 1**
5. Call esp_wifi_set_config(WIFI_IF_AP, &ap_config) again

AxelLin, 2020-09-14:
Hi @liuzfesp 

Thanks for the update and I didn't mean to push you.
Look back the discussion on 14 Aug, you said "it's a bug about esp_wifi_set_storage(WIFI_STORAGE_RAM), we already have fix for it".
So I'm surprised that you said "this is a big change" now.
And I don't understand the "bug about esp_wifi_set_storage(WIFI_STORAGE_RAM)" and the relationship of your explain detail.
(Why storing AP configuration twice can work?)

PS.
I understand it's closed source library, I wouldn't ask the implementation detail.
But I would like to know the user visible impact, **especially which branches are impacted by this bug**.
Is this a new bug (regression)? (If yes, which commit causes the problem?)

liuzfesp, 2020-09-14:
Hi @AxelLin 

This issue exists in all active IDF release for very long time, all branches are impacted by this bug.

Yes, we already have fix for it for more than one month. When I said `it's a big change`, here it doesn't mean the fix is related to a lot of code change, it just means we have to merge the fix very carefully because it's about WiFi NVS change. Generally speaking, we should always take WiFi NVS related change carefully in case it may introduce new bug about OTA (OTA bug is much more serious bug for us).  For this kind of code change, we internally need several round of tests before we can finally merge it.

The root cause is because `esp_wifi_set_config()` returns directly if it finds that the passing configurations are same as the value stored in **RAM**.  In above workaround, the first esp_wifi_set_config call just modifies a parameter `beacon_interval` and make sure it's different than the value stored in RAM, the 2nd call change the value back. Beside `esp_wifi_set_config()`, other WiFi APIs related to NVS storing also have similar issue, we will fix all of them in the fix.

Some background information about the intention of API `esp_wifi_set_storage(WIFI_STORAGE_RAM)`:

- In ESP32, originally this API is only used for debug reason even though is not prefixed with `esp_wifi_internal_`.
- We internally have a lot of stress testing, some test involving frequently WiFi NVS writing/reading, e.g. in WiFi mode switching testing, we may switch the mode for more than 1000000 times in one stress test. Each time we do a mode switching, we have to read/write the WiFi NVS, it will impact the lifetime of WiFi NVS (which is flash) for so frequency NVS read/write. So we introduce API `esp_wifi_set_storage(WIFI_STORAGE_RAM)`. During stress test involves a huge number of NVS read/write, we just store the configuration into RAM to avoid impacting the NVS.

Now our customers have innovative way to use `esp_wifi_set_storage(WIFI_STORAGE_RAM)`, we need to fix the bug anyway.

stdenits, 2020-10-02:
@liuzfesp
Sorry for not answering for so long. I was sick.

How is the AP configuration related to the fact that ESP32 in the STAtion mode cannot connect to the access point? Although it seems to receive the correct username and password.

Also i'm using ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_FLASH)), not RAM only.

stdenits, 2020-10-27:
Could anybody give me some tips to find  out the reason?

Tried to connect to the smartphone AP's, new log attached below.
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6908
load:0x40078000,len:14304
load:0x40080400,len:3672
0x40080400: _init at ??:?

entry 0x40080678
I (29) boot: ESP-IDF v4.3-dev-1561-g357a27760 2nd stage bootloader
I (29) boot: compile time 13:45:36
I (30) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot:  3 storage          Unknown data     01 82 00110000 00100000
I (101) boot: End of partition table
I (105) boot_comm: chip revision: 1, min. application chip revision: 0
I (112) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1c0c4 (114884) map
I (165) esp_image: segment 1: paddr=0x0002c0ec vaddr=0x3ffb0000 size=0x03f2c ( 16172) load
I (172) esp_image: segment 2: paddr=0x00030020 vaddr=0x400d0020 size=0x929bc (600508) map
0x400d0020: _stext at ??:?

I (401) esp_image: segment 3: paddr=0x000c29e4 vaddr=0x3ffb3f2c size=0x00188 (   392) load
I (401) esp_image: segment 4: paddr=0x000c2b74 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/Users/User/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (408) esp_image: segment 5: paddr=0x000c2f80 vaddr=0x40080404 size=0x1567c ( 87676) load
I (454) esp_image: segment 6: paddr=0x000d8604 vaddr=0x400c0000 size=0x00064 (   100) load
I (455) esp_image: segment 7: paddr=0x000d8670 vaddr=0x50000000 size=0x00064 (   100) load
I (473) boot: Loaded app from partition at offset 0x10000
I (473) boot: Disabling RNG early entropy source...
I (486) cpu_start: Pro cpu up.
I (486) cpu_start: Starting app cpu, entry point is 0x40081294
0x40081294: call_start_cpu1 at C:/Users/User/esp-idf/components/esp_system/port/cpu_start.c:124

I (0) cpu_start: App cpu up.
I (500) cpu_start: Pro cpu start user code
I (501) cpu_start: cpu freq: 160000000
I (501) cpu_start: Application information:
I (505) cpu_start: Project name:     wifi_softAP
I (510) cpu_start: App version:      1
I (515) cpu_start: Compile time:     Oct 27 2020 16:57:48
I (521) cpu_start: ELF file SHA256:  116d077734ed02de...
I (527) cpu_start: ESP-IDF:          v4.3-dev-1561-g357a27760
I (533) heap_init: Initializing. RAM available for dynamic allocation:
I (541) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (547) heap_init: At 3FFB8A28 len 000275D8 (157 KiB): DRAM
I (553) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (559) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (566) heap_init: At 40095A80 len 0000A580 (41 KiB): IRAM
I (573) spi_flash: detected chip: gd
I (576) spi_flash: flash io: dio
I (581) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (591) ESP32_Server: app_main() runned on core 0

[... 51 lines cut ...]
I (1961) wifi_init: tcp mbox: 6
I (1971) wifi_init: tcp tx win: 5744
I (1971) wifi_init: tcp rx win: 5744
I (1971) wifi_init: tcp mss: 1440
I (1981) wifi_init: WiFi IRAM OP enabled
I (1981) wifi_init: WiFi RX IRAM OP enabled
I (1991) wifi:Set ps type: 0

I (1991) ESP32_Server: WiFi arp_pkg_calc 
I (2001) ESP32_Server: PSW 123456789 
I (2111) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (2111) wifi:mode : sta (d8:a0:1d:4a:3c:a0)
I (2111) wifi:enable tsf
I (3441) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:13
I (5471) wifi:state: init -> auth (b0)
I (5541) wifi:state: auth -> assoc (0)
I (5551) wifi:state: assoc -> run (10)
I (5581) wifi:connected with arp_pkg_calc, aid = 7, channel 11, BW20, bssid = 20:da:22:7b:a4:61
I (5581) wifi:security: WPA2-PSK, phy: bgn, rssi: -41
I (5581) wifi:pm start, type: 0

I (5621) wifi:AP's beacon interval = 102400 us, DTIM period = 2
```

[image]

HarveyRong-Esp, 2020-10-28:
Hi @stdenits,

Can you provide a wireless packet capture corresponding to the unconnected wifi log?
Or provide the smallest reproduction code and phone model for our local reproduction?

The recommended Macbook's built-in packet capture tool does not require a packet capture card. It can be used to capture 802.11a/b/g/n/ac packets:
https://osxdaily.com/2015/04/23/sniff-packet-capture-packet-trace-mac-os-x-wireless-diagnostics/
Or you can use the tutorial on the official Espressif website to capture packets using Wireshark
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html

stdenits, 2020-10-28:
@HarveyRong-Esp,

Hello.
Thank you for your reply. But i just decided to decline this function and use static STA configuration.
Maybe be later it will work as intended.

closed 2020-10-28 by stdenits

AxelLin, 2020-11-17:
> Hi @AxelLin
> 
> This issue exists in all active IDF release for very long time, all branches are impacted by this bug.

I just notice this issue was closed, which commit fixed this issue?