espressif/esp-idf #5725: esp_https_ota_begin/esp_https_ota_perform much slower than esp_ota_begin/esp_ota_write

korstiaanS, 2020-08-13:
Hi,

Using SDK 4.0.1.

Is it normal behaviour that when I use OTA via `esp_https_ota_begin + esp_https_ota_perform + esp_https_ota_finish` it is more than 5x slower than when I use `esp_ota_begin + esp_ota_write + esp_ota_end`?
For the second (fast) solution I uploaded the .bin file via a curl command.

Can I speed up the https way?

Korstiaan

shubhamkulkarni97, 2020-08-14:
Hi @korstiaanS,

`esp_ota_begin + esp_ota_write + esp_ota_end` APIs are basically used to write OTA data to specified partition.

On the other hand, when `esp_https_ota` APIs are used, HTTP request is sent, TLS handshake is performed, the response from server is received and then data is written to partition using `app_update` APIs. Throughput will also depend on network status.

This will reduce overall throughput with `esp_https_ota` APIs.

To increase throughput with HTTPS, you modify following configuration(in menuconfig):

```
CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=16
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=64
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM=64
CONFIG_ESP32_WIFI_TX_BA_WIN=32
CONFIG_ESP32_WIFI_RX_BA_WIN=32
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=65534
CONFIG_LWIP_TCP_WND_DEFAULT=65534
CONFIG_LWIP_TCP_RECVMBOX_SIZE=64
CONFIG_LWIP_UDP_RECVMBOX_SIZE=64
CONFIG_LWIP_TCPIP_RECVMBOX_SIZE=64
```
**NOTE:** These changes will also increase memory consumption

Hope this helps you.

Thanks,
Shubham

shubhamkulkarni97, 2020-08-20:
@korstiaanS, Is you issue resolved?

korstiaanS, 2020-08-20:
I'm not risking it to change these parameters.
I'll keep the existing ones and accept the fact that it is much slower.

closed 2020-08-20 by korstiaanS