espressif/esp-idf #5726: Increasing esp idf mesh data transfer limit (MTU size)

Suraj-singh999898, 2020-08-13:
"esp_mesh_send" function can't send more than 1500 byte long data because of its internal MTU size (https://github.com/espressif/esp-idf/issues/2816).
I need to transfer 5000byte between nodes.

if possible then the best solution would be some kind of data streaming like HTTP has "esp_http_client_get_chunk_length"
otherwise increasing MTU size also works.

ESP-iPENCIL, 2020-08-14:
esp_mesh_send does not support Fragmentation. One packet size is limited to MESH_MTU( defined in ./esp-idf/components/esp32/include/esp_mesh.h).
You may need to fragment and reassemble the application data by yourself. Sorry for the convinent.

jarus0, 2020-11-27:
thanks, problem solved

closed 2020-11-27 by jarus0

fira-code, 2020-12-16:
@jarus0 hey, I'm curious to know how you dealt with the fragmentation issue.
I'm working on it and facing memory issues so if you can share informations on how you did it, it'll be very useful.
thx

jarus0, 2020-12-17:
So here you have 2 options.( Both working fine)

Option 1 
Switch "esp mesh idf" to "mdf"

Option 2 
Write you own fragments feature.
- Sender first send size of data with CRC.
- Receiver will Merge & verified CRC.
- I also written quene and message identify code so receiver can handle multiple receiving at same time.