espressif/esp-idf #5727: How to recover if esp_https_ota_begin fails?

korstiaanS, 2020-08-13:
Version: 4.0.1

Hi,

I noticed that if the call to `esp_https_ota_begin` fails, because the https server is not active at that moment, the only way to use it again is to restart the esp32.
So if I start the https server after 1 fail, it is too late and the only way to get to work is to restart the esp32.
Is there a way to avoid a complete restart of the esp32?

If I execute `esp_https_ota_begin` without an active https server I get (=normal):

```
E (59859) esp-tls: Failed to connnect to host (errno 104)
E (59863) esp-tls: Failed to open new connection
E (59866) TRANS_SSL: Failed to open a new connection
E (59868) HTTP_CLIENT: Connection failed, sock < 0
E (59872) esp_https_ota: Failed to open HTTP connection: ERROR
E (59875) esp_https_ota: Failed to establish HTTP connection
E (59878) 341_main: ESP HTTPS OTA Begin failed
```
If I then start the https server and try again I get the following errors when trying to use `esp_https_ota_get_img_desc` after the `esp_https_ota_begin`

```
E (79416) esp_https_ota: Connection closed, errno = 104
E (79422) esp_https_ota: Complete headers were not received
E (79424) 341_main: esp_https_ota_read_img_desc failed
```

Will never work again until restart.

Any way to solve this? It can happen a lot that the https server is not available/reachable at the moment of trying.

Korstiaan

shubhamkulkarni97, 2020-08-14:
Hi @korstiaanS,

* If `esp_ota_begin` fails, `esp_https_ota_get_img_desc` will never succeed. For your use case, `esp_ota_begin` needs to be called till it returns ESP_OK.

* While checking for the issue, I found that Openssl server drops connection if `esp_ota_begin` is called in a loop, due to which `Connection closed, errno = 104` is printed in logs. This looks like a server side issue, can you try using python server instead and check if it works?

Command to run python server: `python -m SimpleHTTPServer PORT`

Thanks,
Shubham

korstiaanS, 2020-08-14:
@shubhamkulkarni97,

- it is not `esp_ota_begin` but `esp_https_ota_begin`
- I do ALWAYS check if `esp_https_ota_begin` succeeds before continuing
- my issue has nothing to do with dropping connection
- I said, if it fails once, it will NEVER work again until restart of ESP32.

My question is a real life scenario, a webserver can be down, not reachable, ... when the esp is trying to connect. The issue is that if that happens the ESP cannot recover.

shubhamkulkarni97, 2020-08-14:
@korstiaanS,
> 
> * it is not `esp_ota_begin` but `esp_https_ota_begin`

Sorry for the typo, I meant `esp_https_ota_begin`.

Can you try patch attached below and run `advanced_https_ota` example?

This patch does work for me with python server for your use case(starting server after esp_https_ota_begin fails). ESP32 is able to connect to server and perform OTA Update successfully.

[esp_https_ota_begin.patch.zip](https://github.com/espressif/esp-idf/files/5073682/esp_https_ota_begin.patch.zip)

shubhamkulkarni97, 2020-08-20:
@korstiaanS, Is your issue resolved?

korstiaanS, 2020-08-20:
Hi,

It is strange but yes, if I use a Python webserver instead of an openssl server I don't get that problem. 
I created following script to make this a secure webserver:

```
#!/usr/bin/env python
import BaseHTTPServer, SimpleHTTPServer
import ssl

print("Starting web server on port 8080...")

httpd = BaseHTTPServer.HTTPServer(('10.0.18.16', 8080),
        SimpleHTTPServer.SimpleHTTPRequestHandler)

httpd.socket = ssl.wrap_socket (httpd.socket,
        keyfile='server_certs/ca_key.pem',
        certfile='server_certs/ca_cert.pem', server_side=True)

httpd.serve_forever()

```

closed 2020-08-20 by korstiaanS