espressif/esp-idf #5728: NimBLE MIC failure on GATT writes

errno, 2020-08-13:
## Environment

- Development Kit: none
- Module or chip used: ESP32-WROOM-32
- IDF version: v4.3-dev-771-gc77c4ccf6
- Build System: idf.py
- Compiler version: 8.2.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

I'm trying to achieve bulk data transfer from central (iOS) to peripheral (ESP) via custom GATT service/characteristic.
Central makes writes to GATT characteristic, and awaits for notification which indicates that ESP is ready for next chunk.
And it works fine until at random chunk ESP disconnects with reason 0x3d (MIC failure).
Devices are in close proximity. ESP does nothing else then dumping received data to monitor console.

### Expected Behavior

ESP receives data reliably.

### Actual Behavior

Disconnect caused by MIC failure (0x3d).

### Steps to reproduce

1. Generate intensive writes to GATT characteristic.
2. Wait until it fails.

### Code to reproduce this issue

This is the code handling writes on ESP side:

```cpp
case BLE_GATT_ACCESS_OP_WRITE_CHR:
        ESP_LOGI("--- >>>", "OS_MBUF_PKTLEN - %d", OS_MBUF_PKTLEN(ctxt->om));
        rc = gatt_svr_chr_write(ctxt->om,
                                OS_MBUF_PKTLEN(ctxt->om),
                                OS_MBUF_PKTLEN(ctxt->om),
                                &inlet_buff, NULL);
        // return rc;
        if (rc == 0) {
          for (int j = 0; j < OS_MBUF_PKTLEN(ctxt->om); j++) {
            printf("0x%02x\t", inlet_buff[j]);
          }
          printf("\n");
          struct os_mbuf *om;
          data_tmp = !data_tmp;
          om = ble_hs_mbuf_from_flat(&data_tmp, sizeof(data_tmp));
          rc = ble_gattc_notify_custom(conn_handle, dev_ota_inlet_handle, om);
          ESP_LOGI("rc:", "%d", rc);
        }
        return 0;
```

## Debug Logs

```
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=27 data=0xb9 0x00 0x04 0x00 0x16 0x1b 0x00 0x00 0x00 0x54 0x50 0x53 0x20 0x4f 0x54 0x41 0x20 0x53 0x74 0x61 0x74 0x65 0x1b 0x5b 0x30 0x6d 0x0a 
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=1 len=27 data=0x00 0x00 0x1b 0x5b 0x30 0x3b 0x33 0x31 0x6d 0x45 0x20 0x28 0x25 0x64 0x29 0x20 0x25 0x73 0x3a 0x20 0x65 0x73 0x70 0x5f 0x6f 0x74 0x61 
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=1 len=27 data=0x5f 0x73 0x65 0x74 0x5f 0x62 0x6f 0x6f 0x74 0x5f 0x70 0x61 0x72 0x74 0x69 0x74 0x69 0x6f 0x6e 0x20 0x66 0x61 0x69 0x6c 0x65 0x64 0x21 
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=1 len=27 data=0x20 0x65 0x72 0x72 0x3d 0x30 0x78 0x25 0x64 0x1b 0x5b 0x30 0x6d 0x0a 0x00 0x00 0x1b 0x5b 0x30 0x3b 0x33 0x31 0x6d 0x45 0x20 0x28 0x25 
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=1 len=27 data=0x64 0x29 0x20 0x25 0x73 0x3a 0x20 0x65 0x73 0x70 0x5f 0x68 0x74 0x74 0x70 0x5f 0x63 0x6c 0x69 0x65 0x6e 0x74 0x20 0x63 0x6f 0x6e 0x66 
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=1 len=27 data=0x69 0x67 0x20 0x6e 0x6f 0x74 0x20 0x66 0x6f 0x75 0x6e 0x64 0x1b 0x5b 0x30 0x6d 0x0a 0x00 0x6f 0x74 0x61 0x5f 0x61 0x70 0x70 0x5f 0x63 
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=1 len=27 data=0x6f 0x75 0x6e 0x74 0x20 0x3c 0x20 0x31 0x36 0x20 0x26 0x26 0x20 0x22 0x6d 0x75 0x73 0x74 0x20 0x65 0x72 0x61 0x73 0x65 0x20 0x74 0x68 
rxed att command: prep write req; conn=0 handle=0x001b offset=0
txed att command: prep write rsp; conn=0 handle=0x001b offset=0
host tx hci data; handle=0 length=189
ble_hs_hci_acl_tx(): 0x00 0x00 0xbd 0x00 0xb9 0x00 0x04 0x00 0x17 0x1b 0x00 0x00 0x00 0x54 0x50 0x53 0x20 0x4f 0x54 0x41 0x20 0x53 0x74 0x61 0x74 0x65 0x1b 0x5b 0x30 0x6d 0x0a 0x00 0x00 0x1b 0x5b 0x30 0x3b 0x33 0x31 0x6d 0x45 0x20 0x28 0x25 0x64 0x29 0x20 0x25 0x73 0x3a 0x20 0x65 0x73 0x70 0x5f 0x6f 0x74 0x61 0x5f 0x73 0x65 0x74 0x5f 0x62 0x6f 0x6f 0x74 0x5f 0x70 0x61 0x72 0x74 0x69 0x74 0x69 0x6f 0x6e 0x20 0x66 0x61 0x69 0x6c 0x65 0x64 0x21 0x20 0x65 0x72 0x72 0x3d 0x30 0x78 0x25 0x64 0x1b 0x5b 0x30 0x6d 0x0a 0x00 0x00 0x1b 0x5b 0x30 0x3b 0x33 0x31 0x6d 0x45 0x20 0x28 0x25 0x64 0x29 0x20 0x25 0x73 0x3a 0x20 0x65 0x73 0x70 0x5f 0x68 0x74 0x74 0x70 0x5f 0x63 0x6c 0x69 0x65 0x6e 0x74 0x20 0x63 0x6f 0x6e 0x66 0x69 0x67 0x20 0x6e 0x6f 0x74 0x20 0x66 0x6f 0x75 0x6e 0x64 0x1b 0x5b 0x30 0x6d 0x0a 0x00 0x6f 0x74 0x61 0x5f 0x61 0x70 0x70 0x5f 0x63 0x6f 0x75 0x6e 0x74 0x20 0x3c 0x20 0x31 0x36 0x20 0x26 0x26 0x20 0x22 0x6d 0x75 0x73 0x74 0x20 0x65 0x72 0x61 0x73 0x65 0x20 0x74 0x68 
Number of Completed Packets: num_handles=1
handle:0 pkts:1
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=27 data=0xb9 0x00 0x04 0x00 0x16 0x1b 0x00 0xb4 0x00 0x65 0x20 0x70 0x61 0x72 0x74 0x69 0x74 0x69 0x6f 0x6e 0x20 0x62 0x65 0x66 0x6f 0x72 0x65 
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=1 len=27 data=0x20 0x77 0x72 0x69 0x74 0x69 0x6e 0x67 0x20 0x74 0x6f 0x20 0x69 0x74 0x22 0x00 0x00 0x00 0x2f 0x62 0x75 0x69 0x6c 0x64 0x73 0x2f 0x2d 
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=1 len=27 data=0x79 0x38 0x56 0x41 0x7a 0x4b 0x58 0x2f 0x30 0x2f 0x53 0x6d 0x61 0x72 0x74 0x53 0x74 0x72 0x65 0x61 0x6d 0x2f 0x73 0x6d 0x61 0x72 0x74 
Disconnection Complete: status=0 handle=0 reason=61
Revised our id addr:
0xc3 0xe7 0x14 0x2d 0x30 0x4d 
subscribe event; conn_handle=0 attr_handle=8 reason=2 prevn=0 curn=0 previ=1 curi=0
subscribe event; conn_handle=0 attr_handle=27 reason=2 prevn=1 curn=0 previ=0 curi=0
disconnect; reason=573 handle=0 our_ota_addr_type=1 our_ota_addr=4d:30:2d:14:e7:c3 our_id_addr_type=0 our_id_addr=30:ae:a4:7b:04:de peer_ota_addr_type=0 peer_ota_addr=48:3b:38:57:b4:60 peer_id_addr_type=0 peer_id_addr=48:3b:38:57:b4:60 conn_itvl=18 conn_latency=0 supervision_timeout=72 encrypted=1 authenticated=0 bonded=1

ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0007 len=0
0x07 0x20 0x00 
Command complete: cmd_pkts=5 ogf=0x8 ocf=0x7 status=0 
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0008 len=32
0x08 0x20 0x20 0x10 0x02 0x01 0x06 0x09 0x09 0x46 0x6c 0x61 0x6d 0x69 0x6e 0x67 0x6f 0x02 0x0a 0x03 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
Command complete: cmd_pkts=5 ogf=0x8 ocf=0x8 status=0 
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0006 len=15
0x06 0x20 0x0f 0x30 0x00 0x60 0x00 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x07 0x00 
Command complete: cmd_pkts=5 ogf=0x8 ocf=0x6 status=0 
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000a len=1
0x0a 0x20 0x01 0x01 
Command complete: cmd_pkts=5 ogf=0x8 ocf=0xa status=0 
```
## P.S.

In desperate attempt I've tried to reimplement this service with Bluedroid stack and got a similar result.

Alvin1Zhang, 2020-08-14:
Thanks for reporting, we will look into.

prasad-alatkar, 2020-08-18:
Hi @errno , apologies for delayed response. From the logs, it looks like iPhone is sending disconnect with error_code = `BLE_ERR_CONN_TERM_MIC`. The ESP32 disconnects because it receives `DISCONNECT COMPLETE` HCI event from iPhone.

>Disconnection Complete: status=0 handle=0  reason=61

I tried to reproduce the issue using ESP32 for both central and peripheral, I could not reproduce the issue for ~12Hours GATT writes from central. Similarly, I tried to reproduce the issue on Android phone, but issue was not reproducible for >15 minutes of testing. Could you please try to reproduce the issue using Android device?

errno, 2020-09-02:
Sorry for the delay.

@prasad-alatkar Thanks for your reply. I had the impression that ESP32 is the one who drops a connection.

I'll definitely try to implement Android central. But it will take some time. For now, I switched to L2CAP CoC, and it works OK with iPhone.

closed 2020-09-02 by errno