espressif/esp-idf #5729: After Reinitializing NimBLE, unable to adveritise with BLE_OWN_ADDR_RPA_RANDOM_DEFAULT or BLE_OWN_ADDR_RANDOM

sebirdman, 2020-08-13:
## Environment

- IDF version: 4.0.1
- Build System: cmake

## Problem Description

During my applications runtime, sometimes we deinit and reinit bluetooth for memory reasons. After reinit, it seems that we are unable to call ble_gap_adv_start with either an RPA_RANDOM or RANDOM mac address without it returning an error (530).

PUBLIC mac addresses do not have this issue.

### Expected Behavior

can call ble_gap_adv_start with RPA_RANDOM or RANDOM mac addresses and successfully advertise

### Actual Behavior

cannot call ble_gap_adv_start with RPA_RANDOM or RANDOM mac addresses and successfully advertise

### Steps to reproduce

1. tear down nimble as recommended in the docs here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/nimble/index.html#_CPPv436esp_nimble_hci_and_controller_deinitv
2. reinit using the docs here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/nimble/index.html#_CPPv434esp_nimble_hci_and_controller_initv
3. attempt to advertise with RPA_RANDOM or RANDOM

### Code to reproduce this issue

I can provide code if necessary, but it'll take some work to pull it out of my application.

prasad-alatkar, 2020-08-13:
Hi @sebirdman, As I understand there are two part to the answers, one for RANDOM and second for RPA address type.

 **1. RANDOM:**
- If you want to advertise with `addr_type` as RANDOM (`BLE_OWN_ADDR_RANDOM`),  please set random address using `ble_hs_id_set_rnd` API. You may want to generate random address using `ble_hs_id_gen_rnd`.

**2. RPA (RPA_PUBLIC/ RPA_RANDOM):** 
- To use privacy feature for NimBLE host on ESP32, we need to call `ble_hs_pvcy_rpa_config` in Host-Controller sync callback(or after). For detailed documentation of this API, refer: https://github.com/espressif/esp-nimble/blob/nimble-1.2.0-idf/nimble/host/include/host/ble_hs_pvcy.h#L66). 
- Please use `own_addr_type` as `BLE_OWN_ADDR_RANDOM` while using host based privacy. We do not use addr_types `BLE_OWN_ADDR_RPA_PUBLIC_DEFAULT` and `BLE_OWN_ADDR_RPA_RANDOM_DEFAULT` in host based privacy.
- You may want to add `BLE_SM_PAIR_KEY_DIST_ID` in key distributions of `ble_hs_cfg` (`sm_our_key_dist` & `sm_their_key_dist`).

Please let me know if you still face any issue.

sebirdman, 2020-08-13:
hi @prasad-alatkar, thanks for the fast response

1. Random:

I'm currently setting and generating a random address every time right after i initialize the Bluetooth stack. this includes during re-initialization. 

2. RPA:

That makes sense. Our firmware is not yet using RPA, but when i found this bug i tried all the different address types to see which one worked and this one didn't work.

I should note that ALL the address types work after the first bluetooth stack init, after the second is only when these two stop working.

prasad-alatkar, 2020-08-17:
Hi @sebirdman , I tried to reproduce the scenario, but did not face any issue with setting RANDOM address after multiple NimBLE re-init cycles. I am attaching sample patch for your reference based on [bleprph](https://github.com/espressif/esp-idf/blob/v4.0.1/examples/bluetooth/nimble/bleprph/main/main.c) example: [Sample_patch_for_random_address_after_re-init.txt](https://github.com/espressif/esp-idf/files/5084569/Sample_patch_for_random_address_after_re-init.txt) . Please let me know if it helps you resolve the issue. If you still face the issue, Can you please provide sample patch to reproduce the issue ? If sharing code may take time, is it possible for you to share debug logs for the issue ?

prasad-alatkar, 2020-09-22:
@sebirdman , Can you please provide current update ? Did above shared patch help resolve the issue ?

Alvin1Zhang, 2020-11-05:
@sebirdman Thanks for reporting, would you please help share if any updates for this issue? Thanks.

prasad-alatkar, 2021-01-08:
@sebirdman Could you please help to confirm if the patch was helpful to resolve the issue ?

sebirdman, 2021-01-11:
I've not been able to reproduce this in some time without the patch. i suspect i was doing something odd.

closed 2021-01-11 by sebirdman

Alvin1Zhang, 2021-01-12:
Thanks for sharing the updates.