espressif/esp-idf #5730: After advertising for sometime with NimBLE, advertising start returns BLE_HS_ENOTSYNCED

sebirdman, 2020-08-13:
## Environment
- IDF version: 4.0.1
- Build System:  cmake

## Problem Description

My application advertises bluetooth packets often. eventually ble_gap_adv_start returns BLE_HS_ENOTSYNCED and i can no longer send advertisements.

### Expected Behavior

ble_gap_adv_start returns 0 on advertisements

### Actual Behavior

ble_gap_adv_start sometimes returns BLE_HS_ENOTSYNCED

### Steps to reproduce

1. advertise packets for a while
2. observe

I'm not really sure what i should be doing, or if there's some way to get the device back sync'ed again.

dhrishi, 2020-08-14:
@sebirdman Can you please provide additional details? I tried the below cases using the [bleprph example](https://github.com/espressif/esp-idf/blob/v4.0.1/examples/bluetooth/nimble/bleprph/main/main.c) and they seem to work fine:

1. Periodically, every 5 sec, start/stop advertisement.
2. Start advertisement with a finite duration of 5 sec (provided in `ble_gap_adv_start`) and it gets restarted in `BLE_GAP_EVENT_ADV_COMPLETE` and repeats.
3. Start advertisement forever and keep the device idle.

Please let me know if I am missing anything. Additionally, a code snippet to reproduce your issue would be really helpful.

sebirdman, 2020-08-14:
Hi @dhrishi 

The issue seems to happen several hours into the device in. Up to 12 hours at times. My application is using a mesh similar but predates the standard blemesh. It will sometimes advertise quite often, and we've tested it when advertising as fast as possible  between two devices.

My main use is your example #1. I can try to provide some repro code, but it'll take some time to pull out of my application. 

So far i've only seen this happen on one device in my mesh, i'll see what additional data i can gather.

prasad-alatkar, 2020-08-17:
Hi @sebirdman generally `BLE_HS_ENOTSYNCED` error comes when you try to use NimBLE host before host-controller are synced. As I see, you get this error after successfully running for ~12 HRS, it maybe because host-controller sync is in bad state. Generally, when in bad state, host calls `ble_hs_reset` which resets host and again try to bring host-controller in sync. But from description, it looks like repeated attempt from host to sync with controller is getting failed. Can you please provide debug logs (`menuconfig → Component config → Log output → Default log verbosity` to `Debug`) which depicts the issue ? It will be really helpful if you can provide patch/code to reproduce the issue.

prasad-alatkar, 2020-09-22:
Hi @sebirdman Did you get a chance to collect the debug logs ?

sebirdman, 2020-09-23:
@prasad-alatkar I've not been able to reproduce this anymore. I'm curious if i was doing something wrong. This was rather early in our development lifecycle for nimble.

Alvin1Zhang, 2020-09-24:
Thanks for reporting and updates, feel free to reopen if the issue still happens. Thanks.

closed 2020-09-24 by Alvin1Zhang