espressif/esp-idf #5731: Converting Code into modules with 4.0.1

ip-v1, 2020-08-14:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-Wrover Ethernet Kit
- Kit version: 1.2
- Module or chip used:  ESP32-WROVER-E
- IDF version v4.0.1
- Build System:        cmake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): esp-2019r2  8.2.0
- Operating System:   Windows
- (Windows only) environment type. Command Prompt
- Using an IDE?: Visual Studio Code with ESP IDF Plugin
- Power Supply:         USB

## Problem Description

After getting the basic ethernet code to work, I added http sever code to the same main file and it did what it should. As soon as I tried to create a component from the code, the linker started giving me errors on a function not being found.

### Expected Behavior

Build Complete

### Actual Behavior

Build fails at linker stage

### Steps to reproduce

1. step1 : With VSCode, have the ESP IDF plugin Installed and configured. 
2. step2: clone the repo at https://github.com/ip-v1/esp32_ethernet_http_component
3. step3: build using the build button

### Code to reproduce this issue

https://github.com/ip-v1/esp32_ethernet_http_component

## Debug Logs

> Executing task: cmake -G Ninja .. <

-- Found Git: C:/Users/isingh/AppData/Local/Programs/Git/mingw64/bin/git.exe (found version "2.27.0.windows.1")
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
[... 853 lines cut ...]
CONFIG_TCP_OVERSIZE_MSS=y
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_IP_LOST_TIMER_INTERVAL=120
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
# End of deprecated options

suda-morris, 2020-08-17:
First of all, this is not a esp-idf bugfix or feature request, it's a esp-idf usage question, we high recommend our users to use our [forum](https://esp32.com) to get a quicker response.

For your issue, seems like you have met some problems when using our build system.

* https://github.com/ip-v1/esp32_ethernet_http_component/blob/master/main/CMakeLists.txt#L3
  * Usually, main component doesn't have to specify dependency. It will look for all components in its sight. In your project, `ip_web_server` component is just in your projects' components folder, it's one of the default search path for our build system. So the main component can see it.
* https://github.com/ip-v1/esp32_ethernet_http_component/blob/master/components/ip_web_server/CmakeLists.txt#L3
  * Since this component has declare a dependency on `esp_eth` in the `ip_web_server.h`, you have to mention this **PUBLIC** dependency in the `REQUIRES` command.

Alvin1Zhang, 2020-08-17:
Thanks for reporting, please help raise a post on forum https://esp32.com. Thanks.

closed 2020-08-17 by Alvin1Zhang

ip-v1, 2020-08-17:
Hi and apologies for posting in the wrong place. I have opened a topic on esp32.com as suggested.

> For your issue, seems like you have met some problems when using our build system.
>
> * https://github.com/ip-v1/esp32_ethernet_http_component/blob/master/main/CMakeLists.txt#L3
>
>   * Usually, main [...]

Would this break anything?

> * https://github.com/ip-v1/esp32_ethernet_http_component/blob/master/components/ip_web_server/CmakeLists.txt#L3
>   
>   * Since this component has declare a dependency on `esp_eth` in the `ip_web_server.h`, you have to mention this **PUBLIC** dependency in the `REQUIRES` command.

Where additionally do I need to add this? I basically added it everywhere except for the root CMakeLists.txt

Thanks again for the reply.

ip-v1, 2020-08-18:
Apologies for posting in the wrong place. I have posted a question on ESP32.com 

Thank you.