espressif/esp-idf #5733: Websocket Server over TLS

Francois-Belanger, 2020-08-15:
Hi, every new IOT software we write is hosted over HTTPS (client application like IONIC that connects to the ESP32).  This got the effect that a simple websocket server is not allowed anymore. (An insecure WebSocket connection may not be initiated from a page loaded over HTTPS.)

I need to find a solution to connect from a web page to the ESP32 to transfer realtime data.  I'm currently doing an simple web socket server, but that doesn't work anymore :-(

The simplest solution would be to use a Secure WebSocket Server on the ESP32.  But I can't find any implementation that does that within IDF.  Many Client WSS samples on the web, but not any complete WSS Server.  Why is that ?  I've looked at the current WS I'm using.. But it's using netconn and going to mbedTLS looks complicated !

I'm trying to stream about 10kBytes / sec.

Thanks for the help !

Alvin1Zhang, 2020-08-17:
Thanks for raising this feature request.

david-cermak, 2020-08-27:
Hi @Francois-Belanger 

You can setup a WebSocket server on top of a [`esp_https_server` component](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_https_server.html) adding corresponding ws handlers and setting `is_websocket = true`, so combining these two below example in one:
* https://github.com/espressif/esp-idf/tree/master/examples/protocols/https_server
* https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/ws_echo_server

Here is an example of a wss-echo-server main file. To test it quickly you can just
* copy-paste this to the current [https_server](https://github.com/espressif/esp-idf/tree/master/examples/protocols/https_server) example
* update the certificate [from the example](https://github.com/espressif/esp-idf/blob/master/examples/protocols/https_server/README.md#certificates)

``` c
#include <esp_event.h>
#include <esp_log.h>
#include <esp_system.h>
#include <nvs_flash.h>
#include <sys/param.h>
#include "esp_netif.h"
#include "esp_eth.h"
#include "protocol_examples_common.h"

#include <esp_https_server.h>

static const char *TAG = "wss_echo_server";

static esp_err_t echo_handler(httpd_req_t *req)
{
    uint8_t buf[128] = { 0 };
    httpd_ws_frame_t ws_pkt;
    memset(&ws_pkt, 0, sizeof(httpd_ws_frame_t));
    ws_pkt.payload = buf;
    ws_pkt.type = HTTPD_WS_TYPE_TEXT;
    esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, 128);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "httpd_ws_recv_frame failed with %d", ret);
        return ret;
    }
    ESP_LOGI(TAG, "Received packet with message: %s", ws_pkt.payload);
    ret = httpd_ws_send_frame(req, &ws_pkt);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "httpd_ws_send_frame failed with %d", ret);
    }
    return ret;
}

static const httpd_uri_t ws = {
        .uri        = "/ws",
        .method     = HTTP_GET,
        .handler    = echo_handler,
        .user_ctx   = NULL,
        .is_websocket = true
};

static httpd_handle_t start_wss_echo_server(void)
{
    httpd_handle_t server = NULL;

    // Start the httpd server
    ESP_LOGI(TAG, "Starting server");

    httpd_ssl_config_t conf = HTTPD_SSL_CONFIG_DEFAULT();

    extern const unsigned char cacert_pem_start[] asm("_binary_cacert_pem_start");
    extern const unsigned char cacert_pem_end[]   asm("_binary_cacert_pem_end");
    conf.cacert_pem = cacert_pem_start;
    conf.cacert_len = cacert_pem_end - cacert_pem_start;

    extern const unsigned char prvtkey_pem_start[] asm("_binary_prvtkey_pem_start");
    extern const unsigned char prvtkey_pem_end[]   asm("_binary_prvtkey_pem_end");
    conf.prvtkey_pem = prvtkey_pem_start;
    conf.prvtkey_len = prvtkey_pem_end - prvtkey_pem_start;

    esp_err_t ret = httpd_ssl_start(&server, &conf);
    if (ESP_OK != ret) {
        ESP_LOGI(TAG, "Error starting server!");
        return NULL;
    }

    // Set URI handlers
    ESP_LOGI(TAG, "Registering URI handlers");
    httpd_register_uri_handler(server, &ws);

    return server;
}

static void stop_wss_echo_server(httpd_handle_t server)
{
    // Stop the httpd server
    httpd_ssl_stop(server);
}

static void disconnect_handler(void* arg, esp_event_base_t event_base,
                               int32_t event_id, void* event_data)
{
    httpd_handle_t* server = (httpd_handle_t*) arg;
    if (*server) {
        stop_wss_echo_server(*server);
        *server = NULL;
    }
}

static void connect_handler(void* arg, esp_event_base_t event_base,
                            int32_t event_id, void* event_data)
{
    httpd_handle_t* server = (httpd_handle_t*) arg;
    if (*server == NULL) {
        *server = start_wss_echo_server();
    }
}

void app_main(void)
{
    static httpd_handle_t server = NULL;

    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    /* Register event handlers to start server when Wi-Fi or Ethernet is connected,
     * and stop server when disconnection happens.
     */

#ifdef CONFIG_EXAMPLE_CONNECT_WIFI
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &connect_handler, &server));
    ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, WIFI_EVENT_STA_DISCONNECTED, &disconnect_handler, &server));
#endif // CONFIG_EXAMPLE_CONNECT_WIFI
#ifdef CONFIG_EXAMPLE_CONNECT_ETHERNET
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &connect_handler, &server));
    ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ETHERNET_EVENT_DISCONNECTED, &disconnect_handler, &server));
#endif // CONFIG_EXAMPLE_CONNECT_ETHERNET

    /* This helper function configures Wi-Fi or Ethernet, as selected in menuconfig.
     * Read "Establishing Wi-Fi or Ethernet Connection" section in
     * examples/protocols/README.md for more information about this function.
     */
    ESP_ERROR_CHECK(example_connect());
}

```

Francois-Belanger, 2020-08-27:
Hi @david-cermak this is great and sure looks simple.  Sadly I had to change direction for lack of options, but I will surely use it later.  Can you please add this to the examples ?  I'm sure it will be **very** useful for the community :-)

Thanks again !
Francois

closed 2020-09-16 by espressif-bot

baldhead69, 2020-09-18:
Hi @david-cermak,

I am trying using your suggestion above.

***************************************************************************************************************************
Chrome Javascript:

When i trying to connect from chrome browser, the esp32 log below appear, and the chrome message(undefined error).

I (62861) esp_https_server: performing session handshake
E (63731) esp-tls-mbedtls: mbedtls_ssl_handshake returned -30592
E (63731) esp_https_server: esp_tls_create_server_session failed
W (63771) httpd: httpd_accept_conn: session creation failed
W (63777) httpd: httpd_server: error accepting new connection

If i register a new uri in addition to the websocket uri, and first access this uri, ie: https://192.168.0.8/config
the same messages appear, but, i accept the chrome security exception, and ends up connecting(message: Hello Secure World!).

After that, i can connect to websocket uri, ie: https://192.168.0.8/
if i clean the chrome history(cache), the same problem appear again.

`static const httpd_uri_t get_access_point_page =
    {
        .uri       = "/config",  
        .method    = HTTP_GET,
        .handler   = get_access_point_page_handler,
        .user_ctx  = NULL,
        .is_websocket = false
    };
    httpd_register_uri_handler(server, &get_access_point_page);

static esp_err_t get_access_point_page_handler(httpd_req_t *req)
{
    httpd_resp_set_type(req, "text/html");
    httpd_resp_send(req, "<h1>Hello Secure World!</h1>", HTTPD_RESP_USE_STRLEN);

    return ESP_OK;
}
`
*********************************************************************************************************************************
Android studio app using okhttp3 secure websocket(wss):

Erro message:  java.security.cert.CertPathValidatorException: Trust anchor for certification path not found. 

If i try to connect to "wss://echo.websocket.org", from app, works ok.

Note: Android studio app using okhttp3 websocket(ws), works ok before i am using wss.

**********************************************************************************************************************************

david-cermak, 2020-09-18:
@baldhead69 Have you checked [this note](https://github.com/espressif/esp-idf/tree/master/examples/protocols/https_server/simple#certificates) about using self signed certificates? What kind of CA did you use with your example?
Please note that the `wss://echo.websocket.org` uses well know Let's encrypt certificates.

baldhead69, 2020-09-18:
Hi @david-cermak ,

I read "this note".

"What kind of CA did you use with your example?"
A: I only generated a new certificate using: 
openssl req -newkey rsa:2048 -nodes -keyout prvtkey.pem -x509 -days 3650 -out cacert.pem -subj "/CN=ESP32 HTTPS server example"

I read about "Let's encrypt certificates".

I am new with that, so i'm really lost.

From what I understand, I need to have a domain of my own, register this domain in "yaler" for example, only then i could generate a key on the "Let's encrypt" site.

Thank's.

baldhead69, 2020-09-19:
Hi @david-cermak,

Now i have the problem in the image in annex.

I think that is a okhttp3 lib error.

[image]

baldhead69, 2020-09-19:
Hi @david-cermak ,

Wouldn't you be able to generate a local certificate ?

Perhaps espressif would be interested in making a script to help generate this certificate for esp32 https server using this link:

https://letsencrypt.org/pt-br/docs/certificates-for-localhost/

I think that it need has a hostname, it would not be possible to generate the certificate by the ip, at least i think that the okhttp3 lib does not accept.

I already tried to use mdns with esp32 on my local network and I had problems.

baldhead69, 2020-09-24:
Hi @david-cermak ,

Some suggestion on hostname not verified error ?

Thank's.

baldhead69, 2020-09-26:
Hi @david-cermak ,

I solved the problem partially in the link: "https://github.com/square/okhttp/issues/6276"