espressif/esp-idf #5734: idf Build failed with LWIP_IRAM_OPTIMIZATION

LXY1226, 2020-08-15:
## Environment
|Development Kit|other [WT32-SC01]|
|                               -:|:-                              |
| Kit version                     | N/A                            |
| Module or chip used             | [ESP32-WROVER-B]               |
| IDF version                     | v4.1-rc                        |
| Build System                    | idf.py                         |
| Compiler version                | (crosstool-NG esp-2020r2) 8.2.0|
| Operating System                | Windows                        |
| (Windows only) environment type | Plain Command Prompt.          |
| Using an IDE?                   | Yes [CLion with ESP Plugin]    |
| Power Supply                    | USB                            |

## Problem Description
`esp-idf/tools/ldgen/ldgen.py` Fails while` LWIP_IRAM_OPTIMIZATION = y`;

## Build Log

> FAILED: esp-idf/esp32/ld/esp32.project.ld
<details>
<summary>cmdline</summary>

> cmd.exe /C "cd /D C:\Users\*User*\CLionProjects\esp\build\esp-idf\esp32 && C:\Users\*User*\.espressif\python_env\idf4.1_py3.8_env\Scripts\python.exe C:/Users/*User*/Code/esp-idf/tools/ldgen/ldgen.py --config C:/Users/*User*/CLionProjects/esp/sdkconfig --fragments C:/Users/*User*/Code/esp-idf/components/xtensa/linker.lf C:/Users/*User*/Code/esp-idf/components/esp_ringbuf/linker.lf   C:/Users/*User*/Code/esp-idf/components/spi_flash/linker.lf        C:/Users/*User*/Code/esp-idf/components/espcoredump/linker.lf   C:/Users/*User*/Code/esp-idf/components/esp32/linker.lf    C:/Users/*User*/Code/esp-idf/components/esp32/ld/esp32_fragments.lf     C:/Users/*User*/Code/esp-idf/components/soc/linker.lf      C:/Users/*User*/Code/esp-idf/components/esp_event/linker.lf     C:/Users/*User*/Code/esp-idf/components/esp_wifi/linker.lf C:/Users/*User*/Code/esp-idf/components/lwip/linker.lf  C:/Users/*User*/Code/esp-idf/components/log/linker.lf      C:/Users/*User*/Code/esp-idf/components/heap/linker.lf  C:/Users/*User*/Code/esp-idf/components/freertos/linker.lf C:/Users/*User*/Code/esp-idf/components/newlib/esp32-spiram-rom-functions-c.lf  C:/Users/*User*/Code/esp-idf/components/newlib/newlib.lf   C:/Users/*User*/Code/esp-idf/components/app_trace/linker.lf     C:/Users/*User*/Code/esp-idf/components/esp_gdbstub/linker.lf --input C:/Users/*User*/Code/esp-idf/components/esp32/ld/esp32.project.ld.in --output C:/Users/*User*/CLionProjects/esp/build/esp-idf/esp32/ld/esp32.project.ld --kconfig C:/Users/*User*/Code/esp-idf/Kconfig --env-file C:/Users/*User*/CLionProjects/esp/build/config.env --libraries-file C:/Users/*User*/CLionProjects/esp/build/ldgen_libraries --objdump C:/Users/*User*/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-objdump.exe"
</details>  

> Traceback (most recent call last):
  File "C:/Users/*User*/Code/esp-idf/tools/ldgen/ldgen.py", line 158, in <module>
    main()
  File "C:/Users/*User*/Code/esp-idf/tools/ldgen/ldgen.py", line 134, in main
    mapping_rules = generation_model.generate_rules(sections_infos)
  File "C:/Users/*User*/Code/esp-idf/tools/ldgen/generation.py", line 352, in generate_rules
    self._create_exclusions(mapping_rules, default_rules, sections_infos)
  File "C:/Users/*User*/Code/esp-idf/tools/ldgen/generation.py", line 451, in _create_exclusions
    general_rule.add_exclusion(specific_rule, sections_info)
  File "C:/Users/*User*/Code/esp-idf/tools/ldgen/generation.py", line 100, in add_exclusion
    expansions = do_section_expansion(self, section)
  File "C:/Users/*User*/Code/esp-idf/tools/ldgen/generation.py", line 84, in do_section_expansion
    **expansions = fnmatch.filter(sections_in_obj, section)** [Debugger Screenshot]
  File "C:/Users/*User*/AppData/Local/Programs/Python/Python38/lib/fnmatch.py", line 59, in filter
    for name in names:
TypeError: 'NoneType' object is not iterable
ninja: build stopped: subcommand failed.
ninja failed with exit code 1  

### Steps to reproduce

1. make a copy of Helloworld
2. Tagged `(Top) → Component config → LWIP → Enable LWIP IRAM optimization`

### sdkconfig
[sdkconfig](https://github.com/espressif/esp-idf/files/5078041/sdkconfig.log)
#### Python Debugger Screenshot
[image]

renzbagaporo, 2020-08-27:
In this case it seems that `ethernetif.c` is not being compiled with your application, since CONFIG_ETH_ENABLED is missing from your config. To enable, you have to enable one of the three:

```
CONFIG_ETH_USE_ESP32_EMAC
CONFIG_ETH_USE_SPI_ETHERNET
CONFIG_ETH_USE_OPENETH
```

These seem to be undefined in the config you posted.

m-bruck, 2020-08-28:
But for a WiFi only device this includes unnecessary drivers just to allow the LWIP IRAM optimization?

> In this case it seems that `ethernetif.c` is not being compiled with your application, since CONFIG_ETH_ENABLED is missing from your config. To enable, you have to enable one of the three:
>
> ``` [...]

ESP-Marius, 2022-08-31:
Closing the issue as this does not seem to be a problem anymore on the latest IDF master.

closed 2022-08-31 by ESP-Marius