espressif/esp-idf #5735: ledc_set_pin() function within driver/ledc.h does not reset the GPIO number

albahmed23, 2020-08-15:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it):
    // v4.0.1
- Build System:         idf.py (CMake)
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 8.2.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes: VSC
- Power Supply:         USB

## Problem Description

When you call the ledc_set_pin() function. The function would not reset the GPIO pin number for the new output of the LEDC.

### Expected Behavior

The function should have reset the pin.

### Actual Behavior

The function adds the GPIO to the channel and does not reset the one already on there.

### Steps to reproduce

1. Just run the ledc_set_pin() after you initialize the ledc components 

### Code to reproduce this issue

N/A

Alvin1Zhang, 2020-08-17:
Thanks for reporting, we will look into.

ginkgm, 2021-07-05:
Hi @albahmed23 ,

This is expected behavior. With this flexibility, developers can make some interesting products. If you don't want this, you can call `gpio_reset_pin` to return the IO to a known state.

albahmed23, 2021-07-05:
Thanks @ginkgm for your prompt help. That makes sense.

closed 2021-07-05 by albahmed23