espressif/esp-idf #5736: ESP32 OTA issue

Alex-Smart, 2020-08-16:
Found these two closed issues (similar to what Ive got), but no solution
https://github.com/espressif/esp-idf/issues/5240
https://github.com/espressif/esp-idf/issues/5401

firmware file is accessible, checked it via browser
Ive launched advanced_https_ota example, here is console output:
```
I (3727) advanced_https_ota_example: Starting Advanced OTA example
I (5357) esp_https_ota: Starting OTA...
I (5367) esp_https_ota: Writing to partition subtype 16 at offset 0x110000
D (5367) HTTP_CLIENT: esp_transport_read returned:-1 and errno:128 
E (5367) esp_https_ota: Connection closed, errno = 128
E (5377) esp_https_ota: Complete headers were not received
E (5377) advanced_https_ota_example: esp_https_ota_read_img_desc failed
E (5387) advanced_https_ota_example: ESP_HTTPS_OTA upgrade failed 0
```

ESP-IDF version

> v4.2-dev-1126-gd85d3d969

ESP tools

> * xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
>   - esp-2020r1-8.2.0 (recommended, installed)
> * xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
>   - esp-2020r1-8.2.0 (recommended, installed)
> * esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
>   - 2.28.51-esp-20191205 (recommended, installed)
> * esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
>   - 2.28.51-esp-20191205 (recommended, installed)
> * cmake: CMake build system
>   - 3.16.4 (recommended, installed)
> * openocd-esp32: OpenOCD for ESP32
>   - v0.10.0-esp32-20200406 (recommended, installed)
> * ninja: Ninja build system
>   - 1.10.0 (recommended, installed)
> * idf-exe: IDF wrapper tool for Windows
>   - 1.0.1 (recommended, installed)
> * ccache: Ccache (compiler cache)
>   - 3.7 (recommended, installed)
> 

HTTPS server side

>   Operating System: Debian GNU/Linux 9 (stretch)
>   Kernel: Linux 4.9.0-6-amd64
>   openssl:   Installed: 1.1.0l-1~deb9u1

Alvin1Zhang, 2020-08-17:
Thanks for reporting, we will look into.

bill1davis, 2020-08-17:
OTA is completely broken for me in this branch, was working fine in 3.3.  Both simple and advanced methods are returning the following.  I am loading a relatively large binary, about 2MB.  AWS server, accessible binary.

E (135900) TRANS_SSL: esp_tls_conn_read error, errno=No more processes
W (135900) HTTP_CLIENT: esp_transport_read returned:-26880 and errno:11 
I (136900) wifi:bcn_timout,ap_probe_send_start
I (146220) wifi:bcn_timout,ap_probe_send_start
I (155340) wifi:bcn_timout,ap_probe_send_start
I (164450) wifi:bcn_timout,ap_probe_send_start

shubhamkulkarni97, 2020-08-18:
Hello @Alex-Smart, can you check if any of the suggestions mentioned in  https://github.com/espressif/esp-idf/issues/5240#issuecomment-639292839 work for you?

Hi @bill1davis,
Are you observing issues only with large binary or with small size binaries as well? Can you check if you are observing issues with small binary (e.g. `hello_world` binary)? If yes, can you provide logs setting log level to debug?

Alex-Smart, 2020-08-18:
@shubhamkulkarni97, as you can see in my original post, I saw  [#5240 (comment)](https://github.com/espressif/esp-idf/issues/5240#issuecomment-639292839) , it didnt help...

bill1davis, 2020-08-18:
@shubhamkulkarni97
OTA has been working all day today for me.  I can't really explain it.  Same function calls, same server.
The only clue that something might be amiss is the task watchdog fires.  I was unable to set log to debug.  so much i/o that the system rebooted.

I (33701) esp_https_ota: Starting OTA...
I (33701) esp_https_ota: Writing to partition subtype 16 at offset 0x410000
E (39851) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (39891) task_wdt:  - IDLE0 (CPU 0)
E (39891) task_wdt: Tasks currently running:
E (39891) task_wdt: CPU 0: ipc0
E (39891) task_wdt: CPU 1: IDLE1
I (42981) OTA: bytes read 578

shubhamkulkarni97, 2020-08-20:
@bill1davis, Task watchdog fires while erasing big partitions if `CONFIG_SPI_FLASH_YIELD_DURING_ERASE` is not enabled. Can you try enabling this config option in menuconfig and check if it fixes your issue?

@Alex-Smart Can you share logs setting log level to debug?

Steps to enable debug logs:
* Run `idf.py menuconfig`
* Go to `Component Config > Log Output > Default log verbosity`
* Select Debug

This will help us debug your issue.

bill1davis, 2020-08-20:
@shubhamkulkarni97

I don't see CONFIG_SPI_FLASH_YIELD_DURING_ERASE in menuconfig???

shubhamkulkarni97, 2020-08-21:
@bill1davis, 

Can you paste output of `git describe --tags`?

This config option was added in https://github.com/espressif/esp-idf/commit/3cb655eaabd136700071c9225f33715ea4915917 commit, please check if this change is included in your branch. If not, you will have to upgrade your branch to latest commit.

If it is not possible to update the branch with latest changes, you can cherry-pick above mentioned commit to fix the issue.

bill1davis, 2020-08-24:
@shubhamkulkarni97

I don't see the issue in the 4.1 release.  I think I was mistakenly testing against 4.1-beta2 instead of 4.1-rc.

Alvin1Zhang, 2020-08-25:
Thanks for reporting and glad the issue didn't appear. Feel free to reopen if the issue still happens. Thanks.

closed 2020-08-25 by Alvin1Zhang

GKFernandes, 2021-11-04:
Hi everyone,

I am currently running into this same issue.

I have basically implemented the code in https://github.com/espressif/esp-idf/tree/master/examples/system/ota/advanced_https_ota and I host all of my firmware binaries inside amazon's S3 service.

The thing is with some binaries the OTA function works just fine and with some other ones it doesn't, so I'd like some to sort out the issue.

I've followed @shubhamkulkarni97 advice on other posts about how to set up the server certificate and that it enabled me to work with S3, so I don't think it's server related.

I wonder if this is related to this issue: https://github.com/espressif/esp-idf/issues/6165

My debug output is:

I (179023) OTA: Iniciando OTA
D (179023) HTTP_CLIENT: Begin connect to: https://gkzinotabucket.s3.amazonaws.com:443
D (179023) esp-tls: host:gkzinotabucket.s3.amazonaws.com: strlen 31
D (179063) esp-tls: Enable TCP keep alive. idle: 5, interval: 5, count: 3
D (179233) esp-tls: handshake in progress...
D (181663) HTTP_CLIENT: Write header[3]: GET /RBTEST.bin HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: gkzinotabucket.s3.amazonaws.com
Content-Length: 0

D (182023) HTTP_CLIENT: on_message_begin
D (182023) HTTP_CLIENT: HEADER=x-amz-request-id:PVW09A38SPM6HV1E
D (182033) HTTP_CLIENT: HEADER=x-amz-id-2:FaowKFcR8jh+vMUlMbXkB6APaZ/v5ogcGMnQecvOhuUWbb8NdF0+WxoPrK/YXripD5eHmsPZeDI=
D (182033) HTTP_CLIENT: HEADER=Content-Type:application/xml
D (182043) HTTP_CLIENT: HEADER=Transfer-Encoding:chunked
D (182043) HTTP_CLIENT: HEADER=Date:Thu, 04 Nov 2021 18:44:33 GMT
D (182053) HTTP_CLIENT: HEADER=Server:AmazonS3
D (182063) HTTP_CLIENT: http_on_headers_complete, status=403, offset=266, nread=266
D (182063) HTTP_CLIENT: content_length = -1
I (182073) esp_https_ota: Starting OTA...
I (182073) esp_https_ota: Writing to partition subtype 17 at offset 0x160000
D (182083) HTTP_CLIENT: Chunks were not completely read
D (182093) HTTP_CLIENT: is_data_remain=1, is_chunked=1, content_length=-1
D (182093) HTTP_CLIENT: need_read=289, byte_to_read=289, rlen=254, ridx=0
D (182103) HTTP_CLIENT: http_on_body 243
D (182103) HTTP_CLIENT: http_on_chunk_complete
D (182113) HTTP_CLIENT: http_on_chunk_complete
D (182113) HTTP_CLIENT: http_on_message_complete, parser=3ffdf814
D (182123) HTTP_CLIENT: is_data_remain=0, is_chunked=1, content_length=-1
E (182123) esp_https_ota: Complete headers were not received
E (182133) OTA: esp_https_ota_read_img_desc failed
E (182143) OTA: ESP_HTTPS_OTA upgrade failed 0