espressif/esp-idf #5737: IDF4.2 : PPPOS_Client : in the main() function : issue in esp_modem_stop_ppp() : reason "receive_cb_ctx" not initialized

mmggbe, 2020-08-16:
Dear,

I have an issue with the example PPPOS_Client when sending a SMS with a SIM800l
Api version is : ESP-IDF v4.2-dev-1995-g636b964c8-dirty

At the end of main function of  ppos_client_main(), the function esp_modem_stop_ppp() crashes

```
void app_main(void)
{
...
    esp_mqtt_client_handle_t mqtt_client = esp_mqtt_client_init(&mqtt_config);
    esp_mqtt_client_start(mqtt_client);
    xEventGroupWaitBits(event_group, GOT_DATA_BIT, pdTRUE, pdTRUE, portMAX_DELAY);
    esp_mqtt_client_destroy(mqtt_client);
    /* Exit PPP mode */
    **ESP_ERROR_CHECK(esp_modem_stop_ppp(dte));**

....
}

```
The reason is in function :

```
static void esp_handle_uart_data(esp_modem_dte_t *esp_dte)
{

     if (esp_dte->parent.dce->mode != MODEM_PPP_MODE) {
 ...  
    if (length) {
        esp_dte->receive_cb(esp_dte->buffer, length, esp_dte->**receive_cb_ctx**);
     }
 }
```

The issue is that **receive_cb_ctx** ins not initialized.
Any Idea wher I shoudl look for?
Thanks

Marc

david-cermak, 2020-08-19:
Hi @mmggbe 

Thanks for reporting this issue. There seems to be a problem with switching back to PPP mode in some cases. Just to make sure this is the same issue you reported could you please answer the below questions?

Could you please post full error message/log from your board?
Does this happen intermittently or always?
Why do you think the issue and the reason for that is in `receive_cb_ctx` not initialised in `esp_handle_uart_data()`

mmggbe, 2020-08-20:
HI,
concerning the "Why do you think the issue and the reason for that is in receive_cb_ctx not initialised in esp_handle_uart_data()", I saw it with openocd & eclipse when looking to the variable just before the crash. But now, I cannot reproduce it, but it crashes always between line 169 & 171 below.

The modified code with additional trace is :

static void esp_handle_uart_data(esp_modem_dte_t *esp_dte)
{

    ESP_LOGI(MODEM_TAG, "%s(%d): ",__FUNCTION__, __LINE__);		//MaGe

   	if (esp_dte->parent.dce->mode != MODEM_PPP_MODE) {
        ESP_LOGE(MODEM_TAG, "Error: Got data event in PPP mode");
        /* pattern detection mode -> ignore date event on uart
         * (should never happen, but if it does, we could still
         * read the valid data once pattern detect event fired) */
        return;
    }
    ESP_LOGI(MODEM_TAG, "%s(%d): ",__FUNCTION__, __LINE__); //MaGe

    size_t length = 0;
    uart_get_buffered_data_len(esp_dte->uart_port, &length);

    length = MIN(esp_dte->line_buffer_size, length);
    length = uart_read_bytes(esp_dte->uart_port, esp_dte->buffer, length, portMAX_DELAY);
    /* pass the input data to configured callback */

    ESP_LOGI(MODEM_TAG, "%s(%d): ",__FUNCTION__, __LINE__); //MaGe

    if (length) {
 Line 167:        ESP_LOGI(MODEM_TAG, "%s(%d): ",__FUNCTION__, __LINE__); //MaGe
        esp_dte->receive_cb(esp_dte->buffer, length, esp_dte->receive_cb_ctx);
 Line 169:       ESP_LOGI(MODEM_TAG, "%s(%d): ",__FUNCTION__, __LINE__); //MaGe
    }
  Line 171:  ESP_LOGI(MODEM_TAG, "%s(%d): ",__FUNCTION__, __LINE__); //MaGe

}

The trace is : 
I (15584) esp-modem: esp_handle_uart_data(146): 
I (15584) esp-modem: esp_handle_uart_data(155): 
I (15584) esp-modem: esp_handle_uart_data(164): 
I (15584) esp-modem: esp_handle_uart_data(167): 
Then crash with no more info excepted a 0x0 file opened by openocd

or : 
I (18293) pppos_example: Modem PPP Stopped
I (18293) esp-modem: esp_modem_stop_ppp(545): 
I (18303) esp-modem: esp_modem_dte_change_mode(341): 
I (18303) esp-modem: esp_modem_dte_change_mode(354): ,  Port= 1:
I (18313) esp-modem: esp_modem_dte_change_mode(357): 
I (18313) esp-modem: esp_modem_dte_change_mode(362): 
I (18323) sim800: sim800_set_working_mode(303): 
I (18323) sim800: sim800_set_working_mode(308): 
I (18333) esp-modem: UART cmd: +++
I (18343) esp-modem: Received data: 0
I (18343) esp-modem: esp_handle_uart_data(146): 
I (18343) esp-modem: esp_handle_uart_data(155): 
I (18353) esp-modem: esp_handle_uart_data(164): 
I (18353) esp-modem: esp_handle_uart_data(171): 
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2

I continue to investigate to get a good traces of the local variables before the crash

mmggbe, 2020-08-20:
Hi,
I added in "static void esp_handle_uart_data(esp_modem_dte_t *esp_dte)" :
    **ESP_LOGI(MODEM_TAG, "%s(%d), ptr: buf=%p txr=%p: ",__FUNCTION__, __LINE__, esp_dte->buffer, esp_dte->receive_cb_ctx);**
but never a Null pointer...  
Therfore I did probably something wrong with GDB... I test now without GDB
Sorry for the confusion

Below, the end of the trace : it seems the answer of the modem is not seen by the uart...  I will investigate why during the WE
I (18324) pppos_example: MQTT_EVENT_DATA
I (18324) esp-modem: esp_handle_uart_data(171): 
**TOPIC=/topic/esp-pppos
DATA=esp32-pppos**
I (18384) esp-modem: Received data: 0
I (18384) esp-modem: esp_handle_uart_data(146): 
I (18384) esp-modem: esp_handle_uart_data(155): 
I (18384) esp-modem: esp_handle_uart_data(164), ptr: buf=0x3ffbb31c txr=0x3ffbce8c: 
I (18394) esp-modem: esp_handle_uart_data(171): 
I (18444) esp-modem: Received data: 0
I (18444) esp-modem: esp_handle_uart_data(146): 
I (18444) esp-modem: esp_handle_uart_data(155): 
I (18444) esp-modem: esp_handle_uart_data(164), ptr: buf=0x3ffbb31c txr=0x3ffbce8c: 
I (18454) esp-modem: esp_handle_uart_data(171): 
I (18504) esp-modem: Received data: 0
I (18504) esp-modem: esp_handle_uart_data(146): 
I (18504) esp-modem: esp_handle_uart_data(155): 
I (18504) esp-modem: esp_handle_uart_data(164), ptr: buf=0x3ffbb31c txr=0x3ffbce8c: 
I (18514) esp-modem: esp_handle_uart_data(167): 
I (18514) esp-modem: esp_handle_uart_data(169): 
I (18524) esp-modem: esp_handle_uart_data(171): 
I (18574) esp-modem: Received data: 0
I (18574) esp-modem: esp_handle_uart_data(146): 
I (18574) esp-modem: esp_handle_uart_data(155): 
I (18574) esp-modem: esp_handle_uart_data(164), ptr: buf=0x3ffbb31c txr=0x3ffbce8c: 
I (18584) esp-modem: esp_handle_uart_data(167): 
I (18584) esp-modem: esp_handle_uart_data(169): 
I (18594) esp-modem: esp_modem_stop_ppp(537): 
I (18594) esp-modem: esp_handle_uart_data(171): 
I (18604) pppos_example: Modem PPP Stopped
I (18614) esp-modem: esp_modem_stop_ppp(545): 
I (18614) esp-modem: esp_modem_dte_change_mode(341): 
I (18614) esp-modem: esp_modem_dte_change_mode(354): ,  Port= 1:
I (18624) esp-modem: esp_modem_dte_change_mode(357): 
I (18634) esp-modem: esp_modem_dte_change_mode(362): 
I (18634) sim800: sim800_set_working_mode(303): 
I (18644) sim800: sim800_set_working_mode(308): 
I (18644) esp-modem: UART cmd: +++
I (18664) esp-modem: Received data: 0
I (18664) esp-modem: esp_handle_uart_data(146): 
I (18664) esp-modem: esp_handle_uart_data(155): 
I (18664) esp-modem: esp_handle_uart_data(164), ptr: buf=0x3ffbb31c txr=0x3ffbce8c: 
I (18674) esp-modem: esp_handle_uart_data(171): 
I (18734) esp-modem: Received data: 0
I (18734) esp-modem: esp_handle_uart_data(146): 
I (18734) esp-modem: esp_handle_uart_data(155): 
I (18734) esp-modem: esp_handle_uart_data(164), ptr: buf=0x3ffbb31c txr=0x3ffbce8c: 
I (18744) esp-modem: esp_handle_uart_data(171): 
I (18794) esp-modem: Received data: 0
I (18794) esp-modem: esp_handle_uart_data(146): 
I (18794) esp-modem: esp_handle_uart_data(155): 
I (18794) esp-modem: esp_handle_uart_data(164), ptr: buf=0x3ffbb31c txr=0x3ffbce8c: 
I (18804) esp-modem: esp_handle_uart_data(167): 
I (18804) esp-modem: esp_handle_uart_data(169): 
I (18814) esp-modem: esp_handle_uart_data(171): 
I (18864) esp-modem: Received data: 0
I (18864) esp-modem: esp_handle_uart_data(146): 
I (18864) esp-modem: esp_handle_uart_data(155): 
I (18864) esp-modem: esp_handle_uart_data(164), ptr: buf=0x3ffbb31c txr=0x3ffbce8c: 
I (18874) esp-modem: esp_handle_uart_data(171): 
I (18924) esp-modem: Received data: 0
I (18924) esp-modem: esp_handle_uart_data(146): 
[... 121 lines cut ...]
I (20364) esp-modem: esp_handle_uart_data(164), ptr: buf=0x3ffbb31c txr=0x3ffbce8c: 
I (20374) esp-modem: esp_handle_uart_data(171): 
I (20424) esp-modem: Received data: 0
I (20424) esp-modem: esp_handle_uart_data(146): 
I (20424) esp-modem: esp_handle_uart_data(155): 
I (20424) esp-modem: esp_handle_uart_data(164), ptr: buf=0x3ffbb31c txr=0x3ffbce8c: 
I (20434) esp-modem: esp_handle_uart_data(171): 
I (20484) esp-modem: Received data: 7
W (20484) esp-modem: Pattern Queue Size too small
I (20534) esp-modem: Received data: 0
I (20534) esp-modem: esp_handle_uart_data(146): 
I (20534) esp-modem: esp_handle_uart_data(155): 
I (20534) esp-modem: esp_handle_uart_data(164), ptr: buf=0x3ffbb31c txr=0x3ffbce8c: 
I (20544) esp-modem: esp_handle_uart_data(171): 
E (21654) esp-modem: esp_modem_dte_send_cmd(252): process command timeout
E (21654) sim800: sim800_set_working_mode(310): send command failed
E (21654) esp-modem: esp_modem_dte_change_mode(364): set new working mode:0 failed
E (21664) esp-modem: esp_modem_stop_ppp(547): enter command mode failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400844dc
file: "../main/pppos_client_main.c" line 481
func: app_main
expression: esp_modem_stop_ppp(dte)

abort() was called at PC 0x400844df on core 0
Setting breakpoint at 0x4008568e and returning...

mmggbe, 2020-09-24:
Dear,

Some more update :
The function esp_modem_stop_ppp(dte) send a +++  and expects a NO CARRIER
The function esp_modem_stop_ppp(dte) calls sim800_set_working_mode() which configure sim800_handle_exit_data_mode() as handler.
The issue is that the wrong handler is called by esp_handle_uart_data() which is still in ppp mode and the sim800_handle_exit_data_mode() handler in never called.
Its why the function sim800_set_working_mode() never receives the NO CARRIER and return a time out.

Alvin1Zhang, 2020-11-06:
Thanks for reporting and sorry for slow turnaround. We have a fix under internal reviewing. The issue will be closed automatically once the fix is available on GitHub.

closed 2021-01-05 by espressif-bot