espressif/esp-idf #5739: LOG_FORMAT macro fails with -Wformat=2

m-bruck, 2020-08-17:
- Development Kit: none
- Module or chip used: ESP32
- IDF version: v4.1-rc
- Build System: CMake
- Compiler version: 2020r2
- Operating System: Windows, ESP Command Prompt
- Using an IDE?: No

The LOG_FORMAT macro uses %d (instead of %u or PRIu32) to print the output of `uint32_t esp_log_timestamp()`. This triggers a warning with -Wformat=2 (this is with c++).

.../esp-idf-v4.1-rc/components/log/include/esp_log.h:255:27: error: format '%d' expects argument of type 'int', but argument 4 has type 'uint32_t' {aka 'unsigned int'} [-Werror=format=]

KonstantinKondrashov, 2020-08-18:
Hi @m-bruck!
Thanks for report about that I will fix it on the master and backport it. I will replace '%d' in  LOG_FORMAT to `%u`.

mahavirj, 2021-03-12:
@m-bruck Fixed with https://github.com/espressif/esp-idf/commit/4567b6e866a9dd7d6b4888814d50f02d6115f79d

closed 2021-03-12 by mahavirj