espressif/esp-idf #5741: undefined reference to xQueueGenericCreateStatic

kayleejacob1, 2020-08-17:
I am using the latest IDF in Visual Studio with visual micro 

I have set 
#define configSUPPORT_STATIC_ALLOCATION 1
in FreeRTOSconfig.h

I am using this so that I can allocate memory to the PSRAM.

Code below
TaskHandle_t loopi2cread;
QueueHandle_t readi2cdata;
StaticQueue_t xQueuereadi2cBuffer;
 uint8_t* powerque = (uint8_t*)ps_calloc(2, sizeof(powerdata));
 readi2cdata = xQueueCreateStatic(1, sizeof(powerdata), powerque,&xQueuereadi2cBuffer);

Alvin1Zhang, 2020-08-18:
@kayleejacob1 Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

kayleejacob1, 2020-08-18:
I have discovered through more research that I have to recompile idf for
arduino.  But I am having issues doing that in windows I have followed all
steps but seems that it does not save the file anywhere

Alvin1Zhang, 2020-08-18:
Same with post https://esp32.com/viewtopic.php?f=13&p=63992#p63992

kayleejacob1, 2020-08-18:
yes that is me

Alvin1Zhang, 2020-09-02:
@kayleejacob1 Thanks for reporting, there is already a reply in the [post](https://esp32.com/viewtopic.php?f=13&p=63992#p63992), will close this ticket and let us track there. Thanks.

closed 2020-09-02 by Alvin1Zhang