espressif/esp-idf #5746: Error in UART examples

phildimond, 2020-08-18:
The UART example code for ESP32 has an error. The driver install must come after the configuration. 

For example in the esp-idf/examples/peripherals/uart/uart_echo/ example, the code below in uart_echo_example_main.c will not work if the UART is changed to UART_NUM_2 *unless* the uart_driver_install is moved after the uart_set_pin. 

    /* Configure parameters of an UART driver,
     * communication pins and install the driver */
    uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity    = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        .source_clk = UART_SCLK_APB,
    };
    uart_driver_install(UART_NUM_1, BUF_SIZE * 2, 0, 0, NULL, 0);
    uart_param_config(UART_NUM_1, &uart_config);
    uart_set_pin(UART_NUM_1, ECHO_TEST_TXD, ECHO_TEST_RXD, ECHO_TEST_RTS, ECHO_TEST_CTS);

alisitsyn, 2020-11-06:
Hello @phildimond,

Thank you for the issue.

Could you please provide more information:
1. Which ESP-IDF version you are using?
2. Could you send me the complete code of your example?
3. What kind of board (ESP chip) is used?

I was trying to reproduce the issue with my ESP-WROOM32 board and the echo example from latest master and with release/v4.1. It works correctly with the UART_NUM_2 as well with UART_NUM_1 on my side and is able to return echo. I used the code you provided for initialization of UART as well as original code.

phildimond, 2020-11-10:
ESP 4.1

Code was the example as quoted. I was having trouble with my code so I went back to the example and had the same issue with it, until I swapped the order of the calls.

Devkit-C (new with V3 ESP-WROOM-32D)

Phillip Dimond
Consulting Engineer, Gaming Systems
Phone: 0409-748433
Intl: +61-409-748433

alisitsyn, 2020-11-10:
@phildimond,

Thank you for the information. Generally,  order of the calling uart_param_config and uart_driver_install can be arbitrary. May be I am missing something but can not reproduce it on my Windows and Linux machine with the DevKitC ESP-WROOM-32D_V4 (what I have now) on release/v4.1. In order to fix this I need to confirm the issue and find reason for this. Could you send me your elf and binary file with the sdkconfig.

Thanks.

fromeijn, 2020-12-17:
I'm seeing the same issue when using the `pppos_client` example and using `UART_NUM_2` not sure what the deal is there. UART is not behaving and i'm missing data etc. I'm on [v4.2](https://github.com/espressif/esp-idf/tree/release/v4.2). Chip used for testing is `ESP-WROOM-32U`.  Using `UART_NUM_1` works fine.

AxelLin, 2020-12-18:
> @phildimond,
> 
> Thank you for the information. Generally, order of the calling uart_param_config and uart_driver_install can be arbitrary. 

Are you sure?
Q1: What is the expected driver behavior before calling correct uart_param_config?
Q2: What is the expected driver behavior before calling uart_set_pin?

IMHO, the init order does matter, e.g. 0784426d6b6a0d1e97a7594493b2c80e82070404

AxelLin, 2023-05-07:
> @phildimond,
> 
> Thank you for the information. Generally, order of the calling uart_param_config and uart_driver_install can be arbitrary. 

https://github.com/espressif/esp-idf/issues/6589 show the order does matter.