espressif/esp-idf #5747: corrupted nvs key partition

achraf-boussaada, 2020-08-18:
## Environment

- Development Kit:    ESP32-DevKitC
- Kit version (DevKitC): [v1|v2|v3|v4] - I'm not sure about the version
- Module or chip used:  ESP32-WROOM-32
- IDF version : ESP-IDF v4.3-dev-771-gc77c4ccf6
- Build System:         idf.py
- Compiler version : (crosstool-NG esp-2020r2) 8.2.0
- Operating System:    Linux
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: yes, visual code studio
- Power Supply:        USB

## Problem Description
 I'm having a problem with the flash encryption/nvs encryption. Here's a summary of what I did until now:
1- I enabled flash encryption by doing the following under `idf.py menuconfig`:

Security features --> Enable flash encryption on boot (Development mode)
Bootloader config --> Bootloader log verbosity (Info)
Partition table --> changed offset of partition table from 0x8000 to 0x9000
2- I enabled nvs encryption alongside a custom partition table by doing the following under `idf.py menuconfig`:

Partition table --> Custom partition table CSV
Component config --> NVS --> Enable NVS encryption
this is my custom partition table:
```
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,      data, nvs,     ,        0x6000,
custom_nvs,data,nvs,,0x3000,
phy_init, data, phy,     ,        0x1000,
factory,  app,  factory, ,        1M,
private_keys,data,nvs_keys,,0x1000,encrypted,
```
After using the NVS partition generator here's the output of `idf.py partition_table`:
```
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0xa000,24K,
custom_nvs,data,nvs,0x10000,12K,
phy_init,data,phy,0x13000,4K,
factory,app,factory,0x20000,1M,
private_keys,data,nvs_keys,0x120000,12K,encrypted
```
I used the offsets to flash the content of both partitions to the chip with the esptool.py. I set two flags` --before default_reset` `--after no_reset` and the command `write_flash`.
I checked my efuse table and according to it, the flash encrypted is correctly enabled (I double checked programmatically with `esp_flash_encryption_enabled()`).

here's some relevant info from my efuse table:
```
Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                Flash encryption mode counter                      = 1 R/W (0b0000001)
FLASH_CRYPT_CONFIG (BLOCK0):             Flash encryption config (key tweak bits)           = 15 R/W (0xf)
BLK1 (BLOCK1):                           Flash encryption key                              
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-
```
Finally after executing `idf.py encrypted-flash monitor` the following code snippet throws this: `Failed to read nvs keys. Error: ESP_ERR_NVS_CORRUPT_KEY_PART`:
```
// find partition with nvs_keys
    const esp_partition_t *partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA,
                                                                ESP_PARTITION_SUBTYPE_DATA_NVS_KEYS,
                                                                KEYS_PARTITION_LABEL);

    if (partition == NULL)
    {
        printf("Could not locate nvs_key partition. Aborting.");
    }

    // read nvs_keys from key partition
    nvs_sec_cfg_t cfg;
    esp_err_t err = nvs_flash_read_security_cfg(partition, &cfg);
    if (err != ESP_OK)
    {
        printf("Failed to read nvs keys. Error: %s", esp_err_to_name(err));
    }
```
I really can't seem to understand why the keys partition is corrupted. Am I missing something?

Alvin1Zhang, 2020-08-19:
Thanks for reporting, we will look into.

0xjakob, 2020-08-19:
Hi @achraf-boussaada. 

Thanks for your interest in NVS encryption on ESP32!

Just to check:
Did you do a plaintext-flash first before you do an encrypted-flash? The encrypted-flash feature is mainly for development purposes when you know the key burnt into the eFuse and don't "lock" the device completely.
You have to do a plaintext flash (ordinary flash) the first time after setting up the eFuse encryption keys and corresponding bits. After the plaintext flash, the bootloader checks the eFuses and finds that encryption is enabled. It then automatically encrypts all partitions marked as encrypted. But it seems like 
As far as I know you don't need any special procedure to flash the nvs encryption keys. Just flashing the whole image should work.

I will check out the encryption process again and come back later.

Best,
Jakob

achraf-boussaada, 2020-08-19:
I tired doing a plaintext-flash first but I got an error, which I reported in an other [issue](https://github.com/espressif/esp-idf/issues/5742). I think the chip was soft-bricked somewhere along the line. 

In my project I tried to disable flash encryption and enable it again and doing the whole process again. This time I tried it with the flash encryption in release mode. A plaintext-worked with the following warning:
```
E (330) flash_encrypt: Flash encryption settings error: app is configured for RELEASE but efuses are set for DEVELOPMENT
E (341) flash_encrypt: Mismatch found in security options in bootloader menuconfig and efuse settings. Device is not secure.
```
Unfortunately, I got the same error as before concerning the corrupted nvs keys partition.

I finally got it to work today with a new chip: **ESP32-WROOM-32D**. I enabled flash encryption in release mode, did all the steps mentioned in the issue, the error was gone and I managed to read data from my encrypted custom nvs partition. Nevertheless I still got a warning/error about the development mode:
```
E (330) flash_encrypt: Flash encryption settings error: app is configured for RELEASE but efuses are set for DEVELOPMENT
E (341) flash_encrypt: Mismatch found in security options in bootloader menuconfig and efuse settings. Device is not secure.
```
here's the relevant efuses on the new chip after enabling flash encryption in release mode:
```
Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                Flash encryption mode counter                      = 127 R/W (0b1111111)
UART_DOWNLOAD_DIS (BLOCK0):              Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):             Flash encryption config (key tweak bits)           = 15 R/W (0xf)
CONSOLE_DEBUG_DISABLE (BLOCK0):          Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                     secure boot enabled for bootloader                 = False R/W (0b0)
ABS_DONE_1 (BLOCK0):                     secure boot abstract 1 locked                      = False R/W (0b0)
JTAG_DISABLE (BLOCK0):                   Disable JTAG                                       = True R/W (0b1)
DISABLE_DL_ENCRYPT (BLOCK0):             Disable flash encryption in UART bootloader        = True R/W (0b1)
DISABLE_DL_DECRYPT (BLOCK0):             Disable flash decryption in UART bootloader        = True R/W (0b1)
DISABLE_DL_CACHE (BLOCK0):               Disable flash cache in UART bootloader             = True R/W (0b1)
BLK1 (BLOCK1):                           Flash encryption key                              
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/- 
```
Now I have couple of questions:

1. I thought by flashing the chip for the first time, the needed efuses will be set automatically for the correspondent mode. Why did I have those errors/warning?

2. Can I switch from release mode to development mode so I can reflash it using `idf.py encrypted-flash`?

3. Is it possible to still make this work on the old chip?

0xjakob, 2020-08-19:
@achraf-boussaada 
1. As far as I remember, no. But the encryption is not my main topic, working mostly on NVS... When I did this I had to set some efuses. But don't know whether there's another way...
2. No, the purpose of the release mode is to lock the device so that at most OTA (over-the-air) updates are possible, if you implement them. If there was a way to revert it, then the chip wouldn't be secure, basically.
3. later...

0xjakob, 2020-08-20:
@achraf-boussaada 
Please keep in mind that in release mode you have a limited number flashes. Only plaintext-flashes are possible which set the `FLASH_CRYPT_CNT` each time, until it's all ones. The device with the eFuses above can't be flashed any more.

Regarding your question 3:
Please send us the eFuses of that chip as you did for the other device. These are the main indicators whether the device is brigged or not.

Best,
Jakob

achraf-boussaada, 2020-08-20:
@0xjakob 
thanks for the tips. Concerning my first question, according to the [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html#release-mode) there are 3 relevant eFuses that will be automatically burnt: `download_dis_encrypt` `download_dis_decrypt` and `FLASH_CRYPT_CNT`. If there are extra eFuses that needs to be manually changed that means the documentation is missing some steps and needs to be updated.

For the third questions here are the eFuses for my first chip **ESP32-WROOM-32**:
```
Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                Flash encryption mode counter                      = 127 R/W (0b1111111)
UART_DOWNLOAD_DIS (BLOCK0):              Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):             Flash encryption config (key tweak bits)           = 15 R/W (0xf)
CONSOLE_DEBUG_DISABLE (BLOCK0):          Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                     secure boot enabled for bootloader                 = False R/W (0b0)
ABS_DONE_1 (BLOCK0):                     secure boot abstract 1 locked                      = False R/W (0b0)
JTAG_DISABLE (BLOCK0):                   Disable JTAG                                       = True R/W (0b1)
DISABLE_DL_ENCRYPT (BLOCK0):             Disable flash encryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_DECRYPT (BLOCK0):             Disable flash decryption in UART bootloader        = True R/W (0b1)
DISABLE_DL_CACHE (BLOCK0):               Disable flash cache in UART bootloader             = True R/W (0b1)
BLK1 (BLOCK1):                           Flash encryption key                              
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/- 
```

0xjakob, 2020-08-20:
@achraf-boussaada Did you use a host-generated or ESP32-generated encryption key?
I *think* since `DISABLE_DL_ENCRYPT` hasn't been activated yet, it should still be possible to flash a new image.

And for flashing, did you use `idf.py encrypted-app-flash` or `idf.py encrypted-flash`?
I remember having an issue where NVS flash encryption wouldn't work, too. In the end, I needed to flash the whole image (`idf.py encrypted-flash`) instead of just the application partition (`idf.py encrypted-app-flash`).

achraf-boussaada, 2020-08-20:
@0xjakob I used an ESP32-generated encryption key with this command: 
```
$IDF_PATH/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py encrypt nvs_private.csv nvs_private.bin 0x3000 --keygen --keyfile private_keys.bin
```
and for flashing I used `idf.py encrypted-flash`. But if I understood correctly I can't flash the chip anymore because I did a second try with the release mode, as I mentioned in my previous comment:

> In my project I tried to disable flash encryption and enable it again and doing the whole process again. This time I tried it with the flash encryption in release mode. A plaintext-worked with the [...]

closed 2020-09-16 by espressif-bot

baamiis, 2021-10-31:
Hi after I upgraded to IDF version v4.3 I am getting an error when accessing NVS.
The error is ESP_ERROR_WRONG_ECRYPTION. My previous IDF version which was working and did not get this error with is V4.2.2. 

After the error I reverted back to version V4.2.2 and all works with no error.

Can someone help me with this thank you.

mahavirj, 2021-11-01:
@baamiis 

Can you please confirm that `CONFIG_NVS_ENCRYPTION` value in `sdkconfig` file is same as it was in v4.2.2 release? 

In v4.3 release, NVS encryption is enabled if flash encryption feature is kept enabled. Please see "Breaking Changes" section in release notes for v4.3 at https://github.com/espressif/esp-idf/releases/tag/v4.3 once.

baamiis, 2021-11-01:
Thanks for the response :)
CONFIG_NVS_ENCRYPTION value in sdkconfig file is the same as it in v4.2.2, as you can see in my attached sdkconfig CONFIG_NVS_ENCRYPTION=yes
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/7452387/sdkconfig.txt)
.

mahavirj, 2021-11-01:
@baamiis 

- Can you please confirm you have "nvs_keys" partition in your partition table? (Ref: https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/api-reference/storage/nvs_flash.html#nvs-key-partition)
- Can you please confirm that your application is using `nvs_flash_secure_init` to initialize NVS partition as it was recommended in v4.2 docs? (Ref: https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/api-reference/storage/nvs_flash.html#encrypted-read-write)

baamiis, 2021-11-01:
@mahavirj 
here is my partition table:
# Name,Type,SubType,Offset,Size,Flags,Comments,
nvs,data,nvs,0x9000,0x4000,encrypted,
otadata,data,ota,0xd000,0x2000,encrypted,
phy_init,data,phy,0xf000,0x1000,
ota_0,app,ota_0,0x10000,0x200000,encrypted,
ota_1,app,ota_1,0x210000,0x200000,encrypted,
test,app,test,0x410000,0x100000,encrypted,
nvs_keys,data,nvs_keys,0x510000,0x1000,encrypted,
user_data,data,nvs,0x511000,0x4000,encrypted,
user_nvs,data,nvs,0x515000,0x269000,encrypted,
user_nvs_1,data,nvs,0x77E000,0x269000,encrypted,
user_nvs_2,data,nvs,0x9E7000,0x269000,encrypted,
user_nvs_3,data,nvs,0xC50000,0x269000,encrypted,
log_spiffs,data,spiffs,0xEB9000,0x147000,

Here is how I am initialysing and accessing the nvs:
user_keys_partition = (esp_partition_t *)esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_NVS_KEYS, NVS_KEYS);
	user_nvs_partition = (esp_partition_t *)esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_NVS, USER_NVS);

	nvs_sec_cfg_t nvs_cfg;
	err = nvs_flash_read_security_cfg(user_keys_partition, &nvs_cfg);
	if (err == ESP_ERR_NVS_KEYS_NOT_INITIALIZED || ESP_ERR_NVS_CORRUPT_KEY_PART == err)
	{
		err = nvs_flash_generate_keys(user_keys_partition, &nvs_cfg);
	}

### 	err = nvs_flash_secure_init_partition(user_nvs_partition->label, &nvs_cfg);  This fuction is returning ESP_ERR_NVS_WRONG_ENCRYPTION 
	if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND)
	{
		printf("NVS INIT ERROR");
	}

this ESP_ERR_NVS_WRONG_ENCRYPTION is not defined in V4.2.2 I think it was add in version V4.3

baamiis, 2021-11-01:
@mahavirj 

the function lookup_nvs_partition is checking if the partion is encrypted and if it is it then return ESP_ERR_NVS_WRONG_ENCRYPTION. only in V4.3

but my nvs_user_partition is encrypted.
esp_err_t lookup_nvs_partition(const char* label, NVSPartition **p)
{
    const esp_partition_t* esp_partition = esp_partition_find_first(
            ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_NVS, label);

    if (esp_partition == nullptr) {
        return ESP_ERR_NOT_FOUND;
    }

    if (esp_partition->encrypted) {
        return ESP_ERR_NVS_WRONG_ENCRYPTION;
    }
...........

mahavirj, 2021-11-01:
@baamiis 

> nvs,data,nvs,0x9000,0x4000,encrypted,

`nvs` partition should not have `encrypted` flag enabled. `nvs` partition gets encrypted using software AES-XTS and its entirely different than flash encryption scheme. Its in-fact `nvs_keys` partition that gets encrypted using flash encryption scheme.

This addition check was introduced in https://github.com/espressif/esp-idf/commit/aca9ec28b3d091a73605ee89ab63f8f76a996491, and it correct in a way where having `encrypted` flag to `nvs` is not recommended.

Is it possible to update partition table here or these are on-field devices and you are pushing out update through OTA?

CC @0xjakob @igrr

baamiis, 2021-11-01:
We have 40 thousands devices in the field and yes we are using OTA :(

igrr, 2021-11-01:
@baamiis in IDF 4.x versions, it might be possible to work around this without changing the partition table, by patching the partition encryption flag at run time, like so:
```c++
    ((esp_partition_t*)esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_NVS, "nvs"))->encrypted = false;
```
early in your `app_main`, before initializing NVS.

This happens to work because the pointers returned by esp_partition_find happen to be in DRAM. In future IDF versions this might be changed, so consider this a workaround just for this case.

baamiis, 2021-11-01:
@igrr 

That is what I was just thinking. as long as that structure element is not read only I am all good.

Guys thank you very much @igrr and @mahavirj for your amazing help :)

baamiis, 2021-11-02:
That worked but I now after waking up from the deep sleep I cannot access the nvs. The function nvs_get_blob is always returning error after the waking up from deep sleep. Again this works well in the v4.2.2.

The error that they nvs_get_blob is returning is: ESP_ERR_NVS_NOT_FOUND.

SgtPepperFTW, 2022-01-26:
So can someone please explain again, how should it work? I have encryption enabled, i have partitions nvs and nvs_key, but I still get (ESP_ERR_NVS_CORRUPT_KEY_PART). The key is generated on the side of chip, so I don't know it. How the code should work with nvs?

0xjakob, 2022-03-04:
@SgtPepperFTW Could you pleaase open a new issue? This issue has been closed, there has been code merged for it. It might be that there is a different issue which also result in `ESP_ERR_NVS_CORRUPT_KEY_PART`. We furthermore need the IDF version and chip you are using (absolutely necessary!).

Actually it should work out-of-the-box once you enabled flash-encryption in menuconfig.