espressif/esp-idf #5749: MQTT uri, username, password setting from std::string doenst work

jarus0, 2020-08-18:
release/v4.0
esp-idf example > https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt/tcp

Below method not working
```
std::string testuri = "someUri";
std::string testusername = "someUsername";
std::string testpassword = "somePassword";

talk.mqtt_cfg.uri = testuri .c_str();
talk.mqtt_cfg.username = testusername .c_str();
talk.mqtt_cfg.password = testpassword .c_str();
```

but by just changing below (directly setting from char array)
```
talk.mqtt_cfg.uri = "someUri";
talk.mqtt_cfg.username = "someUsername";
talk.mqtt_cfg.password = "somePassword";
```

laukik-hase, 2020-08-18:
@jarus0 Can you try this?

```c
strcpy((char *)mqtt_cfg.uri, testuri.c_str());
strcpy((char *)mqtt_cfg.username, testusername.c_str());
strcpy((char *)mqtt_cfg.password, testpassword.c_str());
```

jarus0, 2020-08-19:
Thanks for looking into this 
below are the results.

```
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40001406  PS      : 0x00060530  A0      : 0x8012a53c  A1      : 0x3ffbb560  
A2      : 0x00000000  A3      : 0x3ffcdcd0  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x7474716d  A9      : 0x3ffbb530  
A10     : 0x00000000  A11     : 0x3f414b9c  A12     : 0x00000010  A13     : 0x3ffcdcfc  
A14     : 0x73734e47  A15     : 0x564c706e  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x400013f9  LEND    : 0x4000140d  LCOUNT  : 0xffffffff  

ELF file SHA256: a89cafe8ffb3022a

Backtrace: 0x40001403:0x3ffbb560 0x4012a539:0x3ffbb570 0x4012d26b:0x3ffbb6e0 0x400d08b6:0x3ffbb870 0x40092302:0x3ffbb890
0x4012a539: loadingCredentialsFromFile() at /home/suraj/jarusWork/ComforTick-jEnterprise-ESP32/main/main.h:962

0x4012d26b: app_main at /home/suraj/jarusWork/ComforTick-jEnterprise-ESP32/main/main.cpp:1714

0x400d08b6: main_task at /home/suraj/esp/esp-idf/components/esp32/cpu_start.c:553

0x40092302: vPortTaskWrapper at /home/suraj/esp/esp-idf/components/freertos/port.c:143

Rebooting...

```

ESP-Marius, 2020-08-19:
As @laukik-hase said, `c_str()` is the way to go.

Your new error looks unrelated to this though. Looks like you are trying to write to a NULL pointer in `loadingCredentialsFromFile()`

jarus0, 2020-08-19:
Actully

jarus0, 2020-08-19:
Actully mqtt_cfg.uri and other parameters are define const char*

whenever I try to modify them 
for example, if I do (here "talk" is my class object)

talk.mqtt_cfg.password[2] = 's';

error: assignment of read-only location '*(talk.JMeshWebsocketCommunication::mqtt_cfg.esp_mqtt_client_config_t::password + 2)'
    

I tried this > strcpy, memcpy, strncpy, memmove
also tried > talk.mqtt_cfg.password = (char*)malloc ( password1.length() * sizeof(char) );

but none worked

ESP-Marius, 2020-08-19:
I was a bit quick here, what laukik suggested won't work. `mqtt_cfg.uri`is an uninitialized pointer, you cant copy to it. 

Just doing `mqtt_cfg.uri = testuri .c_str();` should work. You can't modify them after afterwards though, as you say, they are const.

If you want to change it you should point `mqtt_cfg.uri` to a new string.

jarus0, 2020-08-19:
Yes, I know and do as you said, 
but wait I am creating minimal example and creating git repository so you can have complete look.

jarus0, 2020-08-19:
Please have a look the example 
git repository > [](https://github.com/jarus0/mqttCredentialsSettingTest)

Changes that I made in 
1> Convert C example code to C++
2> Change router ssid and password by menuconfig
3> MQTT Credentials assigning from std::string (line num 97 to 109 in main.cpp)

original git repository > https://github.com/espressif/esp-idf/tree/release/v4.1/examples/protocols/mqtt/tcp

Thanks

ESP-Marius, 2020-08-19:
Your issue seems to be unrelated to the `std::string` stuff, try zero initializing your struct ` esp_mqtt_client_config_t mqtt_cfg = {0};`, right now it's filled with random data, which may cause the init call to do all sorts of strange things.

jarus0, 2020-08-19:
Your logic was right struct filled with a random value.
`{0}` didn't work but I find a workaround.

Thanks for your time

closed 2020-08-19 by jarus0