espressif/esp-idf #5751: partitions raw data access

chegewara, 2020-08-18:
Hi,
i couldnt find any API to access partition raw data, by sector and offset. It would be useful with USB MSC.

Thanks

laukik-hase, 2020-08-18:
@chegewara Is this what you are looking for?

[ESP-IDF SPI Flash API: API Reference - Partition Table](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spi_flash.html#api-reference-partition-table)

chegewara, 2020-08-19:
@laukik-hase Thanks, but this API i am already using to OTA update with USB MSC, what i mean is rather raw access to SD/MMC card.

chegewara, 2020-08-20:
I found a way to read/write raw sectors from SD card using arduino-esp32. I am guessing it will be possible with esp-idf too if i could do it with arduino, so issue closed for now, but i hope espressif will add component that will allow this.

closed 2020-08-20 by chegewara