espressif/esp-idf #5754: pppos_client example failed to connect to MQTT broker through BG96 (DCE)

ganeshkumarsw, 2020-08-18:
## Environment

Development Kit: ESP32 WROOM 32 Module
Module or chip used: [ESP32-WROOM-32]
IDF version (idf.py --version): ESP-IDF v4.1
Build System: [idf.py]
Compiler version (run xtensa-esp32-elf-gcc --version to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
Operating System: [Windows]
(Windows only) environment type: [ESP Command Prompt].
Using an IDE?: [NO]
Power Supply: [external 3.3V]

## Problem description 

"connection reset by peer" when trying to connect with an MQTT broker.

## Monitor Log

I (31) boot: ESP-IDF v4.1-dirty 2nd stage bootloader
I (31) boot: compile time 08:41:42
I (31) boot: chip revision: 1
I (40) boot.esp32: SPI Speed      : 40MHz
I (40) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (42) boot: Partition Table:
I (43) boot: ## Label            Usage          Type ST Offset   Length
I (45) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (47) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (49) boot:  2 factory          factory app      00 00 00010000 00100000
I (50) boot: End of partition table
I (52) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x13748 ( 79688) map
I (84) esp_image: segment 1: paddr=0x00023770 vaddr=0x3ffb0000 size=0x021d0 (  8656) load
I (88) esp_image: segment 2: paddr=0x00025948 vaddr=0x40080000 size=0x00404 (  1028) load
I (89) esp_image: segment 3: paddr=0x00025d54 vaddr=0x40080404 size=0x0a2c4 ( 41668) load
I (108) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x49920 (301344) map
I (223) esp_image: segment 5: paddr=0x00079948 vaddr=0x4008a6c8 size=0x00bec (  3052) load
I (231) boot: Loaded app from partition at offset 0x10000
I (231) boot: Disabling RNG early entropy source...
I (231) cpu_start: Pro cpu up.
I (232) cpu_start: Application information:
I (233) cpu_start: Project name:     pppos_client
I (234) cpu_start: App version:      v4.1-dirty
I (236) cpu_start: Compile time:     Aug 18 2020 15:45:11
I (237) cpu_start: ELF file SHA256:  ebf2524e4b47e2ab...
I (239) cpu_start: ESP-IDF:          v4.1-dirty
I (240) cpu_start: Starting app cpu, entry point is 0x400812dc
I (226) cpu_start: App cpu up.
I (243) heap_init: Initializing. RAM available for dynamic allocation:
I (244) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (246) heap_init: At 3FFB3B80 len 0002C480 (177 KiB): DRAM
I (247) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (249) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (251) heap_init: At 4008B2B4 len 00014D4C (83 KiB): IRAM
I (252) cpu_start: Pro cpu start user code
I (263) spi_flash: detected chip: generic
I (263) spi_flash: flash io: dio
I (272) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
ppp phase changed[1]: phase=0
I (276) uart: queue free spaces: 30
E (426) esp-modem: esp_dte_handle_line(85): DTE has not yet bind with DCE
E (476) esp-modem: esp_dte_handle_line(85): DTE has not yet bind with DCE
I (4676) pppos_example: Module: BG96
I (4676) pppos_example: Operator: **Valid Operator**
I (4676) pppos_example: IMEI: **Valid IMEI**
I (4676) pppos_example: IMSI: **Valid IMSI**
I (4876) pppos_example: rssi: 15, ber: 99
I (5076) pppos_example: Battery voltage: 3639 mV
I (5276) pppos_example: Modem PPP Started
[... 414 lines cut ...]
        .port = 1883,
        .keepalive = 3600,
        .event_handle = mqtt_event_handler,
    };
    esp_mqtt_client_handle_t mqtt_client = esp_mqtt_client_init(&mqtt_config);
    esp_mqtt_client_start(mqtt_client);
    xEventGroupWaitBits(event_group, GOT_DATA_BIT, pdTRUE, pdTRUE, portMAX_DELAY);
    esp_mqtt_client_destroy(mqtt_client);
    /* Exit PPP mode */
    ESP_ERROR_CHECK(esp_modem_stop_ppp(dte));
    /* Destroy the netif adapter withe events, which internally frees also the esp-netif instance */
    esp_modem_netif_clear_default_handlers(modem_netif_adapter);
    esp_modem_netif_teardown(modem_netif_adapter);
    xEventGroupWaitBits(event_group, STOP_BIT, pdTRUE, pdTRUE, portMAX_DELAY);
#if CONFIG_EXAMPLE_SEND_MSG
    const char *message = "Welcome to ESP32!";
    ESP_ERROR_CHECK(example_send_message_text(dce, CONFIG_EXAMPLE_SEND_MSG_PEER_PHONE_NUMBER, message));
    ESP_LOGI(TAG, "Send send message [%s] ok", message);
#endif
    /* Power down module */
    ESP_ERROR_CHECK(dce->power_down(dce));
    ESP_LOGI(TAG, "Power down");
    ESP_ERROR_CHECK(dce->deinit(dce));
    ESP_ERROR_CHECK(dte->deinit(dte));
}

Alvin1Zhang, 2020-08-19:
Thanks for reporting, we will look into.

david-cermak, 2020-08-20:
Hi @ganeshkumarsw 

Could you connect to any other broker? (have you tried the default `mqtt.eclipse.org`)
I've tested your example with `mqtt://broker.hivemq.com` and it connected to the server correctly. Might the `hivemq.com` refuse connections from some geo-locations or form your internet provider?

Could you please quickly test if the connection works when sending AT commands to the module in a manual way? It should be very easy to test using TCP command, such as:
```
AT+QIOPEN=1,0,"TCP","broker.hivemq.com",....
```

ganeshkumarsw, 2020-08-20:
Hi,

I tried other brokers as well but still no luck. Using AT commands for MQTT it works seamlessly but not in PPP mode.

david-cermak, 2020-08-20:
Thanks for the answers. In general I would suggest asking your operator for any possible restrictions, maybe in PPP mode?

> To my surprise, the example works for SIM868 but not for bg96.

AFAICT the SIM868 modem is a `2G` device, so might seem that the connection works okay over `2G` network, but doesn't over `LTE`? Probably also a question to your internet provider.

slechta, 2020-09-18:
I have very similar issue with SIM800 modem. PPP initializes and I get IP, mask, gateway and dns.
(Then I get the "Got IP Event")
The issue is that I do not get any connection.

closed 2021-09-10 by espressif-bot