espressif/esp-idf #5755: .elf file [WinError 2] The system cannot find the file specified

thomasnugent, 2020-08-18:
## Environment

- Development Kit: Sparkfun ESP32 Thing
- Module or chip used: ESP32
- IDF version: v4.1-rc and release/v4.1
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:  Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: VS Code, have changed to LF End of Line Sequence
- Power Supply: USB

## Problem Description

I've tried to build and flash my ESP today without changing any code (i.e. from what I can remember... and I've tried countless previous git commits). Admittedly, I can get `hello-world` and a few other projects working, but not any which use [this IMU code successfully](https://github.com/thomasnugent/esp-mpu9250/tree/feature/esp32) (which has worked for months).

I believe the only things I've done is:
* solder the pins of my SparkFun ESP32 Thing to headers, but I highly doubt that's the issue.
* copied some "calibration values" from cmd.exe output to the source code, which may have copied over CRLF characters (`^M`). But now previous commits don't work either...

I've tried redownloading and installing esp-idf with no luck.

I'd say the main thing that I can think that could be causing this is the "Project Version" as reported by CMake (output is below, I still use `idf.py build`):

```
-- Project version: MCU-to-Unity-and-PCB-v1-4-g8a7056c-dirty
```

This version is reported no matter what project I build/flash, including `hello-world`.  For context, "MCU-to-Unity-and-PCB-v1" is a git tag I added to a previous commit.

I'd love any help I can get, thanks in advance!

### Steps to reproduce

1. Clone [this IMU code](https://github.com/thomasnugent/esp-mpu9250/tree/feature/esp32)
2. `idf.py flash monitor`

## Debug Logs

```
C:\Users\tomnu\Documents\honoursthesis\ESP32\simulimb-esp>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\users\tomnu\documents\honoursthesis\esp32\simulimb-esp\build
Executing "ninja all"...
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Project version: MCU-to-Unity-and-PCB-v1-4-g8a7056c-dirty
-- Building ESP-IDF components for target esp32

Project build complete. To flash, run this command:
C:\Users\tomnu\.espressif\python_env\idf4.1_py3.7_env\Scripts\python.exe ..\..\..\esp-idf\components\esptool_py\esptool\esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build\bootloader\bootloader.bin 0x8000 build\partition_table\partition-table.bin 0x10000 build\simulimb.bin
or run 'idf.py -p (PORT) flash'

C:\Users\tomnu\Documents\honoursthesis\ESP32\simulimb-esp>idf.py flash monitor
Adding "flash"'s dependency "all" to list of commands with default set of options.
Executing action: all (aliases: build)
Running ninja in directory c:\users\tomnu\documents\honoursthesis\esp32\simulimb-esp\build
Executing "ninja all"...
[1/4] cmd.exe /C "cd /D C:\Users\tomnu...*************************************"
Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[2/4] Performing build step for 'bootloader'
ninja: no work to do.
Executing action: flash
Choosing default port b'COM5' (use '-p PORT' option to set a specific serial port)
Running esptool.py in directory c:\users\tomnu\documents\honoursthesis\esp32\simulimb-esp\build
Executing "C:\Users\tomnu\.espressif\python_env\idf4.1_py3.7_env\Scripts\python.exe C:\Users\tomnu\Documents\esp-idf\components/esptool_py/esptool/esptool.py -p COM5 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash @flash_project_args"...
esptool.py -p COM5 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 simulimb.bin
esptool.py v2.9-dev
Serial port COM5
Connecting........_
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 26MHz
MAC: fc:f5:c4:01:1d:40
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 3072 bytes to 103...
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 759.1 kbit/s)...
Hash of data verified.
Compressed 15520 bytes to 10812...
Wrote 15520 bytes (10812 compressed) at 0x00001000 in 0.3 seconds (effective 456.5 kbit/s)...
Hash of data verified.
Compressed 159792 bytes to 87208...
Wrote 159792 bytes (87208 compressed) at 0x00010000 in 2.2 seconds (effective 578.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory c:\users\tomnu\documents\honoursthesis\esp32\simulimb-esp

Executing "C:\Users\tomnu\.espressif\python_env\idf4.1_py3.7_env\Scripts\python.exe C:\Users\tomnu\Documents\esp-idf\tools/idf_monitor.py -p COM5 -b 115200 --toolchain-prefix xtensa-lx106-elf- c:\users\tomnu\documents\honoursthesis\esp32\simulimb-esp\build\simulimb.elf -m 'C:\Users\tomnu\.espressif\python_env\idf4.1_py3.7_env\Scripts\python.exe' 'C:\Users\tomnu\Documents\esp-idf\tools\idf.py'"...
--- idf_monitor on COM5 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:1260
load:0x40078000,len:10968
xtensa-lx106-elf-addr2line -pfiaC -e c:\users\tomnu\documents\honoursthesis\esp32\simulimb-esp\build\simulimb.elf 0x40078000: [WinError 2] The system cannot find the file specified
load:0x40080400,len:3184
xtensa-lx106-elf-addr2line -pfiaC -e c:\users\tomnu\documents\honoursthesis\esp32\simulimb-esp\build\simulimb.elf 0x40080400: [WinError 2] The system cannot find the file specified
entry 0x400805ec
xtensa-lx106-elf-addr2line -pfiaC -e c:\users\tomnu\documents\honoursthesis\esp32\simulimb-esp\build\simulimb.elf 0x400805ec: [WinError 2] The system cannot find the file specified
$\�YH���+ˋ�r�Ei�����ѕ�5R*�͕��ɕ���U����ѕ�5R��Ʌ�ѕɥ镑��ͥ���*�͕��ɕ�5R��Ʌ�ѕɥ镑��ͥ���*�͕��ɕ�5R�abort() was called at PC 0x400d4850 on core 0
xtensa-lx106-elf-addr2line -pfiaC -e c:\users\tomnu\documents\honoursthesis\esp32\simulimb-esp\build\simulimb.elf 0x400d4850: [WinError 2] The system cannot find the file specified

ELF file SHA256: 4679623952d24103

Backtrace: 0x400837c8:0x3ffb7310 0x40083b79:0x3ffb7330 0x400d4850:0x3ffb7350 0x400d32c8:0x3ffb7380 0x400d3513:0x3ffb7440
```

After this line, the ESP reboots and the same error messages occur forever.

## Other items if possible

sdkconfig and .elf file included.
[bugreport.zip](https://github.com/espressif/esp-idf/files/5093123/bugreport.zip)

thomasnugent, 2020-08-19:
I'm pretty convinced that this is a problem with line endings in my source code... I believe it's happened before but I was able to reverse it with a different git commit.

Has anyone had issues with this on Windows?

More environment settings and debug output:
Windows, VSCode, `git config --get core.autocrlf` returns `false`

```
C:\Users\tomnu\Documents\honoursthesis\ESP32\simulimb-esp>idf.py flash monitor
Adding "flash"'s dependency "all" to list of commands with default set of options.
Executing action: all (aliases: build)
Running ninja in directory c:\users\tomnu\documents\honoursthesis\esp32\simulimb-esp\build
Executing "ninja all"...
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Project version: MCU-to-Unity-and-PCB-v1-5-g7e6e385-dirty
-- Building ESP-IDF components for target esp32
-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- Adding linker script C:/Users/tomnu/Documents/honoursthesis/ESP32/simulimb-esp/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Components: ahrs app_trace app_update bootloader bootloader_support cxx driver efuse esp32 esp_adc_cal esp_common esp_eth esp_event esp_netif esp_ringbuf esp_rom esp_wifi espcoredump esptool_py freertos heap log lwip main mbedtls mpu9250 newlib nvs_flash partition_table perfmon pthread soc spi_flash tcpip_adapter vfs wpa_supplicant xtensa
-- Component paths: C:/Users/tomnu/Documents/honoursthesis/ESP32/simulimb-esp/components/ahrs C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/app_trace C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/app_update C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/bootloader C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/bootloader_support C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/cxx C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/driver C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/efuse C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/esp32 C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/esp_adc_cal C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/esp_common C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/esp_eth C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/esp_event C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/esp_netif C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/esp_ringbuf C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/esp_rom C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/esp_wifi C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/espcoredump C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/esptool_py C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/freertos C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/heap C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/log C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/lwip C:/Users/tomnu/Documents/honoursthesis/ESP32/simulimb-esp/main C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/mbedtls C:/Users/tomnu/Documents/honoursthesis/ESP32/simulimb-esp/components/mpu9250 C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/newlib C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/nvs_flash C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/partition_table C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/perfmon C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/pthread C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/soc C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/spi_flash C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/tcpip_adapter C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/vfs C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/wpa_supplicant C:/Users/tomnu/Documents/esp-idf-v4.1-rc/components/xtensa
-- Configuring done
```

thomasnugent, 2020-08-19:
My sincere apologies! Turns out it was an issue with my IMU (MPU9250) which was resetting the board when attempting to initialise itself.

If anyone's still following this, I would love some guidance on those two points I raised (since I couldn't find anything online):

* can line endings screw up an ESP-IDF project in Windows (from my own testing just now, both LF and CRLF work just fine)
* what does "Project Version" mean from CMake? Why does it seem to follow my git tag name?

closed 2020-08-19 by thomasnugent

projectgus, 2020-09-28:
Hi @thomasnugent ,

Sorry for the delayed reply:

>  can line endings screw up an ESP-IDF project in Windows (from my own testing just now, both LF and CRLF work just fine)

No it can't, but git may report that the directory is "dirty" depending on how it's configured to handle line ending.

>  what does "Project Version" mean from CMake? Why does it seem to follow my git tag name?

If the project directory is a git repo, this is the output of `git describe --tags --dirty`.  `MCU-to-Unity-and-PCB-v1-5-g7e6e385-dirty` can be translated as "Tag MCU-to-Unity-and-PCB-v1, plus 5 additional commits on top, current commit has 7e6e385, dirty working copy with uncommitted changes".

The "git describe" help should give more details about this.