espressif/esp-idf #5757: esp32 mqtt over ssl issue

sowmimalu, 2020-08-19:
hai ,
   i am trying implement mqtt over ssl and espi-df v4.0 dev.
tried with (examples/protocols/mqtt/ssl) example code
with default key is not connecting to server 
when i am trying with my server and key i am getting the following error.
E (40589) esp-tls: mbedtls_x509_crt_parse returned -0x262e
E (40599) esp-tls: Failed to open new connection
E (40599) TRANS_SSL: Failed to open a new connection
E (40599) MQTT_CLIENT: Error transport connect

how to implement mqtt over ssl and what are the steps to folllow ?
help me out solve this.

thankyou,
Sowmiya

ESP-Marius, 2020-08-19:
Hi,

> tried with (examples/protocols/mqtt/ssl) example code
with default key is not connecting to server

Are you saying the example doesnt work in when connecting to the example server? Or that it doesnt work when switching to _your_ server?

> when i am trying with my server and key i am getting the following error.
E (40589) esp-tls: mbedtls_x509_crt_parse returned -0x262e

Like the error says, it failed to parse your certificate. Make sure that your certificate is parseable (e.g. test it using openssl/online checker) and that you are embedding it correctly in your app.

sowmimalu, 2020-08-19:
tired with default key and url which is in example code     mqtts://mqtt_eclipse_org port : 8883 but it not connecting to server
after tried with my sever and key   below is my error 
E (4755) esp-tls: mbedtls_ssl_handshake returned -0x2700
I (4755) esp-tls: Failed to verify peer certificate!
I (4755) esp-tls: verification info:! The certificate Common Name (CN) does not match with the expected CN

E (4775) esp-tls: Failed to open new connection
E (4775) TRANS_SSL: Failed to open a new connection
E (4785) MQTT_CLIENT: Error transport connect

here i attached my code with the changes with key hardcoded . strucking in this place what are the changes to do in ssl example ?
what is the mistake here ? help me out to solve this..

#include <stdio.h>
#include <stdint.h>
#include <stddef.h>
#include <string.h>
#include "esp_wifi.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "esp_event_loop.h"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
#include "freertos/queue.h"
#include "freertos/event_groups.h"

#include "lwip/sockets.h"
#include "lwip/dns.h"
#include "lwip/netdb.h"

#include "esp_log.h"
#include "mqtt_client.h"

#include <time.h>
#include <sys/time.h>
#include "esp_log.h"
#include "esp_attr.h"
#include "esp_sleep.h"
#include "nvs_flash.h"

#include "lwip/err.h"
#include "lwip/apps/sntp.h"

static const char *TAG = "MQTTS_EXAMPLE";

static EventGroupHandle_t wifi_event_group;
const static int CONNECTED_BIT = BIT0;

static esp_err_t wifi_event_handler(void *ctx, system_event_t *event)
{
    switch (event->event_id) {
        case SYSTEM_EVENT_STA_START:
            esp_wifi_connect();
            break;
        case SYSTEM_EVENT_STA_GOT_IP:
            xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);

            break;
        case SYSTEM_EVENT_STA_DISCONNECTED:
            esp_wifi_connect();
            xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
            break;
        default:
            break;
    }
    return ESP_OK;
}
static void wifi_init(void)
[... 105 lines cut ...]
	.transport = MQTT_TRANSPORT_OVER_SSL,
    };

    ESP_LOGI(TAG, "[APP] Free memory: %d bytes", esp_get_free_heap_size());
    esp_mqtt_client_handle_t client = esp_mqtt_client_init(&mqtt_cfg);
    esp_mqtt_client_start(client);
}

void app_main()
{
    ESP_LOGI(TAG, "[APP] Startup..");
    ESP_LOGI(TAG, "[APP] Free memory: %d bytes", esp_get_free_heap_size());
    ESP_LOGI(TAG, "[APP] IDF version: %s", esp_get_idf_version());

    esp_log_level_set("*", ESP_LOG_INFO);
    esp_log_level_set("MQTT_CLIENT", ESP_LOG_VERBOSE);
    esp_log_level_set("TRANSPORT_TCP", ESP_LOG_VERBOSE);
    esp_log_level_set("TRANSPORT_SSL", ESP_LOG_VERBOSE);
    esp_log_level_set("TRANSPORT", ESP_LOG_VERBOSE);
    esp_log_level_set("OUTBOX", ESP_LOG_VERBOSE);
     nvs_flash_init();
    wifi_init();
   mqtt_app_start();

}

sowmimalu, 2020-08-19:
I (5663) event: sta ip: 192.168.43.206, mask: 255.255.255.0, gw: 192.168.43.1
I (5663) MQTTS_EXAMPLE: [APP] Free memory: 4370172 bytes
I (5663) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (5673) MQTTS_EXAMPLE: Other event id:7
I (5873) mbedtls: ssl_tls.c:8021 => handshake

I (5873) mbedtls: ssl_cli.c:3405 client state: 0

I (5873) mbedtls: ssl_tls.c:2751 => flush output

I (5883) mbedtls: ssl_tls.c:2763 <= flush output

I (5883) mbedtls: ssl_cli.c:3405 client state: 1

I (5893) mbedtls: ssl_tls.c:2751 => flush output

I (5893) mbedtls: ssl_tls.c:2763 <= flush output

I (5903) mbedtls: ssl_cli.c:774 => write client hello

I (5913) mbedtls: ssl_tls.c:3180 => write handshake message

I (5913) mbedtls: ssl_tls.c:3337 => write record

I (5923) mbedtls: ssl_tls.c:2751 => flush output

I (5923) mbedtls: ssl_tls.c:2770 message length: 242, out_left: 242

I (5933) mbedtls: ssl_tls.c:2775 ssl->f_send() returned 242 (-0xffffff0e)

I (5943) mbedtls: ssl_tls.c:2803 <= flush output

I (5943) mbedtls: ssl_tls.c:3470 <= write record

I (5953) mbedtls: ssl_tls.c:3314 <= write handshake message

I (5953) mbedtls: ssl_cli.c:1106 <= write client hello

I (5963) mbedtls: ssl_cli.c:3405 client state: 2

I (5963) mbedtls: ssl_tls.c:2751 => flush output

I (5973) mbedtls: ssl_tls.c:2763 <= flush output

I (5983) mbedtls: ssl_cli.c:1499 => parse server hello

I (5983) mbedtls: ssl_tls.c:4305 => read record

I (5993) mbedtls: ssl_tls.c:2532 => fetch input

I (5993) mbedtls: ssl_tls.c:2693 in_left: 0, nb_want: 5

I (6183) mbedtls: ssl_tls.c:2717 in_left: 0, nb_want: 5

I (6183) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (6183) mbedtls: ssl_tls.c:2738 <= fetch input

I (6193) mbedtls: ssl_tls.c:2532 => fetch input

I (6193) mbedtls: ssl_tls.c:2693 in_left: 5, nb_want: 70

I (6203) mbedtls: ssl_tls.c:2717 in_left: 5, nb_want: 70

I (6203) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 65 (-0xffffffbf)

I (6213) mbedtls: ssl_tls.c:2738 <= fetch input

I (6223) mbedtls: ssl_tls.c:4379 <= read record

I (6223) mbedtls: ssl_cli.c:1781 server hello, total extension length: 21

I (6233) mbedtls: ssl_cli.c:1970 <= parse server hello

I (6243) mbedtls: ssl_cli.c:3405 client state: 3

I (6243) mbedtls: ssl_tls.c:2751 => flush output

I (6253) mbedtls: ssl_tls.c:2763 <= flush output

I (6253) mbedtls: ssl_tls.c:5440 => parse certificate

I (6263) mbedtls: ssl_tls.c:4305 => read record

I (6263) mbedtls: ssl_tls.c:2532 => fetch input

I (6273) mbedtls: ssl_tls.c:2693 in_left: 0, nb_want: 5

I (6283) mbedtls: ssl_tls.c:2717 in_left: 0, nb_want: 5

I (6283) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (6293) mbedtls: ssl_tls.c:2738 <= fetch input

I (6303) mbedtls: ssl_tls.c:2532 => fetch input

I (6303) mbedtls: ssl_tls.c:2693 in_left: 5, nb_want: 879

I (6313) mbedtls: ssl_tls.c:2717 in_left: 5, nb_want: 879

I (6313) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 874 (-0xfffffc96)

I (6323) mbedtls: ssl_tls.c:2738 <= fetch input

I (6343) mbedtls: ssl_tls.c:4379 <= read record

W (6343) mbedtls: ssl_tls.c:5713 x509_verify_cert() returned -9984 (-0x2700)

I (6343) mbedtls: ssl_tls.c:5244 => send alert message

I (6353) mbedtls: ssl_tls.c:3337 => write record

I (6353) mbedtls: ssl_tls.c:2751 => flush output

I (6363) mbedtls: ssl_tls.c:2770 message length: 7, out_left: 7

I (6373) mbedtls: ssl_tls.c:2775 ssl->f_send() returned 7 (-0xfffffff9)

I (6373) mbedtls: ssl_tls.c:2803 <= flush output

I (6383) mbedtls: ssl_tls.c:3470 <= write record

I (6383) mbedtls: ssl_tls.c:5257 <= send alert message

I (6393) mbedtls: ssl_tls.c:5810 <= parse certificate

I (6393) mbedtls: ssl_tls.c:8031 <= handshake

E (6403) esp-tls: mbedtls_ssl_handshake returned -0x2700
I (6413) esp-tls: Failed to verify peer certificate!
I (6413) esp-tls: verification info:   ! The certificate Common Name (CN) does not match with the expected CN
  ! The certificate validi
E (6433) esp-tls: Failed to open new connection
E (6433) TRANS_SSL: Failed to open a new connection
E (6433) MQTT_CLIENT: Error transport connect
I (6443) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED

here is my log when i hardcoded key for my server.
awaiting for ur reply...

thanks,
Sowmiya

ESP-Marius, 2020-08-20:
Now you are posting a different error though?  -0x2700 vs -0x262e, did you fix your original problem then? 

The new error message pretty much says what your problem is, `The certificate Common Name (CN) does not match with the expected CN`

The common name of the certificate needs to match the host you are connecting to. My guess is that you are connecting using an IP, so whatever you have as a common name doesn't match this.

sowmimalu, 2020-08-20:
yaah thank you for your quick reply , i am connecting with an ip, same ip as the host name on server side too still i m getting the same issue strucking here for the past one week .
the steps followed  :
1. default key which is not working for me without doing any changes on code.
2. my sever ip and ca is connecting with mqtt.fx and not into my idf and above is my code i done the changes using tls1.2
then is there any mistake on code side 
3. tried with some online different online server  same connecting with mqtt.fx getting same error on idf.
could you please share me if any sample code for this if i m doing anything wrong here?

ESP-Marius, 2020-08-20:
1. I just tested the mqtt/ssl example (commit id: 820621687ce04b3c8) and it connects just fine (no changes except the wifi password). If you are saying that you can't even get the example to work then you should start troubleshooting here before trying to change anything. Then there must be something wrong with your Wifi connection/firewall etc.

2. MQTT.fx might not be as strict with checking the certificate as IDF (mbedtls) is. If you are are connecting directly using an IP address, but your certificate specifies a name as the common name then this will _not_ work.

sowmimalu, 2020-08-20:
1.default url and key which is in example code is not connecting with mqtt.fx also  but my server ip and key connecting with mqtt i dont know what is the mistake i am doing here .
2.i am connecting device with my wifi mobile network.
could you please share me the ssl code which you tested ?

ESP-Marius, 2020-08-20:
I already linked it: https://github.com/espressif/esp-idf/commit/820621687ce04b3c85276ca0eb3fe72479573cfd

sowmimalu, 2020-08-20:
yaah got it , default not connecting with 
[mqtt_eclipse_org.txt](https://github.com/espressif/esp-idf/files/5101053/mqtt_eclipse_org.txt)

mqtt.fx , is that default key is valid ?  it is in pem format

ESP-Marius, 2020-08-20:
Are you asking me why you cant connect with mqtt.fx using that certificate? If yes, then I have no idea, you need get help from the mqtt.fx guys then....

And yes, it's a valid certificate. You can even go to https://mqtt.eclipse.org/ and look at the certificate chain they are using and compare it if you want to. If you want to know if it's a valid .pem certificate then just check it yourself? There's a bunch of online certificate parsers out there.

In general, github issues are not the place to ask general "how do I do this" type of questions, we recommend using our forum: https://www.esp32.com/ for that.

ESP-Marius, 2020-08-21:
That's good!

There's an option to turn off common name checking in MQTT added recently (https://github.com/espressif/esp-mqtt/commit/5e8950e681e5419d85e0ffc0d72a78cdeb894fd7). But the IDF repo still doesn't reference this commit (there's an in-progress MR to update it).

ESP-Marius, 2020-08-24:
Not in the lastest IDF yet, but soon.

ESP-Marius, 2020-09-17:
As of ffd7f39189278924fe81c94187167073342ba6a7 it's now possible to disable common name checking. Of course we still recommend keeping it on to avoid potential MITM attacks.

Closing issue.

closed 2020-09-17 by ESP-Marius

sowmimalu, 2020-09-17:
Thanks for your reply, i am using dev 4.0 now i have to to update same version git for this feature ?

ESP-Marius, 2020-09-17:
@sowmimalu release/v4.0 currently doesnt have the update yet, but hopefully it shouldnt take too long before it's backported and published here.

sowmimalu, 2020-09-17:
which git version i should now for this update ?

sowmimalu, 2020-09-17:
which git release version i should update for this now ?

ESP-Marius, 2020-09-17:
master, but you could also checkout the latest mqtt submodule yourself if you wanna keep using release/v4.0

sowmimalu, 2020-09-17:
yaah , but i need to disable the common name checking option in ssl

sowmimalu, 2020-09-17:
could you please share the git link of master which is the latest release ?

ESP-Marius, 2020-09-17:
Not sure which you are asking for, latest commit on esp-idf master is https://github.com/espressif/esp-idf/commit/8bc19ba893e5544d571a753d82b44a84799b94b1, lastet esp-mqtt commit is https://github.com/espressif/esp-mqtt/commit/fcd7f0d0839a1054084220f5786722e135c6e660

sowmimalu, 2020-09-17:
because of , currently i am using v4.0 dev there is no option to disable common name checking , for this i need  to update the latest commit