espressif/esp-idf #5758: Heap tracing 'Standalone' couse HTTPS client crash

lbaltoro, 2020-08-19:
## Environment

- Development Kit:      [ESP32-Ethernet-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1.2]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it):
    // v4.2-dev-2077-g7dfe4b468
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     [Windows]
- (Windows only) environment type: [Plain Command Prompt].
- Using an IDE?: [No]
- Power Supply:         [external 5V]

## Problem Description

I'm working on bigger project with ethernet works with MQTT, HTTPS.
Module has often crash with:
'Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.'
when try to connect i.e. google.com by HTTPS.
I tried other HTTPS clients based on pure mbedtls library and module has still error.

I've isolated the problem to the simplest example.
I had combined:
esp-idf\examples\ethernet\basic\ with esp_http_client example
and run, and it works.
But when I set Heap tracing to 'Standalone' it couse crashing.

### Expected Behavior
HTTPS Client works with enabled Heap tracing
### Actual Behavior
ESP crashes after few HTTPS requests when 'Heap tracing' option is set to Standalone
### Steps to reproduce

1. Add to esp-idf\examples\ethernet\basic  https request example
2. Enable Heap tracing to Standalone

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.

void app_main(void) {
.....

   xTaskCreate(&https_get_task, "https_get_task", 8192, NULL, 5, NULL);
}

static void https_get_task(void *pvParameters) {
	esp_http_client_config_t config = {
        .host = "google.com",
        .path = "/",
        .transport_type = HTTP_TRANSPORT_OVER_SSL,
    };
	while(1) {
		esp_http_client_handle_t client = esp_http_client_init(&config);
		esp_err_t err = esp_http_client_perform(client);

		if (err == ESP_OK) {
			
			ESP_LOGI(TAG, "HTTPS Status = %d, content_length = %d",
					esp_http_client_get_status_code(client),
					esp_http_client_get_content_length(client));
		} else {
			ESP_LOGE(TAG, "Error perform http request %s", esp_err_to_name(err));
		}
		
		esp_http_client_cleanup(client);
		vTaskDelay(1000 / portTICK_PERIOD_MS);
	}
	vTaskDelete(NULL);
}

```

## Debug Logs

```
I (5467) eth_example: ~~~~~~~~~~~
I (5467) eth_example: ETHIP:192.168.0.248
I (5477) eth_example: ETHMASK:255.255.255.0
I (5477) eth_example: ETHGW:192.168.0.101
I (5487) eth_example: ~~~~~~~~~~~
I (12997) eth_example: HTTPS Status = 200, content_length = -1
I (17277) eth_example: HTTPS Status = 200, content_length = -1
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400f949c: 22891289 0000f01d 81004136
0x400f949c: mbedtls_mpi_free at C:/Users/Sdarm/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:142

Core  0 register dump:
PC      : 0x400f94a0  PS      : 0x00060930  A0      : 0x00040020  A1      : 0x3ffbf350
0x400f94a0: mbedtls_mpi_free at C:/Users/Sdarm/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:144

A2      : 0x3ffbf370  A3      : 0x3ffc8ce8  A4      : 0x3ffc80e8  A5      : 0x00000000
A6      : 0x3ffc810c  A7      : 0x00060520  A8      : 0x00000000  A9      : 0x3ffbf330
A10     : 0x3ffc80c8  A11     : 0x00000008  A12     : 0xa7caa262  A13     : 0x3ffc8cec
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace:0x400f949d:0x3ffbf350 0x0004001d:0x3ffbf370 |<-CORRUPTED
0x400f949d: mbedtls_mpi_free at C:/Users/Sdarm/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:142

ELF file SHA256: 6ddf7841d14b34c5

Rebooting...
```

## Other items if possible

[coredump.log](https://github.com/espressif/esp-idf/files/5094809/coredump.log)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5094814/sdkconfig.txt)
[ethernet_example_main.txt](https://github.com/espressif/esp-idf/files/5094837/ethernet_example_main.txt)

Alvin1Zhang, 2020-08-19:
Thanks for reporting and letting us know, we will look into.

0xjakob, 2020-09-03:
@lbaltoro We have found the issue, a bug we've been hunting for a few weeks already. Your post helped us fixing it! We'll first apply the fix on master. Then backport it to v4.2.

closed 2020-09-16 by espressif-bot