espressif/esp-idf #5759: When Bluetooth and wiiFI turn on together

poisonousfish, 2020-08-19:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32D
- IDF version : V4.1 
- Build System:         idf.py
- Compiler version :xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r2) 8.2.0
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?:  ESP IDF Explorer
- Power Supply:         Battery（Ensure adequate output）

## Problem Description
When I'm transmitting data over TCP, the following prompts sometimes come up.
I had this problem1 at first.
I think WIFI PS mode is probably the cause of this problem1
Then I encountered the second problem
When I have wifi and Bluetooth on, I can't set WIFI PS to WIFI_PS_NONE
WIFI_PS_MIN_MODEM or WIFI_PS_MAX_MODEM is fine

PS:I found that other people also encountered this problem when working with wifi Mesh. You said that you have fixed this bug, but the latest V4.1 stable version I used still has this problem

当我使用TCP进行传输数据的时候，时不时的会出事wifi断开.
这是我的问题1
我认为可能是wifi的power saver mode 导致wifi连接不稳
然后我遇到的第二个问题
当我开启蓝牙和wifi的时候我不能将ps设置为WIFI_PS_NONE
如果设置为WIFI_PS_MIN_MODEM 或者WIFI_PS_MAX_MODEM 就是可以的

PS:我发现别人进行wifi mesh时候也遇到这个问题，你们表示已经修复这个bug，但是我使用的最新的V4.1稳定版还是出现这个问题

## problem 1.
I (55868) wifi:bcn_timout,ap_probe_send_start
I (64978) wifi:bcn_timout,ap_probe_send_start
I (79538) wifi:bcn_timout,ap_probe_send_start
I (88938) wifi:bcn_timout,ap_probe_send_start

## problem 2.
I (1311) wifi:wifi firmware version: 3ea4c76
I (1311) wifi:config NVS flash: enabled
I (1311) wifi:config nano formating: disabled
I (1311) wifi:Init dynamic tx buffer num: 32
I (1321) wifi:Init data frame dynamic rx buffer num: 32
I (1321) wifi:Init management frame dynamic rx buffer num: 32
I (1331) wifi:Init management short buffer num: 32
I (1331) wifi:Init static rx buffer size: 1600
I (1341) wifi:Init static rx buffer num: 10
I (1341) wifi:Init dynamic rx buffer num: 32
I (1351) wifi:mode : sta (a4:cf:12:64:5f:68)
I (1351) wifi:Set ps type: 0
E (1351) wifi:Error! Should enable WiFi modem sleep when both WiFi and Bluetooth are enabled!!!!!!

## last
I just want to be able to have a steady connection and transmission when Bluetooth and wifi are on

我只是想要在开启wifi和蓝牙时进行一个稳定数据传输。

liuzfesp, 2020-08-20:
Hi @poisonousfish,

For problem#2, as following log indicates, it's a must to enable WiFi modem sleep when both WiFi and BT are enabled, you can't configure to WIFI_PS_NONE in WiFi/BT coexist mode.
`E (1351) wifi:Error! Should enable WiFi modem sleep when both WiFi and Bluetooth are enabled!!!!!!`

For problem#1, it's maybe caused by AP's problem, could you help to do below test:

- Two ESP32 device, one is WiFi only and disable WiFi sleep (WIFI_PS_NONE), another is in WiFi/BT coexist mode and enabled WiFi sleep.
- Test and check whether both of them has problem#1 at the same time. If yes, it means AP has problem of sending beacon, otherwise problem#1 may be related to ESP32.
- It will be very helpful if you can capture the WiFi packets also.

poisonousfish, 2020-08-20:
> Hi @poisonousfish,
>
> For problem#2, as following log indicates, it's a must to enable WiFi modem sleep when both WiFi and BT are enabled, you can't configure to WIFI_PS_NONE in WiFi/BT coexist mode. [...]

Yes, I did the test
Two ESP32 device, one is WiFi only and disable WiFi sleep (WIFI_PS_NONE), another is in WiFi/BT coexist mode and enabled WiFi sleep.
WiFi only, and disabling WiFi sleep (WIFI_PS_NONE) is fine，but another still happen  problem#1

For question 2, that means I can only separate Bluetooth from wifi for stable transmission

liuzfesp, 2020-08-20:
Hi @poisonousfish, for problem#1, how about WiFi only + enable sleep mode?

poisonousfish, 2020-08-20:
> Hi @poisonousfish, for problem#1, how about WiFi only + enable sleep mode?

Let me run a test and get back to you later

liuzfesp, 2020-08-24:
Hi @poisonousfish could you help to capture the WiFi packets between ESP32 and AP?

poisonousfish, 2020-08-25:
> Hi @poisonousfish could you help to capture the WiFi packets between ESP32 and AP?

I'm not sure what I should do，i never capture the WiFi packets between ESP32 and AP

esp-HarveyRong, 2020-09-01:
@poisonousfish, 
The recommended Macbook's built-in packet capture tool does not require a packet capture card. It can be used to capture 802.11a/b/g/n/ac packets:
https://osxdaily.com/2015/04/23/sniff-packet-capture-packet-trace-mac-os-x-wireless-diagnostics/
Or you can use the tutorial on the official Espressif website to capture packets using Wireshark
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html

vitorrussi, 2020-09-02:
I'm having the same problem 2 with the wifi/bluetooth coexistence, even though i set esp_wifi_set_ps(WIFI_PS_MIN_MODEM), after mwifi_start() it automatically put ps type to 0, and then crashes. wifi without bluetooth works fine, and i'm using release/v1.0. they can actually coexist or i have to manually disable esp-wifi-mesh to use bluetooth?

liuzfesp, 2020-09-03:
HI @vitorrussi, please don't disable Wifi modem sleep for WiFi/BT coexistence.
BTW, are you using IDF release/v1.0? Is that correct?

poisonousfish, 2020-09-03:
> @poisonousfish,
> The recommended Macbook's built-in packet capture tool does not require a packet capture card. It can be used to capture 802.11a/b/g/n/ac packets: [...]

I will conduct the test and reply to you as soon as possible

poisonousfish, 2020-09-15:
> @poisonousfish,
> The recommended Macbook's built-in packet capture tool does not require a packet capture card. It can be used to capture 802.11a/b/g/n/ac packets: [...]

Sadly, I can't get direct access to AP and ESP32, and I don't have a suitable AP for testing right now.
But my ESP32 is almost stable work.

However, I had a problem with a ESP32 that was running on wifi while it were stalled for more than 45 seconds and not like on connect wifi.

I (2887) wifi:security: WPA2-PSK, phy: b, rssi: -61
I (2897) wifi:pm start, type: 0
**I (2947) wifi:AP's beacon interval = 102400 us, DTIM period = 1**
The pauses here exceed 45S and sometimes exceed 60S

The same software and hardware, the other ESP32 won't encounter, and there's little difference between the two locations.
When I switched to ESP32, the software worked fine.
What could possibly contribute to this phenomenon?
The problematic ESP32 is otherwise working fine.

closed 2020-12-13 by espressif-bot

sim1234, 2021-12-14:
To fix this issue on Arduino platform just set `WiFi.setSleep(true);`

ingeniero-david, 2022-10-07:
@vitorrussi in menuconfig you need to enable the "Enable mesh PS (power save) function" option in MDF Mwifi section.

[image]

guianmon99, 2022-11-09:
hi @ingeniero-david I just tried enabling the mesh PS like you said but i am still have the same error could you please tell me if you were able to use esp-MDFf with ble at the same time.

ingeniero-david, 2022-11-09:
> hi @ingeniero-david I just tried enabling the mesh PS like you said but i am still have the same error could you please tell me if you were able to use esp-MDFf with ble at the same time.

Hi @guianmon99 I have used esp-mdf with classic bt, could you share your code?

guianmon99, 2022-11-10:
Hi @ingeniero-david  thank you for your promptly response. 

static mdf_err_t wifi_init()
{
    mdf_err_t ret          = nvs_flash_init();
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();

    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        MDF_ERROR_ASSERT(nvs_flash_erase());
        ret = nvs_flash_init();
    }

    MDF_ERROR_ASSERT(ret);
}
    MDF_ERROR_ASSERT(esp_netif_init());
    MDF_ERROR_ASSERT(esp_event_loop_create_default());
    MDF_ERROR_ASSERT(esp_wifi_init(&cfg));
    MDF_ERROR_ASSERT(esp_wifi_set_storage(WIFI_STORAGE_FLASH));
    MDF_ERROR_ASSERT(esp_wifi_set_mode(WIFI_MODE_STA));
    MDF_ERROR_ASSERT(esp_wifi_set_ps(WIFI_PS_NONE));
    MDF_ERROR_ASSERT(esp_mesh_set_6m_rate(false));
    MDF_ERROR_ASSERT(esp_wifi_start());

    return MDF_OK;
} Here is the configuration for the wifi that I use.
         mwifi_init_config_t cfg = MWIFI_INIT_CONFIG_DEFAULT();
         mwifi_config_t config   = {
        .channel   = CONFIG_MESH_CHANNEL,
        .mesh_id   = CONFIG_MESH_ID,
        .mesh_type = CONFIG_DEVICE_TYPE,
    }; 
And here is the start of the mesh. 
MDF_ERROR_ASSERT(mdf_event_loop_init(event_loop_cb));
    MDF_ERROR_ASSERT(wifi_init());
    MDF_ERROR_ASSERT(mwifi_init(&cfg));
    MDF_ERROR_ASSERT(mwifi_set_config(&config));
    MDF_ERROR_ASSERT(mwifi_start());
I am using nimble stack, but i suspect the wifi configuration to be the problem.

guianmon99, 2022-11-10:
Hi @ingeniero-david  thank you for your promply response. 
here is the code for initialising the wifi that I use 
static mdf_err_t wifi_init()
{
    mdf_err_t ret          = nvs_flash_init();
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();

    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        MDF_ERROR_ASSERT(nvs_flash_erase());
        ret = nvs_flash_init();
    }

    MDF_ERROR_ASSERT(ret);

    MDF_ERROR_ASSERT(esp_netif_init());
    MDF_ERROR_ASSERT(esp_event_loop_create_default());
    MDF_ERROR_ASSERT(esp_wifi_init(&cfg));
    MDF_ERROR_ASSERT(esp_wifi_set_storage(WIFI_STORAGE_FLASH));
    MDF_ERROR_ASSERT(esp_wifi_set_mode(WIFI_MODE_STA));
    MDF_ERROR_ASSERT(esp_wifi_set_ps(WIFI_PS_MIN_MODEM));
    MDF_ERROR_ASSERT(esp_mesh_set_6m_rate(false));
    MDF_ERROR_ASSERT(esp_wifi_start());

    return MDF_OK;
}

mwifi_init_config_t cfg = MWIFI_INIT_CONFIG_DEFAULT();
    mwifi_config_t config   = {
        .channel   = CONFIG_MESH_CHANNEL,
        .mesh_id   = CONFIG_MESH_ID,
        .mesh_type = CONFIG_DEVICE_TYPE,
    };

    static rgb_t node_task = {0};
    node_task.node = NULL;
    static counters_t root= {0};

    uint32_t offmode[6]={192,0,0,0,67,192};/*colors in offline mode rgb values */
    //static TaskHandle_t led_task2 = NULL;
    /**
     * @brief Set the log level for serial port printing.
     */
    esp_log_level_set("*", ESP_LOG_INFO);
    esp_log_level_set(TAG, ESP_LOG_DEBUG);

    /**
     * @brief Initialize wifi mesh.
     */
    MDF_ERROR_ASSERT(mdf_event_loop_init(event_loop_cb));
    MDF_ERROR_ASSERT(wifi_init());
    MDF_ERROR_ASSERT(mwifi_init(&cfg));
    esp_wifi_set_ps(WIFI_PS_MIN_MODEM);
    MDF_ERROR_ASSERT(mwifi_set_config(&config));
    MDF_ERROR_ASSERT(mwifi_start());
and here is the initialization of the mesh that is giving me troubles. just for reference both ble and wifi mesh work on their own just fine.

ingeniero-david, 2022-11-10:
@guianmon99 your code looks correct, another thing to keep in mind is that all nodes in the mesh must have enabled the "Enable mesh PS (power save) function" option in MDF Mwifi section even if they do not use bluetooth. Could you share your error?

guianmon99, 2022-11-10:
Hi @ingeniero-david  my error is this: Error! Should enable WiFi modem sleep when both WiFi and Bluetooth are exactly.
Also an update I was able to connect to ble in one of the nodes by starting the ble advertising after the parent connection event of the mesh. but when i tried to do this on the root after the child connected i got the same error.  I could do a workaround with this, but i would prefer to connect to ble using the root node is this normal.  I am using the exact same function to initialize both the wifi stack and mesh  in the root node. thankyou for your help btw.