espressif/esp-idf #5760: [Security] Bluedroid ignore +6 PIN

dragahassan, 2020-08-19:
- D**evelopment Kit:** -
- **Kit version: DevKitC:** v4
- M**odule or chip used:**  ESP32-WROOM-32D
- **IDF version:** v4.2-dev-1905-g625bd5eb1
- **Build System:** idf.py
- **Compiler version:** xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- **Operating System:** Windows
- **(Windows only) environment type:** ESP Command Prompt
- **Power Supply:** USB

The Bluedroid BLE stack take only first six digit for PIN, and ignore completely the rest.

1: Use example: [gatt_security_server](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_security_server)
2: Build & Flash
3: In your Phone/Tablet search and pair with ESP

The PIN is `123456`, but you can correctly pair with `123456999999`, that's because Bluedroid BLE stack take only first six digit.

xiewenxiang, 2020-08-20:
Hi dragahassan,

I tried to reproduce the problem you mentioned, when I entered the wrong key or long sizes key,  all of these will pair fail.
So, I want to confirm with you thar do you have checked the status bit for `ESP_GAP_BLE_AUTH_CMPL_EVT`?

In ESP-IDF, there is actually a check for smp passkey.

`esp-idf/components/bt/host/bluedroid/stack/smp/smp_api.c`

```c
#define BTM_MAX_PASSKEY_VAL (999999)
#define BTM_MIN_PASSKEY_VAL (0)

void SMP_PasskeyReply (BD_ADDR bd_addr, UINT8 res, UINT32 passkey)
{
    tSMP_CB *p_cb = & smp_cb;
    UINT8   failure = SMP_PASSKEY_ENTRY_FAIL;

    SMP_TRACE_EVENT ("SMP_PasskeyReply: Key: %d  Result:%d",
                     passkey, res);

    /* If timeout already expired or has been canceled, ignore the reply */
    if (p_cb->cb_evt != SMP_PASSKEY_REQ_EVT) {
        SMP_TRACE_WARNING ("SMP_PasskeyReply() - Wrong State: %d", p_cb->state);
        return;
    }

    if (memcmp (bd_addr, p_cb->pairing_bda, BD_ADDR_LEN) != 0) {
        SMP_TRACE_ERROR ("SMP_PasskeyReply() - Wrong BD Addr");
        return;
    }

    if (btm_find_dev (bd_addr) == NULL) {
        SMP_TRACE_ERROR ("SMP_PasskeyReply() - no dev CB");
        return;
    }

    if (passkey > BTM_MAX_PASSKEY_VAL || res != SMP_SUCCESS) {
        SMP_TRACE_WARNING ("SMP_PasskeyReply() - Wrong key len: %d or passkey entry fail", passkey);
        /* send pairing failure */
        smp_sm_event(p_cb, SMP_AUTH_CMPL_EVT, &failure);

    } else if (p_cb->selected_association_model == SMP_MODEL_SEC_CONN_PASSKEY_ENT) {
        smp_sm_event(&smp_cb, SMP_SC_KEY_READY_EVT, &passkey);
    } else {
        smp_convert_string_to_tk(p_cb->tk, passkey);
    }

    return;
}
```

dragahassan, 2020-08-20:
I just start completely everything from scratch to be sure, and this security problem **still exist**.

**1:** I create a new Windows 10 Virtualbox! (_because my OS have multiple modified ESP-IDF_)
**2:** Setup ESP IDF using [esp-idf-tools-setup-2.3.exe](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html) (Select new setup for Python & Git)
**3:** Set OUT capability in file `example_ble_sec_gatts_demo.c` from `ESP_IO_CAP_NONE` to `ESP_IO_CAP_OUT`
**4:** Build `idf.py -p COM3 build`
**5:** Clean `ESP32 device idf.py -p COM3 erase_flash`
**6:** Flash `idf.py -p COM3 flash`

In your Phone/Tablet search and pair for `ESP_BLE_SECURITY`, when ask for PIN set the correct pin `123456` then add random numbers.. like full PIN will be `123456789012`., the pairing will success!.

dragahassan, 2020-08-20:
Correct PIN is 123456, I set PIN 123456789012 from LG Phone (Android 10)

```
I (526) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (572) BTDM_INIT: BT controller compile version [c1cbe45]
I (572) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (682) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (922) SEC_GATTS_DEMO: app_main init bluetooth
I (1062) SEC_GATTS_DEMO: The number handle = 8
I (1092) SEC_GATTS_DEMO: advertising start success
I (14072) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
I (14442) SEC_GATTS_DEMO: The passkey Notify number:123456
W (37732) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK
I (37912) SEC_GATTS_DEMO: key type = ESP_LE_KEY_LENC
I (37912) SEC_GATTS_DEMO: key type = ESP_LE_KEY_PENC
I (37912) SEC_GATTS_DEMO: key type = ESP_LE_KEY_LID
I (38242) SEC_GATTS_DEMO: key type = ESP_LE_KEY_PID
I (38242) SEC_GATTS_DEMO: remote BD_ADDR: 5112441c7493
I (38242) SEC_GATTS_DEMO: address type = 1
I (38242) SEC_GATTS_DEMO: pair status = success
I (38252) SEC_GATTS_DEMO: auth mode = ESP_LE_AUTH_REQ_SC_MITM_BOND
I (38262) SEC_GATTS_DEMO: Bonded devices number : 1

I (38262) SEC_GATTS_DEMO: Bonded devices list : 1

I (38272) SEC_GATTS_DEMO: 51 12 44 xx xx xx
```

dragahassan, 2020-08-20:
Correct PIN is 123456, I set PIN 090909010203

```
I (526) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (560) BTDM_INIT: BT controller compile version [c1cbe45]
I (560) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (660) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (900) SEC_GATTS_DEMO: app_main init bluetooth
I (1050) SEC_GATTS_DEMO: The number handle = 8
I (1080) SEC_GATTS_DEMO: advertising start success
I (18150) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
I (18440) SEC_GATTS_DEMO: The passkey Notify number:123456
W (24130) BT_SMP: Commitment check fails

I (24140) SEC_GATTS_DEMO: remote BD_ADDR: 5112441c7493
I (24150) SEC_GATTS_DEMO: address type = 1
I (24150) SEC_GATTS_DEMO: pair status = fail
I (24150) SEC_GATTS_DEMO: fail reason = 0x51
I (24150) SEC_GATTS_DEMO: Bonded devices number : 0

I (24160) SEC_GATTS_DEMO: Bonded devices list : 0

W (27130) BT_BTM: btm_sec_clr_temp_auth_service() - no dev CB

I (27140) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x16
I (27160) SEC_GATTS_DEMO: advertising start success
```

dragahassan, 2020-08-20:
[image]

chegewara, 2020-08-20:
Just idea, maybe smartphone is truncating passkey to only 6 digits?
As you can see on screenshot its only 6 digits displayed.

dragahassan, 2020-08-20:
@chegewara No, because the phone allow 12 digit, also the same thing on Asus tablet., and the log `The passkey Notify number:123456` it's not the pin received but the ESP pin.

dragahassan, 2020-08-20:
@chegewara in other hand maybe you are right, ESP set max pin len to 6 maybe, so device truncating the pin even if user set 12 digit!.

dragahassan, 2020-08-20:
Correct PIN is 123456, I set PIN 123456666666.

**SMP Verbose log:**

```
I (0) cpu_start: App cpu up.
I (444) heap_init: Initializing. RAM available for dynamic allocation:
I (451) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (457) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (463) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (469) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (475) heap_init: At 3FFC9100 len 00016F00 (91 KiB): DRAM
I (481) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (488) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (494) heap_init: At 40092E04 len 0000D1FC (52 KiB): IRAM
I (500) cpu_start: Pro cpu start user code
I (519) spi_flash: detected chip: generic
I (519) spi_flash: flash io: dio
W (519) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (530) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (576) BTDM_INIT: BT controller compile version [c1cbe45]
I (576) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (676) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (916) SEC_GATTS_DEMO: app_main init bluetooth
D (916) BT_SMP: SMP_Init
D (916) BT_SMP: SMDBG l2c smp_l2cap_if_init
D (1006) BT_SMP: SMP_Register state=0
D (1056) BT_SMP: smp_encrypt_data

I (1066) SEC_GATTS_DEMO: The number handle = 8
D (1076) BT_SMP: static passkey 123456
I (1096) SEC_GATTS_DEMO: advertising start success
D (34206) BT_SMP: SMDBG l2c smp_connect_callback

I (34206) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
D (34206) BT_SMP: SMP_Pair state=0 br_state=0 flag=0x0

D (34216) BT_SMP: SMDBG l2c smp_connect_callback

D (34216) BT_SMP: smp_connect_callback()  for pairing BDA: 416d2dc5d8d1  Event: connected

D (34226) BT_SMP: main smp_sm_event

D (34226) BT_SMP: SMP Role: Slave State: [SMP_STATE_IDLE (0)], Event: [L2CAP_CONN_EVT (18)]
D (34236) BT_SMP: State change: SMP_STATE_IDLE(0) ==> SMP_STATE_WAIT_APP_RSP(1)
D (34246) BT_SMP: smp_send_app_cback p_cb->cb_evt=1

D (34256) BT_SMP: io_cap = 4
D (34256) BT_SMP: callback_rc=0  p_cb->cb_evt=1

D (34256) BT_SMP: rcvd auth_req: 0x0d, io_cap: 0                         loc_oob_flag: 0 loc_enc_size: 16,local_i_key: 0x03, local_r_key: 0x03

D (34276) BT_SMP: lmp_version_below LMP version 9 < 8
D (34276) BT_SMP: set auth_req: 0x0d, local_i_key: 0x03, local_r_key: 0x03

D (34286) BT_SMP: main smp_sm_event

D (34286) BT_SMP: SMP Role: Slave State: [SMP_STATE_WAIT_APP_RSP (1)], Event: [API_IO_RSP_EVT (20)]
D (34296) BT_SMP: State change: SMP_STATE_WAIT_APP_RSP(1) ==> SMP_STATE_PAIR_REQ_RSP(3)
D (34306) BT_SMP: smp_process_io_response

D (34306) BT_SMP: State change: SMP_STATE_PAIR_REQ_RSP(3) ==> SMP_STATE_SEC_REQ_PENDING(2)
D (34316) BT_SMP: smp_send_cmd on l2cap cmd_code=0xb

D (34326) BT_SMP: smp_build_security_request

D (34326) BT_SMP: opcode=11 auth_req=0xd
D (34336) BT_SMP: smp_send_msg_to_L2CAP
D (34336) BT_SMP: result state = SMP_STATE_SEC_REQ_PENDING
[... 3669 lines cut ...]
D (57906) BT_SMP: smp_proc_pairing_cmpl

D (57906) BT_SMP: send SMP_COMPLT_EVT reason=0x0 sec_level=0x4

D (57916) BT_SMP: smp_reset_control_value

D (57916) BT_SMP: smp_remove_fixed_channel

D (57926) BT_SMP: smp_cb_cleanup

D (57926) BT_SMP: result state = SMP_STATE_IDLE

D (57926) BT_SMP: result state = SMP_STATE_IDLE

I (58036) SEC_GATTS_DEMO: key type = ESP_LE_KEY_PID
I (58036) SEC_GATTS_DEMO: remote BD_ADDR: 416d2dc5d8d1
I (58036) SEC_GATTS_DEMO: address type = 1
I (58036) SEC_GATTS_DEMO: pair status = success
I (58046) SEC_GATTS_DEMO: auth mode = ESP_LE_AUTH_REQ_SC_MITM_BOND
I (58056) SEC_GATTS_DEMO: Bonded devices number : 1

I (58056) SEC_GATTS_DEMO: Bonded devices list : 1

I (58066) SEC_GATTS_DEMO: 41 6d 2d c5 d8 d1
```

xiewenxiang, 2020-08-21:
Hi @dragahassan

Does the actual application scenario is ESP32 display the passkey and the phone enters the passkey? 
If so, it seems that the phone just used the previous 6 digits data.

Could you provide  the model of the mobile phone? I want to test the phone if I can find it.

xiewenxiang, 2020-08-24:
@dragahassan 

I am very very sorry that I cannot reproduce the question you mentioned. Maybe there are some differences between our operations

I use the gatt_security_server, A little bit of modification， `esp_ble_io_cap_t iocap = ESP_IO_CAP_NONE` => `esp_ble_io_cap_t iocap = ESP_IO_CAP_OUT`.  I'm sure you made this change too, otherwise ESP32 won't display the key here.

I reproduce the scenario you mentioned with several phones(itouch & iphone 8). The pairing was not successful, but the connection was not be disconnected. The connection was still in the clear text transmission state.

Do you hope that the connection will break when the pairing fails?

xiewenxiang, 2020-08-24:
[image]

xiewenxiang, 2020-08-24:
[image]

xiewenxiang, 2020-08-24:
ESP32 display the key: 123456
I input: 123456112233 & 123456666666 & 123456123456
All of them pair fail.

dragahassan, 2020-08-24:
Strange!, thank you for this check, maybe we have different kit or different SDK ?

I'm using: 

- **Kit version: DevKitC:** v4
- **Module or chip used:**  ESP32-WROOM-32D
- **IDF version:** v4.2-dev-1905-g625bd5eb1
- **Build System:** idf.py
- **Compiler version:** xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0

dragahassan, 2020-08-24:
I just test again with 2 other ESP32 but same kit version (_ESP32-WROOM-32D_) and Samsung S20+, pairing success.
probably this problem exit only on SDK v4.2-dev-XXXX.

xiewenxiang, 2020-08-25:
I completely tested in accordance with the conditions you said, but still failed to reproduce, all of which were pair fail.

Could you provide the bin file you generated? I want to test your firmware directly.

The attachment is the bin I generated.

`0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 sec_gatts_demo.bin`

[build.tar.gz](https://github.com/espressif/esp-idf/files/5120952/build.tar.gz)

dragahassan, 2020-08-25:
I test it with your bin file's, and it's success with 123456666666, I record a video (_from virtualbox_), I'm sure this security issue still exist, but I can't figure out yet the exact cause.

[Windows_10_64-2020-08-25T12-44-25-927300900Z.zip](https://github.com/espressif/esp-idf/files/5123850/Windows_10_64-2020-08-25T12-44-25-927300900Z.zip)

[build.zip](https://github.com/espressif/esp-idf/files/5123965/build.zip)

If we use the same non modified example, and the same SDK version and compiler.. why we have different size ? my firmware bin file (_sec_gatts_demo.bin_) is 690K, while your's 704K ? I know enabling log and other stuff in menuconfig can increase the size, but we are okay that we use non modified example, except for enabling `ESP_IO_CAP_OUT`.

closed 2021-08-23 by Alvin1Zhang