espressif/esp-idf #5763: xCoreID member missing from branch feature/freertos_plus_tlsf_allocator

ammaree, 2020-08-20:
I am trying to use the preview of FreeRTOS 10.x but have found that some functionality is missing. 2 items so far include:
a) xCoreId from the TaskStatus_t structure
b) pxTaskGetStackStart() function

Any plans to implement this?

ammaree, 2020-08-20:
I also noticed that the rest of the IDF components in this feature are about 2 months behind master commits, any plan to bring this more up to date?

ammaree, 2020-09-01:
@igrr @projectgus Any comment please?

uLipe, 2020-09-08:
Hi @ammaree , the support of FreeRTOS 10 is under final review, since it is a very sensible changing, we need to make sure this will not get anything broken. 

The next update will probably the support merge on the master, so this branch will not evolve much more than its current state.

About your questions:
a - We're figuring a way to make this part less dependent platform, as we plan to make our FreeRTOS 10 close as possible of the upstream FreeRTOS 10, although we're planning to re-add this field in another way without breaking the existing implementation.

b - This function seems not present in both FreeRTOS v8 and v10 public API. At the moment it seems there are no plans to support this function.

Please let me know if this information was helpful to you and don't hesitate to ask more questions :)

ammaree, 2020-09-08:
@uLipe 

Not sure if I am missing anything but xCoreID is already in the TCB structure so it is just a case of adding it to TaskStatus_t and setting it to the xCoreID value?

As far as pxTaskGetStackStart() is concerned, we can work without it but the absence of xCoreID in TaskStatus_t makes our task utilization tracing senseless.

Any idea when the support merge on master will happen, and whether xCoreID support will be included by then?

uLipe, 2020-09-09:
@ammaree , you're right, this member is already present, but this is a field is a dependent platform one, that is it, it is not present in FreeRTOS upstream API, so the initial plan is to put it back when FreeRTOS 10 changes got merged (we freeze the features of the experimental branch) or find a way that is more compatible to FreeRTOS upstream API without leaving the feature missing.

In any of the ways are chosen, the master branch with FreeRTOS10 support will keep the tracing of coreID of tasks :)

uLipe, 2020-09-15:
@ammaree , just to update:

* The issue with xCoreID missing member in FreeRTOS 10 is being fixed, it will keep the current usage, so it will not break your current task tracing;

* The pxTaskGetStackStart() was present in an early version of the IDF but got removed even in recent releases and in master branch, I'll investigate what it could be done at this point.

closed 2020-10-15 by espressif-bot

projectgus, 2020-10-15:
Hi @ammaree,

Thanks for being patient while we worked through this. The FreeRTOS 10 support has been merged to master in the linked commit, and will part of the upcoming ESP-IDF v4.3 release.

We haven't updated the feature preview branch for the last few fixes/changes, so you'll need to switch to master to get this. The preview branch will be removed soon.