espressif/esp-idf #5764: Secure Boot - flash isn't possible after building a secure boot enabled bootloader

achraf-boussaada, 2020-08-20:
## Environment

- Development Kit:    none
- Module or chip used:  ESP32-WROOM-32E
- IDF version : ESP-IDF v4.3-dev-907-g6c17e3a64
- Build System:         idf.py
- Compiler version : (crosstool-NG esp-2020r2) 8.2.0
- Operating System:    Linux
- Using an IDE?: yes, visual code studio
- Power Supply:        USB

## Problem Description

I'm trying to activate Secure Boot+ Flash Encryption on the ESP32 Chip but while following the documentation to enable Secure Boot V2 I had an error after running `idf.py bootloader` from the [6th step](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v2.html#how-to-enable-secure-boot-v2):
```
Executing action: bootloader
Running ninja in directory /workspace/esp/hello_world/build
Executing "ninja bootloader"...
[1/3] Performing build step for 'bootloader'
ninja: no work to do.

Bootloader build complete. To flash, run this command:
Traceback (most recent call last):
  File "/workspace/esp/esp-idf/tools/idf.py", line 790, in <module>
    main()
  File "/workspace/esp/esp-idf/tools/idf.py", line 708, in main
    cli(sys.argv[1:], prog_name=PROG)
  File "/workspace/.espressif/python_env/idf4.3_py3.6_env/lib/python3.6/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/workspace/.espressif/python_env/idf4.3_py3.6_env/lib/python3.6/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/workspace/.espressif/python_env/idf4.3_py3.6_env/lib/python3.6/site-packages/click/core.py", line 1290, in invoke
    return _process_result(rv)
  File "/workspace/.espressif/python_env/idf4.3_py3.6_env/lib/python3.6/site-packages/click/core.py", line 1224, in _process_result
    value = ctx.invoke(self.result_callback, value, **ctx.params)
  File "/workspace/.espressif/python_env/idf4.3_py3.6_env/lib/python3.6/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/workspace/esp/esp-idf/tools/idf.py", line 626, in execute_tasks
    self._print_closing_message(global_args, tasks_to_run.keys())
  File "/workspace/esp/esp-idf/tools/idf.py", line 514, in _print_closing_message
    print_flashing_message("Bootloader", "bootloader")
  File "/workspace/esp/esp-idf/tools/idf.py", line 477, in print_flashing_message
    cmd += flasher_args[key]["offset"] + " "
KeyError: 'bootloader'
```
Normally the next step is to flash the bootloader with a command of this sort `esptool.py write_flash`, but I got an error instead of providing me with the needed command as stated in the documentation.

Executing `idf.py flash monitor` didn't flash the bootloader and threw an error and the eFuses table is still the same and none of the relevant eFuses for Secure Boot or Flash Encryption were burnt.

Alvin1Zhang, 2020-08-21:
Thanks for reporting, we will look into.

projectgus, 2020-08-25:
Hi @achraf-boussaada ,

Glad you found a way to keep working. This is a bug in idf.py, we disable flashing of the bootloader by default when Secure Boot is on (to avoid accidental flashes) and idf.py didn't handle this case correctly.

A fix has been committed in commit dde6222cdceca978b2588fbf0eb25b8255741fba and will be pushed to GitHub soon. As it happens the fix was submitted before you reported the issue so this issue won't be updated automatically, but the link should "come live" once it is. Sorry for the inconvenience.

Alvin1Zhang, 2020-10-22:
@achraf-boussaada Thanks for reporting, the fix commit https://github.com/espressif/esp-idf/commit/dde6222cdceca978b2588fbf0eb25b8255741fba has come live, feel free to reopen if the issue still happens. Thanks.

closed 2020-10-22 by Alvin1Zhang