espressif/esp-idf #5765: memory error

ThomWarmerdam, 2020-08-20:
## Environment

- Development Kit:     both on ESP32-CAM and AI-THINKER
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  ESP32-S
- IDF version (run ``git describe --tags`` to find it):
    // v4.0-dev-837-g58df1d93b
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     Linux
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: No
- Power Supply:        external 5V

## Problem Description

When writing to memory some other memory location change too. It could be hardware, it could be software. I can reproduce the issue (in exactly the same way) on other ESP boards from other vendors. The software involved are (only) the compiler libraries. Between generating and diagnosing the problem there is only C code without function calls. When playing with the data I noticed that it also depends on the most significant bit of the bytes I write.

### Expected Behavior

I want that only the locations I write to to change

### Actual Behavior
neighboring bytes occasionally change but in a very deterministic way

### Steps to reproduce
Erase memory, sparsely write bytes, check if the other addresses are still erased.

// If possible, attach a picture of your setup/wiring here.
SW only

### Code to reproduce this issue

static void test(void)
{   
    int k, maxs = WAT_DSRF1 * WAT_NRTPL1;
    int8_t *ptr;

#if 1
    ptr = wp->tmpl1;    // from psram, and will generate errors
#else
    ptr = malloc(maxs); // reports no output
    if (!ptr) {outs("no memory..\n"); return;}
#endif

    for (k = 0; k < maxs; k++) ptr[k] = 0xff;

    for (k = 0; k < maxs; k += 21) // must be multiple of 3 for rgb colors
    {
        ptr[k+0] = 0x80;
        ptr[k+1] = 0x80;
        ptr[k+2] = 0x80;
    }

    for (k = 0; k < maxs; k += 3)
    {
        if ((ptr[k] != ptr[k+1]) || (ptr[k] != ptr[k+2]))
        {
            outs("mismatch found at adr = "); outh((int) &ptr[k]);
            outs(" ("); outh(0xff & ptr[k]);
            outs(" "); outh(0xff & ptr[k+1]);
            outs(" "); outh(0xff & ptr[k+2]);
            outs(")\n");
            outs("data before ("); outh(0xff & ptr[k-3]);
            outs(" "); outh(0xff & ptr[k-2]);
            outs(" "); outh(0xff & ptr[k-1]);
            outs(")\n");
            outs("data after ("); outh(0xff & ptr[k+3]);
            outs(" "); outh(0xff & ptr[k+4]);
            outs(" "); outh(0xff & ptr[k+5]);
            outs(")\n");
        }
    }
}   

## Debug Logs

**case #if 1**
water> water test
mismatch found at adr = 3f8e7c84 (ff 0 ff)
data before (ff ff ff)
data after (80 80 80)
water> 

**case #if 0**
water> water test
water> 

**On slightly other code I have seen:**
water> water init
mismatch found at adr = 3f896ce2 (0 0 ff)
data before (81 81 81)
data after (ff ff ff)
mismatch found at adr = 3f899022 (0 0 ff)
data before (81 81 81)
data after (ff ff ff)
mismatch found at adr = 3f89ac42 (0 0 ff)
data before (81 81 81)
data after (ff ff ff)
mismatch found at adr = 3f89b122 (0 0 ff)
data before (81 81 81)
data after (ff ff ff)
mismatch found at adr = 3f89d2e2 (0 0 ff)
data before (81 81 81)
data after (ff ff ff)
mismatch found at adr = 3f8a01c2 (0 0 ff)
data before (81 81 81)
data after (ff ff ff)
mismatch found at adr = 3f8a2922 (0 0 ff)
data before (81 81 81)
data after (ff ff ff)
mismatch found at adr = 3f8a2e02 (0 0 ff)
data before (81 81 81)
data after (ff ff ff)
mismatch found at adr = 3f8a2ec2 (0 0 ff)
data before (81 81 81)
data after (ff ff ff)
mismatch found at adr = 3f8d8462 (0 0 ff)
data before (7f 7f 7f)
data after (ff ff ff)
mismatch found at adr = 3f8d8b82 (0 0 ff)
data before (7f 7f 7f)
data after (ff ff ff)
mismatch found at adr = 3f8dace2 (0 0 ff)
data before (7f 7f 7f)
data after (ff ff ff)
mismatch found at adr = 3f8dc2a2 (0 0 ff)
data before (7f 7f 7f)
data after (ff ff ff)
mismatch found at adr = 3f8dd502 (0 0 ff)
data before (7f 7f 7f)
data after (ff ff ff)
mismatch found at adr = 3f8dd982 (0 0 ff)
data before (7f 7f 7f)
data after (ff ff ff)
mismatch found at adr = 3f8e3b62 (0 0 ff)
data before (7f 7f 7f)
data after (ff ff ff)

The striking thing is that it only happened on certain addresses; addresses that end at 0xn2, where n is even. On the simple example above the address is ending on 4.

ThomWarmerdam, 2020-08-20:
[image]
I have included a picture of the memory. I wrote only gray tones and the colored spots are the bugs. On the top half, we see in the upper right corner a red pixel produced by the program I  submitted. On the bottom half the picture caused by my regular program. I can make that program perform much worse (by writing a color) or much flawless by a different use of the sign bit.

igrr, 2020-09-02:
@ThomWarmerdam Do I understand correctly that the buffer is located in the external RAM? Is this issue reproducible when using the internal RAM as a buffer?

ThomWarmerdam, 2020-09-02:
Hi Ivan,

No, The 'strange' behavior I do not see when I work with internal RAM.

Thom.

closed 2020-09-16 by espressif-bot

igrr, 2020-09-16:
Hi @ThomWarmerdam, the linked commit updates the toolchain to work around the PSRAM cache issue. The change has also been backported to release/v4.0 branch, if you prefer to use that one. Please update the toolchain and try again.