espressif/esp-idf #5766: Guru Meditation (LoadProhibited) in _Unwind_RaiseException with ESP-IDF v3.3.2 toolchain

rtyle, 2020-08-20:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32 rev 1
- IDF version (run ``git describe --tags`` to find it): v3.3.2
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         external 3.3V

## Problem Description

Same as https://github.com/espressif/esp-idf/issues/5360 but with ESP-IDF v3.3.2 toolchain.

C++ exception handling and processor interrupt handling collide during stack unwinding.
This results in a fatal crash.

### Expected Behavior

No crash and reboot

### Actual Behavior

Continuous crash and reboot ...

...
I (0) cpu_start: Starting scheduler on APP CPU.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400e49d1  PS      : 0x00060c30  A0      : 0x800e2a9e  A1      : 0x3ffb5e10  
0x400e49d1: _Unwind_RaiseException at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/unwind.inc:136

A2      : 0x00000007  A3      : 0x3ffb5ee0  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060021  A7      : 0x00000000  A8      : 0x800e49d1  A9      : 0x3ffb5df0  
A10     : 0x00000000  A11     : 0x3ffb5e18  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x3ffb5e50  A15     : 0x00000002  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000018  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: a260619ec653f82ac9b78fa8d62fe746ab3f1d80bc5e6d2648f1b2c3424b0e5f

Backtrace: 0x400e49d1:0x3ffb5e10 0x400e2a9b:0x3ffb5ee0 0x400d0b06:0x3ffb5f20 0x40085e91:0x3ffb5f40
0x400e49d1: _Unwind_RaiseException at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/unwind.inc:136

0x400e2a9b: __cxa_throw at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc:82

0x400d0b06: main_task at /home/ross/eclipse/artlight/esp-idf/components/esp32/cpu_start.c:551

0x40085e91: vPortTaskWrapper at /home/ross/eclipse/artlight/esp-idf/components/freertos/port.c:403

Rebooting...

### Steps to reproduce

1. compile the following code (below, based on that provided in other issue)
2. flash
3. monitor

### Code to reproduce this issue

```cpp
#include <esp_timer.h>

unsigned factorial(unsigned n) {return 2 > n ? 1 : n * factorial(n - 1);}

static void periodicWork(void * notUsed) {factorial(19);}

extern "C" void app_main(void) {
    // perform some work periodically with the side-effect of interrupt processing
    uint64_t constexpr period {19};
    const esp_timer_create_args_t timerArgs {
        periodicWork,
        nullptr,
        ESP_TIMER_TASK,
        "timer"
    };
    esp_timer_handle_t timer {nullptr};
    esp_timer_create(&timerArgs, &timer);
    esp_timer_start_periodic(timer, period);

    // throw and catch exceptions as fast as we can
    while (true) {
        try {
            throw 0;
        } catch (...) {}
    }
}
```

Alvin1Zhang, 2020-08-21:
Thanks for reporting, we will look into.

0xjakob, 2020-08-21:
@rtyle Thanks for the reminder!
In fact, we need to backport the toolchain fix to older versions, too. The reason why that hasn't happened yet is that it isn't a simple fix in IDF but involves other repos, too. But I'll assign it a higher priority internally. Our gcc bugfix got improved and integrated into the gcc respository, so that's the fix which we will probably release as a new toolchain version. After integration into IDF, we will backport it.

Best,
Jakob

0xjakob, 2020-08-23:
@rtyle ETA was last month. I put this into high priority now. We want to fix and backport this within the next weeks. Will keep you updated.

plehmkuhl-ampere, 2020-10-02:
@0xjakob hey there! Made any progress on this issue? :)

0xjakob, 2020-10-13:
@plehmkuhl-ampere Sorry for the long delay and thank you for your patience (if there's still some left). We're beginning to update the toolchain for the older versions of IDF now. Since it's not the usual gcc 8 which we already ported down to version 4.0, the procedure and source is a bit different. This is why it got a bit lower priority.

rtyle, 2021-04-23:
It has been 8 months. Is this going to be fixed?

AxelLin, 2022-02-23:
FYI: v3.3 is going to EOL by end of this month.
https://www.espressif.com/sites/default/files/advisory_downloads/AR2022-002%20End-of-Life%20Advisory%20for%20ESP-IDF%20v3.3%20Release%20Branch%20EN.pdf

0xjakob, 2022-07-05:
We are going to close this issue since the release is EOL. Sorry that we couldn't find the resources in time to fix this issue on v3.3.

closed 2022-07-05 by 0xjakob