espressif/esp-idf #5767: `esp_hid `documentation is not included into programming guide

v1993, 2020-08-21:
## Problem Description

I'd like to learn how to use `esp_hid` component. However, it appears that its documentation is missing from online [programming guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/search.html?q=esp_hidh_init&check_keywords=yes&area=default#), while still being in files:

https://github.com/espressif/esp-idf/blob/6c17e3a64c02eff3a4f726ce4b7248ce11810833/components/esp_hid/include/esp_hidh.h#L105-L111

### Expected Behavior

Documentation for `esp_hid` component is available with the rest of docs.

### Actual Behavior

Documentation for `esp_hid` component is present in headers but is missing from online guide.

### Steps to reproduce

1. Visit ESP32 [programming guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/index.html) (latest).
2. Try to find any docs for `esp_hid` component.

Wth-Esp, 2021-09-16:
Hi @v1993,

Please refer to the demo we provided now, we will provide a programing guide after the USB firmware is supported.
Please wait for the update.

Thanks

AEFeinstein, 2021-11-08:
@Wth-Esp is there a development branch for USB firmware support? I'm curious to see how it's being done.

Wth-Esp, 2021-11-09:
@AEFeinstein  I'm sorry that I'm on another project now, my colleague will answer your question ok?

All changes will first be synced on the master branch on Github, so it's always right to update your local master branch to see the features.

Thanks

AEFeinstein, 2021-11-09:
@Wth-Esp , that's fine. I hope your other project is going well! I wasn't sure if development branches were visible, or if changes are private until they are synced with master. 

I'm also willing to help contribute here. I did some weekend work to get an ESP32-S2 enumerated as an HID gamepad with some small tweaks, but that was using `tusb` not `esp_hid`, which I didn't realize existed (https://github.com/AEFeinstein/esp-idf/commit/64f4918db0a687e093ad841704f8c9cb6a98db87). My changes do work, but I read the contribution guide and they're nowhere near ready for that.

Wth-Esp, 2021-11-09:
@AEFeinstein 

We always welcome the contributions~

The developing branches are not visible on github and I'm not very sure whom to ask, and whether the changes you made is proper to merged in the master branch. So, could you please make a PR and some of my colleagues will handle it.

Thanks