espressif/esp-idf #5768: Unable to reconnect to softAP after disconnecting from it.

mntolia, 2020-08-21:
Development Kit: ESP32-Wrover-Kit
Kit version (for WroverKit/PicoKit/DevKitC): v4
Module or chip used: ESP32-WROOM-32
IDF version : 4
Build System: CMake
Operating System: Linux
Using an IDE?: PlatformIO on Visual Studio Code
Power Supply: USB

I am using Arduino as an ESP-IDF Component

## Problem Description

When the esp32 disconnects from the WiFi it starts a softAP so that I can send commands to it via a webserver. The first time I connect to it I don't have any issues. But all devices(phones, laptops) fails to connect to it the second time unless I restart the esp32.
This is what gets printed to the serial monitor. I am not sure what to make of it. 
```
I (1311155) wifi:station: 50:3e:aa:ee:c1:01 join, AID=2, bgn, 40U
␛[0;31mE (1312315) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:␛[0m
␛[0;31mE (1312315) task_wdt:  - IDLE1 (CPU 1)␛[0m
␛[0;31mE (1312315) task_wdt:  - async_tcp (CPU 0/1)␛[0m
␛[0;31mE (1312315) task_wdt: Tasks currently running:␛[0m
␛[0;31mE (1312315) task_wdt: CPU 0: IDLE0␛[0m
␛[0;31mE (1312315) task_wdt: CPU 1: loopTask␛[0m
I (1313185) wifi:station: 70:bb:e9:5c:c4:1e leave, AID = 1, bss_flags is 134243, bss:0x3ffd7d30
I (1313185) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<0,0>, prof:1
I (1314975) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<0,0>, prof:1
I (1314985) wifi:station: 70:bb:e9:5c:c4:1e join, AID=1, bgn, 20
I (1315155) wifi:station: 50:3e:aa:ee:c1:01 leave, AID = 2, bss_flags is 134242, bss:0x3ffe5ef8
I (1315155) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<0,0>, prof:1
␛[0;31mE (1317315) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:␛[0m
␛[0;31mE (1317315) task_wdt:  - IDLE1 (CPU 1)␛[0m
␛[0;31mE (1317315) task_wdt:  - async_tcp (CPU 0/1)␛[0m
␛[0;31mE (1317315) task_wdt: Tasks currently running:␛[0m
␛[0;31mE (1317315) task_wdt: CPU 0: IDLE0␛[0m
␛[0;31mE (1317315) task_wdt: CPU 1: loopTask␛[0m
I (1318985) wifi:station: 70:bb:e9:5c:c4:1e leave, AID = 1, bss_flags is 134242, bss:0x3ffd7d30
I (1318985) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<0,0>, prof:1
␛[0;31mE (1322315) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:␛[0m
␛[0;31mE (1322315) task_wdt:  - IDLE1 (CPU 1)␛[0m
␛[0;31mE (1322315) task_wdt:  - async_tcp (CPU 0/1)␛[0m
```
## Code

This  is the event handler function if at all it helps:
```cpp
void WiFiEvent(WiFiEvent_t event)
{
    Serial.printf("[WiFi-event] event: %d\n", event);
    fmem.begin(flash_namespace);
    String ssid = fmem.getString(ssid_key);
    String password = fmem.getString(password_key);
    switch (event)
    {
    case SYSTEM_EVENT_WIFI_READY:
        Serial.println("WiFi interface ready");
        break;
    case SYSTEM_EVENT_SCAN_DONE:
        Serial.println("Completed scan for access points");
        break;
    case SYSTEM_EVENT_STA_START:
        Serial.println("WiFi client started");
        break;
    case SYSTEM_EVENT_STA_STOP:
        Serial.println("WiFi clients stopped");
        break;
    case SYSTEM_EVENT_STA_CONNECTED:
        Serial.println("Connected to access point");
        toggle_softap(false);
        toggle_mdnsserver();
        //MQTT attemps auto reconnection. Just the initial connection needs to be done
        if (!mqtt_connected)
        {
            String mqtt_username = fmem.getString(mqtt_username_key);
            String mqtt_password = fmem.getString(mqtt_password_key);
            if (!mqtt_username.isEmpty() && !mqtt_password.isEmpty())
            {
                if (setup_mqtt_client(mqtt_event_handler_cb, mqtt_username, mqtt_password) == 0)
                {
                    mqtt_connected = true;
                }
            }
        }
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        Serial.println("Disconnected from WiFi access point");
        toggle_softap();
        break;
    case SYSTEM_EVENT_STA_AUTHMODE_CHANGE:
        Serial.println("Authentication mode of access point has changed");
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        Serial.print("Obtained IP address: ");
        Serial.println(WiFi.localIP());
        break;
    case SYSTEM_EVENT_STA_LOST_IP:
        Serial.println("Lost IP address and IP address is reset to 0");
        break;
    case SYSTEM_EVENT_STA_WPS_ER_SUCCESS:
        Serial.println("WiFi Protected Setup (WPS): succeeded in enrollee mode");
        break;
    case SYSTEM_EVENT_STA_WPS_ER_FAILED:
        Serial.println("WiFi Protected Setup (WPS): failed in enrollee mode");
        break;
    case SYSTEM_EVENT_STA_WPS_ER_TIMEOUT:
        Serial.println("WiFi Protected Setup (WPS): timeout in enrollee mode");
        break;
    case SYSTEM_EVENT_STA_WPS_ER_PIN:
        Serial.println("WiFi Protected Setup (WPS): pin code in enrollee mode");
        break;
    case SYSTEM_EVENT_AP_START:
        Serial.println("WiFi access point started");
        toggle_dnsserver(false);
        toggle_mdnsserver(true);
        if (ssid && password)
        {
            WiFi.begin(ssid.c_str(), password.c_str(), 6);
        }
        break;
    case SYSTEM_EVENT_AP_STOP:
        Serial.println("WiFi access point  stopped");
        break;
    case SYSTEM_EVENT_AP_STACONNECTED:
        Serial.println("Client connected");
        wifisoftap_client_connected = true;
        break;
    case SYSTEM_EVENT_AP_STADISCONNECTED:
        Serial.println("Client disconnected");
        wifisoftap_client_connected = false;
        break;
    case SYSTEM_EVENT_AP_STAIPASSIGNED:
        Serial.println("Assigned IP address to client");
        break;
    case SYSTEM_EVENT_AP_PROBEREQRECVED:
        Serial.println("Received probe request");
        break;
    default:
        break;
    }
}
```
What could be the issue?

Alvin1Zhang, 2020-08-21:
Thanks for reporting, we will look into.

HarveyRong-Esp, 2020-09-09:
Hi @mntolia, 

```
   [0;31mE (1312315) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:␛[0m
␛[0;31mE (1312315) task_wdt:  - IDLE1 (CPU 1)␛[0m
␛[0;31mE (1312315) task_wdt:  - async_tcp (CPU 0/1)␛[0m
␛[0;31mE (1312315) task_wdt: Tasks currently running:␛[0m
␛[0;31mE (1312315) task_wdt: CPU 0: IDLE0␛[0m
␛[0;31mE (1312315) task_wdt: CPU 1: loopTask␛[0m
```
It seems to be caused by the appearance of task_wdt in loopTask, Please check lookTask related code.

mntolia, 2020-09-11:
The issue did not occur after I upgraded to the latest espressif platform. Thanks anyway!

mntolia, 2020-09-27:
Hi,
This issue is occurring again. This is the code in the loop function:
```cpp
void loop()
{
    //Process DNS Queries if SoftAP is activated
    if (wifisoftap_state && WiFi.softAPgetStationNum() > 0)
    {
        dnsServer.processNextRequest();
    }
    delay(1000);
}
```

mntolia, 2020-09-27:
I figured it out. It happend when async_tcp does not close a connection

HarveyRong-Esp, 2020-11-05:
Hi @mntolia，
Can you provide the minimal code for us to emerge locally? Please note that once the code is published here, it means it will be open source.

closed 2021-08-23 by Alvin1Zhang