espressif/esp-idf #5769: BLE host reset (Nimble) causes runtime error Core 0 panic'ed

DonSimonus, 2020-08-21:
## Environment

- Development Kit: None
- Module or chip used:  ESP32-WROVER-E
- IDF version: v4.0.1-412-g820621687
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System: Windows
- Environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply: External 3.3V

## Problem Description

If the ESP BLE central (Nimble BLE) tries to connect to a peripheral which disappears while discovering (e.g. turned off) the Nimble host will reset. This sometimes leads to a runtime error.

### Expected Behavior
BLE host can handle with disappearing peripherals during connection establishment (connecting, pairing, discovering).

### Actual Behavior
Disappearing BLE peripherals during connection establishment (connecting?, pairing?, discovering) causes BLE Nimble host resets (sometimes).

### Steps to reproduce

1. Set BLE (Nimble) central to scanning mode
2. Turn on a peripheral
3. Turn off the peripheral during connection establishment

### Code to reproduce this issue
I cannot provide my application code, but from the backtrace it looks like the problem is not in my application.

## Debug Logs

```
...
I (390422) ble: Sending BLE_SCANNING(error=ESP_FAIL)
I (390573) ble: Sending BLE_CONNECTING(error=ESP_OK)
I (390736) ble: Sending BLE_PAIRING(error=ESP_OK)
I (392742) ble: Sending BLE_DISCOVERING(error=ESP_OK)
E (392744) ble: Service discovery FAILED (0x7)
E (392746) ble: start_scanning() FAILED (22)
E (392758) ble: Nimble Host reset (reason=19)

Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
Core 0 register dump:
PC      : 0x4001742d  PS      : 0x00060733  A0      : 0x8004a7b4  A1      : 0x3ffb5400
A2      : 0x3ffb2c3c  A3      : 0x4002863e  A4      : 0x3ffb40d8  A5      : 0x3ffb8360
A6      : 0x3ffb8360  A7      : 0x00000000  A8      : 0x80089180  A9      : 0x3ffb53e0
A10     : 0xffffffff  A11     : 0x3ffb2eb4  A12     : 0x00060723  A13     : 0x00000001
A14     : 0x00060723  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x00000003
EXCVADDR: 0x4002863e  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff

ELF file SHA256: 9e8cb0fef476e79a

Backtrace: 0x4001742a:0x3ffb5400 0x4004a7b1:0x3ffb5420 0x4004b466:0x3ffb5440 0x40046c1f:0x3ffb5460 0x4010ed61:0x3ffb5490 0x401100fb:0x3ffb54b0 0x40054d4b:0x3ffb5500 0x40055afd:0x3ffb5520 0x40115a76:0x3ffb5540 0x40051b82:0x3ffb5560 0x40052a59:0x3ffb5580 0x4001a637:0x3ffb55a0 0x40019d11:0x3ffb55d0 0x40055b4d:0x3ffb55f0 0x40107403:0x3ffb5610 0x401079d9:0x3ffb5630 0x40090bc9:0x3ffb5660
0x4010ed61: r_lld_evt_init_wrapper at ??:?

0x401100fb: r_lld_init at ??:?

0x40115a76: r_rwip_reset_ext at ??:?

0x40107403: r_rw_schedule at ??:?

0x401079d9: btdm_controller_task at ??:?

0x40090bc9: vPortTaskWrapper at C:/Work/esp-idf/components/freertos/port.c:143

I (17027) esp_core_dump_flash: Save core dump to flash...
I (17033) esp_core_dump_common: Found tasks: (16)!
D (17039) esp_core_dump_port: Current task 3ffb56f4 EXIT/PC/PS/A0/SP 713d481 4001742d 60733 8004a7b4 3ffb5400
[... 164 lines cut ...]
.coredump.tasks.data 0x3ffd2a00 0x220 RW
.coredump.tasks.data 0x3ffd17d8 0x16c RW
.coredump.tasks.data 0x3ffd1610 0x1c0 RW
.coredump.tasks.data 0x3ffe1270 0x16c RW
.coredump.tasks.data 0x3ffe10d0 0x198 RW
.coredump.tasks.data 0x3ffbd0f8 0x16c RW
.coredump.tasks.data 0x3ffbcf30 0x1c0 RW
.coredump.tasks.data 0x3ffba294 0x16c RW
.coredump.tasks.data 0x3ffcee40 0x454 RW
.coredump.tasks.data 0x3ffb9fb0 0x16c RW
.coredump.tasks.data 0x3ffb9e00 0x1a8 RW
.coredump.tasks.data 0x3ffd65f4 0x16c RW
.coredump.tasks.data 0x3ffd63f0 0x1fc RW
.coredump.tasks.data 0x3ffb9a3c 0x16c RW
.coredump.tasks.data 0x3ffb7cd0 0x1c4 RW
.coredump.tasks.data 0x3ffb773c 0x16c RW
.coredump.tasks.data 0x3ffb7570 0x1c4 RW
.coredump.tasks.data 0x3ffd4128 0x16c RW
.coredump.tasks.data 0x3ffd3f10 0x210 RW
.coredump.tasks.data 0x3ffdfaf4 0x16c RW
.coredump.tasks.data 0x3ffdf940 0x1ac RW

===================== ESP32 CORE DUMP END =====================
===============================================================
```

Alvin1Zhang, 2020-08-21:
Thanks for reporting, we will look into.

chhajedji, 2020-09-24:
Hi @DonSimonus 

 I tried to reproduce the issue with given details, unfortunately I could not reproduce the issue. However from the nature of the issue it looks looks like known issue. I am attaching BT lib to try, please let me know if it resolves the issue. 

Replace the file at `$IDF_PATH/components/bt/controller/libbtdm_app.a` with this file. Let me know if it solves the issue.

[lib.tar.gz](https://github.com/espressif/esp-idf/files/5275889/lib.tar.gz)

chhajedji, 2020-10-04:
@DonSimonus @Alvin1Zhang please close this issue.

Alvin1Zhang, 2020-10-12:
@DonSimonus Thanks for reporting, feel free to reopen if any updates. Thanks.

closed 2020-10-12 by Alvin1Zhang