espressif/esp-idf #5779: .bss segment place psram, flash erase errors

DURBIN-LI, 2020-08-24:
### describe
config allow .bss segment place in external memory, OTA upgradeing, flash erase will appear program errors
**idf version：V4.2**

tks，best wish

Alvin1Zhang, 2020-08-24:
@DURBIN-LI Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

DURBIN-LI, 2020-08-24:
@Alvin1Zhang update issue

DURBIN-LI, 2020-08-24:
@Alvin1Zhang issue has been update

Alvin1Zhang, 2020-08-24:
Thanks for updates and providing more details, we will look into.

shubhamkulkarni97, 2020-08-25:
Hi @DURBIN-LI,

I could reproduce the issue by checking out to the commit you mentioned (i.e. `v4.2-dev-1575-ga3520970f`). However, this issue is fixed and is not observed on latest release/v4.2 branch.

Updating your branch to latest commit should fix the issue.

Thanks,
Shubham

DURBIN-LI, 2020-08-25:
@shubhamkulkarni97 Hi，Updated v4.2 branch，the issue has been fixed
tks，best wish

closed 2020-08-25 by DURBIN-LI