espressif/esp-idf #5781: esp_hid: heap corruption when using classic bt

v1993, 2020-08-24:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
    // v4.3-dev-907-g6c17e3a64
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

When using `esp_hidh_dev_output_set` with BT Classic device, heap corruption occurs. Error looks like this:

```
CORRUPT HEAP: Bad tail at 0x3ffb7f23. Expected 0xbaad5678 got 0xbaad5690
```

Problematic allocation is in `esp_bt_hidh_dev_report_write`:

https://github.com/espressif/esp-idf/blob/6c17e3a64c02eff3a4f726ce4b7248ce11810833/components/esp_hid/src/bt_hidh.c#L245

It have off-by one error that can be fixed by replacing `14` with `15`. As 14 is a magic number without obvious meaning I'm not sure it it is the only place having this issue (I'm just starting to use this component). It could be in device mode counterpart as well and/or BLE interface too.

### Expected Behavior

Using `esp_hidh_dev_output_set` does not corrupt heap.

### Actual Behavior

`esp_hidh_dev_output_set` corrupts heap when used with BT Classic.

### Steps to reproduce

1. Set heap corruption detection to at least "light"
2. Call `esp_hidh_dev_output_set` on valid BT Classic HID device
3. Either wait until detector triggers down BT stack or check using `heap_caps_check_integrity_all(true)`

### Code to reproduce this issue

Can't really provide at it depends on HID device you are interfacing with (report ID/length will vary). Insert call with valid output HID report to `esp_hid_host` example at `ESP_HIDH_OPEN_EVENT` case.

## Debug Logs

N/A as crash is triggered by heap corruption

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

v1993, 2020-08-24:
Additional problem with this function: if allocation fails, there is no way to know it did (function will return `ESP_OK`).

me-no-dev, 2020-08-24:
Because not the full bluedroid classic HID host stack is implemented, some upper layer functionality is missing. This offset was one such case that had to be hacked in. I will try to dig up why it might have changed and how to better define it.

closed 2020-10-15 by espressif-bot