espressif/esp-idf #5782: requirements.txt (esp-idf/requirements.txt) has a dependency issue that cause the build to fail

hasantrq, 2020-08-24:
For the latest master branch.

[image]

The requirements.txt pins **gdbgui>=0.13.2.0** which requires **pygdbmi<=0.9.0.2**. 

However the latest version of **gdbgui, 0.14.0.0** requires ** pygdbmi<0.11, >=0.10.0.0b0**

Please take a look into this

skieast, 2020-08-24:
Looks like pygdmi has a pre-release 10.0.0 which should fix this. I have changed my requirements.txt to gdbgui==0.13.2.0 for now. 
I guess once released both dependencies will change.

leeebo, 2020-08-25:
I gdbgui>=0.13.2.0 pygdbmi<=0.9.0.2.

hasantrq, 2020-08-25:
> Fixed the version gdbgui==0.13.2.0 pygdbmi==0.9.0.2 can temporarily skip the problem.

I am running a CD pipeline and it automatically replicates the esp-idf repo. and I its fairly difficult to change the requirements.txt over there. Can I use a pip command to somehow downgrade after the newest package has been installed?

leeebo, 2020-08-25:
> > Fixed the version gdbgui==0.13.2.0 pygdbmi==0.9.0.2 can temporarily skip the problem.
> 
> I am running a CD pipeline and it automatically replicates the esp-idf repo. and its fairly difficult to change the requirements.txt over there. Can I use a pip command to somehow downgrade after the newest package has been installed?

Yes, using `pip install` you can install any available version. eg, change dir to `/home/{user}/.espressif/python_env/idf4.3_py3.6_env/bin` then run `./pip3 install gdbgui==0.13.2.0 pygdbmi==0.9.0.2`

Francois-Belanger, 2020-08-30:
Same issue on Mac OsX running "pip3 install gdbgui==0.13.2.0 pygdbmi==0.9.0.2". fixed the issue.  Thanks

feklee, 2020-09-02:
Same issue with branch `release/v4.2`:

    gdbgui 0.14.0.0 has requirement pygdbmi<0.11,>=0.10.0.0b0, but you'll have pygdbmi 0.9.0.2 which is incompatible.

rafasimionato, 2020-09-04:
This is preventing me to succeed ESP-IDF installation at my system. I am using Ubuntu over WSL in Windows 10. See logs bellow when running ./install.sh:

Successfully uninstalled pygdbmi-0.10.0.0
ERROR: After October 2020 you may experience errors when installing or updating packages. This is because pip will change the way that it resolves dependency conflicts.

We recommend you use --use-feature=2020-resolver to test your packages with the new resolver before it becomes the default.

gdbgui 0.14.0.0 requires pygdbmi<0.11,>=0.10.0.0b0, but you'll have pygdbmi 0.9.0.2 which is incompatible.
Successfully installed pygdbmi-0.9.0.2

rafasimionato, 2020-09-04:
> Fixed the version gdbgui==0.13.2.0 pygdbmi==0.9.0.2 can temporarily skip the problem.

Worked for me ... I am using Ubuntu over WSL in Windows 10.

closed 2020-09-16 by espressif-bot